PartialVolume/shredos.x86_64 #157: 23.5 hangs on IBM x3650 while 23.3 works

uli42, 2023-07-18:
I have tested the latest 23.5 pre-release on old IBM x3650 machines using PXE. grub manages to load the shredos image but shredos gets stuck while booting. I have increased to log level to 7 and the console to tty and now I see this.

[image]

Afterthat nothing happens anymore.

PartialVolume, 2023-07-18:
Does it work with nomodeset on the command line?

uli42, 2023-07-18:
Will test tomorrow.

PartialVolume, 2023-07-18:
If it does work, can you do a lspci and see what graphics it's running. I can check it's not been left out off 23.5 for some reason.

uli42, 2023-07-19:
`nomodeset` does not help:
[image]

Here's the lspci output from a similar machine:
$ lspci -vv -s 04:00.0
04:00.0 VGA compatible controller: Matrox Electronics Systems Ltd. G200eR2 (prog-if 00 [VGA controller])
        Subsystem: IBM Device 0405
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 64 (4000ns min, 8000ns max), Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 20
        NUMA node: 0
        Region 0: Memory at c6000000 (32-bit, prefetchable) [size=16M]
        Region 1: Memory at c57fc000 (32-bit, non-prefetchable) [size=16K]
        Region 2: Memory at c5800000 (32-bit, non-prefetchable) [size=8M]
        Expansion ROM at <unassigned> [disabled]
        Capabilities: [dc] Power Management version 1
                Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Kernel driver in use: mgag200
        Kernel modules: mgag200

PartialVolume, 2023-07-19:
That snapshot doesn't make sense to me. 
[image]

nomodeset should stop the kernel switching to the DRM drivers. Is nomodeset in both grub files? Legacy & EFI.

ShredOS 23.5 has the mgag200 DRM driver, so it's not a missing driver issue. Earlier versions of ShredOS don't have that driver. 
```
Symbol: DRM_MGAG200 [=y]                                                                                                 │  
  │ Type  : tristate                                                                                                         │  
  │ Defined at drivers/gpu/drm/mgag200/Kconfig:2                                                                             │  
  │   Prompt: Matrox G200                                                                                                    │  
  │   Depends on: HAS_IOMEM [=y] && DRM [=y] && PCI [=y] && MMU [=y]                                                         │  
  │   Location:                                                                                                              │  
  │     -> Device Drivers                                                                                                    │  
  │       -> Graphics support                                                                                                │  
  │ (1)     -> Matrox G200 (DRM_MGAG200 [=y])                                                                                │  
  │ Selects: DRM_GEM_SHMEM_HELPER [=y] && DRM_KMS_HELPER [=y] && I2C [=y] && I2C_ALGOBIT [=y]
```

uli42, 2023-07-19:
even with 'module_blacklist=mgag200drmfb' it behaves the same.

BTW: in the setup I also had --autonuke and --autopoweroff for nwipe and the system powered off within a minute or less after above output. So it _seems_ that nwipe was started and I was only not seeing the output. Shredding should take much longer, though, so this situation is weird...

uli42, 2023-07-19:
Update: with nomodeset it is working! I had accidently added nomodeset to the nwipe options...

PartialVolume, 2023-07-19:
Excellent, so it looks like there might possibly be some bug in the mgag200drmfb driver., also can't rule out a hardware issue with the matrox chip unless it works fine with other distros?

PartialVolume, 2023-07-19:
>even with 'module_blacklist=mgag200drmfb' it behaves the same.

I'm not 100% sure about this as I can't test it, but I think blacklisting may not have worked because the builtin driver is called mgag200, so  'module_blacklist=mgag200' may have worked.

The driver is:
/kernel/drivers/gpu/drm/mgag200/mgag200.ko

PartialVolume, 2023-11-08:
Please reopen this issue if not fixed by 2023.08.2_25.0_x86-64_0.35

closed 2023-11-08 by PartialVolume