PartialVolume/shredos.x86_64 #159: Automatically wipe all attached storage media without the system one

Nebuli1, 2023-07-23:
Hello, I am trying to make a flash drive that will clear all connected media after itself. However, I don't know how to safely exclude the flash drive itself. I have read that excluding the drive by sda does not guarantee that it will always be my pendrive, I tried to exclude by uuid but the packages are missing. How do you guys deal with this? 

My configuration file.
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 nwipe_options="--method=zero --verify=off --noblank --autonuke --autopoweroff /dev/sda"
}

PartialVolume, 2023-07-23:
If I've understood your use case correctly, all you need to do is add `--nousb` , this will exclude all devices from being wiped that are attached via USB, it will still enumerate all SATA, SAS, NvMe etc.

```
set default="0"
set timeout="0"

menuentry "shredos" {
linux /boot/shredos console=tty3 loglevel=3 nwipe_options="--method=zero --verify=off --noblank --autonuke --autopoweroff --nousb /dev/sda"
}
```

PartialVolume, 2023-07-23:
However, if you only want to exclude USB devices that contain ShredOS, but enumerate all other USB , SATA, SAS and NvMe  devices then at the moment that's not possible except by identifying what the device name is and using --exclude, however like you say a USB device name can change so that command is not reliable on the linux command line.

If the above feature is what you are looking for and --nousb isn't sufficient then I can add this as a new feature. I can have ShredOS determine which drives contain ShredOS and exclude those from the wipe. I'd probably go with the option --noshredos.

On a connected issue, I need to add an option that delays nwipes launch as often the USB devices haven't initialised so often you don't see any USB devices until you Control C and nwipe restarts.

ITJamie, 2023-07-25:
i think it would be a good idea to have something like --no-wipe-boot-drive. so it can ignore the bootdrive (USB, sata or whatever) but wipe everything else. maybe even variants, --no-wipe-boot-drive

Nebuli1, 2023-07-28:
I've been reading for days and doing tests. I can't manage without help. Actually the nwipe boot delay option and the --no-wipe-boot-drive parameter solves my problem. 
The problem occurred because my disk cleaning computer has no nvme or PCI slot so I connect nvme drives via USB adapter.

PartialVolume, 2023-07-28:
@Nebuli1 I'm assuming your NvMe doesn't initially appear. You are aware that if you Control C in nwipe/ShedOS then press space-bar before starting your wipe all the USB attached devices will now be listed, when running in GUI mode or are you trying to autonuke with no GUI.

It's probably time I put a small check in the code to launch nwipe once the USB devices have been detected and launch anyway after 5 seconds.

Nebuli1, 2023-07-28:
Thanks I didn't think about autonuke without GUI, I will test.

For now I'm working as you suggest.  USB delay will be great. I've read other topics where you wrote about the autonuke problem so I guess this will be the easiest and fastest yet effective solution.

fieldofgreen, 2023-10-19:
> If the above feature is what you are looking for and --nousb isn't sufficient then I can add this as a new feature. I can have ShredOS determine which drives contain ShredOS and exclude those from the wipe. I'd probably go with the option --noshredos.

any update on this feature? I can exclude the internal disk of the device I use just fine since it always appears first in the list and I swear my USB used to always show in the second slot but It hasn't been doing that recently. Instead it is usually the last drive in the list or second to last. If I am always wiping 2 drives at a time I believe I can predict where the USB will be in the list but if I only do one drive then the USB will move and be visible again.

Nebuli1, 2023-10-20:
please take a peek at this PartialVolume topic

https://github.com/PartialVolume/shredos.x86_64/issues/163

Wait and see if an update has been issued. I for one can't wait, but I have time to prepare my equipment :)

PartialVolume, 2023-10-20:
Thanks for the reminder, I'll make sure this is implemented in the next release.

PartialVolume, 2023-11-06:
Just to let you know, I'm now working on building the latest ShredOS with kernel 6.4.16 with the latest nwipe v0.35 plus adding #163. Should hopefully release this very soon.

fieldofgreen, 2023-11-22:
Was the ability to exclude the Boot drive added to the most recent build or is that still in the works?

PartialVolume, 2023-11-22:
> Was the ability to exclude the Boot drive added to the most recent build or is that still in the works?

No, that will be in the next version. Hopefully shouldn't be too long, just got to add a few more changes.

Nebuli1, 2023-12-13:
@PartialVolume Has any solution been implemented to exclude the Shredos system drive?

[image]

PartialVolume, 2023-12-13:
No, I've not added that option yet,  I've not forgotten.

I will go with  --no-wipe-boot-drive as the option name. It will be in the next release I do, just not sure when I'm going to write the code as I'm pretty busy leading up to Christmas, but I'll see if I can squeeze it in. It will exclude both the ShredOS and Ventoy USB drives so the USB drive you booted from does not show up in the list of enumerated drives in either gui or non gui modes.

Nebuli1, 2023-12-14:
Of course I understand and appreciate. Thank you. I will look for this option in the next update.

Nebuli1, 2024-02-27:
@PartialVolume Hi. Please don't take me the wrong way. I am asking in good faith. Did you manage to implement a new option ? I cannot find it. Thanks.

PartialVolume, 2024-02-27:
It's currently being tested, it will be committed to the master code in the next couple of days. The kernel command to enable it will be `shredos_exclude_boot_disc`

PartialVolume, 2024-02-28:
I'm also making it a little easier for Ventoy users. They can unpack the .img etc edit the grub.cfg and repack etc but that's all very time consuming. Instead all you need to do is create the following empty file on the USB  stick that you don't want to appear in nwipe's interactive or get wiped in autonuke mode. `/etc/shredos/shredos_exclude`  the PDF's will still written to the excluded drive it will just appear invisible to nwipe.

Nebuli1, 2024-02-28:
@PartialVolume ok, as soon as you post the updated img image I will immediately try to test the solution. Thanks

PartialVolume, 2024-03-02:
committed #228

PartialVolume, 2024-03-04:
Updated [README.md-Exclude ShredOS boot drive](https://github.com/PartialVolume/shredos.x86_64?tab=readme-ov-file#how-to-exclude-the-fat-formatted-shredos-boot-drive-from-nwipe-interactive-and-autonuke-modes) with two alternate methods to exclude the ShredOS fat formatted boot drive.

Nebuli1, 2024-03-19:
Thank you. I did the tests and unfortunately my flash drive was wiped ;/ I attach my CFG file.

set default="0"
set timeout="0"
menuentry "shredos" {
linux /boot/shredos console=tty3 loglevel=3 shredos_exclude_boot_disc="yes" nwipe_options="--method=zero --verify=off --noblank --exclude=/dev/sda --PDFreportpath=/ --autopoweroff --autonuke" nomodeset
}

Have I done something wrong?

PartialVolume, 2024-03-19:
You don't mention whether you built ShredOS from source yourself.  I've committed the changes to the source but not published a release yet, so no new .iso / .img files.

However, if you are building from source, let me know and I'll take a look at the problem.

Nebuli1, 2024-03-19:
Sorry I didn't read that. I downloaded the .img. I will wait for the new .img version. :) Thanks.

Nebuli1, 2024-05-31:
@PartialVolume I must admit that I am actively following Github. I've been trying to compile img unfortunately on Windows and giving up ;/ I read recently that you've released an RC version of the
https://github.com/PartialVolume/shredos.x86_64/issues/257#issuecomment-2140333330
Is the shredos_exclude_boot_disc=yes option implemented in it?

Is this a good configuration file? 

set default="0"
set timeout="0"
shredos_exclude_boot_disc="yes"
menuentry "shredos" {
linux /boot/shredos console=tty3 loglevel=3 shredos_exclude_boot_disc="yes" nwipe_options="--method=zero --verify=off --noblank --exclude=/dev/sda --PDFreportpath=/ --autopoweroff --autonuke" nomodeset
}

PartialVolume, 2024-05-31:
>Is the shredos_exclude_boot_disc=yes option implemented in it?

Yes.

>Is this a good configuration file?

Not quite, you seem to have a second `shredos_exclude_boot_disc` inside the nwipe_options. That should be removed. Also the --PDFreportpath is unnecessary as you have it set to / which is it's default value in ShredOS, so that can be removed.

With the above changes made it would look as shown below. I have also put nomodeset directly after loglevel and before any ShredOS or nwipe_options. Not for any valid technical reason, just because I like to see any ShredOS or nwipe options came after everything else. That's also the way I test it.

```
set default="0"
set timeout="0"

menuentry "shredos" {
linux /boot/shredos console=tty3 loglevel=3 shredos_exclude_boot_disc="yes" nwipe_options="--method=zero --verify=off --noblank --autopoweroff --autonuke"
}
```

Nebuli1, 2024-06-01:
I tested the solution. Everything works perfectly, thank you.

PartialVolume, 2024-06-01:
> I tested the solution. Everything works perfectly, thank you.

Excellent, thanks for letting me know.