PartialVolume/shredos.x86_64 #165: Unable to detect internal SSD on Surface 3

jimmothy-beep, 2023-10-03:
Hi there,

as mentioned in the title, Shredos is unable to detect the internal SSD regardless of how many times I restart the software.

PartialVolume, 2023-10-03:
Hi,

What are the details? such as but not limited to

- make/model of the computer
- make/model of the SSD
- configuration, i.e how many drives on the system?
- version of ShredOS
- snapshot of what you see
- nwipe log
- etc

PartialVolume, 2023-10-03:
Hi,

Can you post the output of lspci, that should tell me what I need to know. Thanks.

PartialVolume, 2023-10-03:
Also, take a look in the bios and see if their is any settings for the SSD, should be AHCI mode rather than anything else like raid.

PartialVolume, 2023-11-08:
I'm adding adding the Microsoft Surface platform specific device drivers to the next release 2023.08.2_25.0_x86-64_0.35, due to be released imminently. Also adding ChromeOS specific hardware support.

Hopefully the Microsoft Surface specific device drivers will fix this.

PartialVolume, 2023-11-08:
Please reopen issue if not fixed by 2023.08.2_25.0_x86-64_0.35.

closed 2023-11-08 by PartialVolume