PartialVolume/shredos.x86_64 #166: Hp Proliant G8 Microserver

fieldofgreen, 2023-10-04:
I'm not sure if I'm misunderstanding something obvious here but I'm trying to run shredOS on this server to just use it as a host for wiping multiple drives with USB sata connectors. I've gotten it working fine before with multiple devices so it's not the peripherals. 
It's my understanding that I should be able to plug in my USB with shredos, turn on the server and it boots into shredos instead of anything else that is on the server. I dont know if that requires the slow hp booting screens or not. However I have been completely unable to get it to even try booting into the USB and I dont know why. I have tried different ports, using the SD card slot, I even tried the newest version in the preview section here. Nothing. My next thought that I haven't been able to try yet is using the hp provisioning tool to install windows 10 on the machine and maybe that will allow the server to recognize the drive maybe?? 
I want to try upgrading firmware and things however that process is convoluted and pay walled by hp so ive currently tabled that thought unless I need that firmware then I guess I need to try to track it down somehow. Or just go back to my janky little setup using a Datto device and running out of USB slots. I know the server was able to turn on just fine before I started messing with it. Any thoughts? 

[image]

PartialVolume, 2023-10-04:
Sounds like a bios issue. I've got a load of HP towers, none of which will boot from USB, the bios makes no mention of being able to boot from USB so I've always assumed some HP systems don't boot from USB. In that case I've put ShredOS on CD/DVD and added a SATA CD/DVD. That may or may not be the case with your server but interesting that it's HP.

fieldofgreen, 2023-10-04:
So it does boot from the cd then? Because if that is the case I can get a cd drive in there.

PartialVolume, 2023-10-04:
USB booting sounds a bit flaky on these HP Proliant G8 servers. I was reading some forum and this seemed to work for some.

>I know it's going to sound crazy but try to boot your gen8 and wait a few seconds before inserting your usb (let's say 8 seconds).
>
>Then, when you see the boot menu, press F11 & F3 to boot from USB.
>
>I had issues with USB boot too on GEN 8.

Others said using a different brand of USB stick worked for them.

fieldofgreen, 2023-10-06:
Ok so I have tried getting a disk drive to work and it doesn't seem like it wants to try it. Not sure why at the moment so I will table that for now. 
I tried FreeDOS on a USB and it booted, So this tells me that the server can indeed boot from a USB. It just doesnt like ShredOS for some reason. At the moment I am Getting Ventoy to see if that will boot and possibly trying the ISO version of ShredOS.

fieldofgreen, 2023-10-06:
Ok good news. Ventoy boots. I tried to boot ShredOS 2021.08.2_23.5 as that was the most recent one I was trying to get working to avoid missing drivers and such. It then Dumped me into Grub and I'll be honest I've never dealt with Grub before.

fieldofgreen, 2023-10-06:
Alot of trial and error later. I found that I could in fact boot ShredOS. But only the ISO worked. I found out that Ventoy was being booted into BIOS mode for Legacy rather than UEFI, Thanks HP. So now I can go back to setting up my little station and wiping drives again. I'm using shredos-2021.08.2_23_x86-64_0.34_20221231.iso 
Im not sure exactly why the img files dont work but I will run with it for now. Now to see how many drives I can get the box to recognize at once. 
If you would like me to test something with my setup or want any specs just let me know.

PartialVolume, 2023-10-06:
>It then Dumped me into Grub

Interesting it got as far as grub but then grub didn't load the ShredOS image from the USB stick.

I would probably first confirm that the files on the USB stick are present and correct. What's the program you are using to write the shredos.img file onto the USB and does this same usb stick boot on other PCs?

PartialVolume, 2023-10-06:
>Im not sure exactly why the img files dont work but I will run with it for now. Now to see how many drives I can get the box to recognize at once.

Glad you got it working with the .iso

fieldofgreen, 2023-10-06:
> I would probably first confirm that the files on the USB stick are present and correct. What's the program you are using to write the shredos.img file onto the USB and does this same usb stick boot on other PCs?

Well I used Rufus to put my original USB together that I was using on other devices. I also used Rufus to try to boot the newest version. I only got somewhere with Ventoy 

I guess it's not a huge deal as I have set a few drives to wipe and it is much slower than when I was using my datto alto 3 v2. It's a shame since I was enjoying more USB slots

PartialVolume, 2023-10-09:
According to the specs here https://www.storagereview.com/review/hp-proliant-microserver-gen8-review

The G8 only has two V3 USB ports at the rear. The other 5 are v2 which would be a bottleneck for most newish drives and maybe even old drives too.

fieldofgreen, 2023-10-09:
:/ yeah. no worries I now have learned what I can do to get it to run on just about anything I need now.