PartialVolume/shredos.x86_64 #169: Unable to boot 0.35

Makowo, 2023-11-10:
On a HP proliant dl360 g9 in legacy off USB made with Rufus, I tried the previous two releases and got to the "booting the kernel." But it is frozen at that point with no keyboard reaction.

[image]

PartialVolume, 2023-11-10:
Do you know what the graphics adapter is on that system. If possible can you post lspci and lsusb from a working Linux distro?

In the meantime, if you add nomodeset to the Linux command line in grub.cfg. which will disable DRM graphics and run in simple frame buffer mode, does it the work?

PartialVolume, 2023-11-10:
also check the usb stick for dmesg.txt. if it exists post it here. Thanks.

PartialVolume, 2023-11-10:
You may also want to try this .. no idea if it works or not but posted 4 days ago.

How do I enable nvidia DRM Modeset?
To enable DRM (Direct Rendering Manager) kernel mode setting, set modeset=1 and fbdev=1 kernel module options for nvidia_drm module. The latter is required to tell the nvidia driver to provide its own framebuffer device instead of relying on efifb or vesafb , which don't work under simpledrm .4 days ago
https://wiki.archlinux.org › title › N...
[NVIDIA - ArchWiki](https://wiki.archlinux.org/title/NVIDIA#:~:text=To%20enable%20DRM%20(Direct%20Rendering,don't%20work%20under%20simpledrm%20.)

Makowo, 2023-11-10:
I'll give those a shot and provide the file on Monday when I'm back at work, Thank you for the quick response!

PartialVolume, 2023-11-12:
> set modeset=1 and fbdev=1

Having tried this on one of my Nvidia based systems I think using `nomodeset` in both grub.cfg files is your best bet. modeset=1 and fbdev=1 didn't work for me.

```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 nomodeset
}
```

PartialVolume, 2023-11-12:
@Makowo Are you using a graphics card with displaylink connector or the build in VGA connector?

PartialVolume, 2023-11-12:
Also see https://github.com/PartialVolume/shredos.x86_64/discussions/168#discussioncomment-7548030

Makowo, 2023-11-12:
I am using the VGA output on the back of the server, there is no PCIe devices connected.

PartialVolume, 2023-11-13:
In that case I would put `nomodeset` in both grub.cfg files. It will prevent switching to DRM graphics and use simple frame buffer mode. Also the dmesg.txt file on the USB stick will be useful assuming it exists. If it does exist we know nwipe is running even though you can't see it and proves it's a graphics issue.

PartialVolume, 2023-11-13:
In that case the kernel must be hanging before it reaches nwipe launch.

Leave nomodeset on the Linux command line but change the log level to 7 and after rebooting does the screen show more messages?

Makowo, 2023-11-13:
Indeed it does, Here's a photo of where it's staying. I'll make a video if it's needed. [image]

PartialVolume, 2023-11-13:
Yes, a video would be great. Thanks.

Makowo, 2023-11-13:
Don't have a great way of sharing this at the moment so this'll have to do for now. https://youtube.com/shorts/Xw5pMSpSMdY?si=bJLNhK7tHuijiqfV

PartialVolume, 2023-11-13:
Has your camera got a slowmotion video option. That's a bit fast.

Makowo, 2023-11-13:
I completely forget that phones have these features now https://youtu.be/2i7ZUOZgxSY?si=G2dZcF5hRf1Pd07b

PartialVolume, 2023-11-14:
Can you take the same slomo video but this time remove `nomodeset` from both grub files but leave the loglevel at 7, I should then see the Nouveau video driver load and see if it generates errors.

Makowo, 2023-11-14:
Here it is. https://youtu.be/fvPPo7-cBMw?si=72b1iuycUoy2y9fr

PartialVolume, 2023-11-14:
What's the CPU on that system?

Looks like it's switching framebuffers ok to the mgag20drmfb although the 20 second blank screen between the switch is a bit long.

Can you add  `spectre_v2=off` on the Linux command line in both grub.cfg files. Keep loglevel at 7.

Makowo, 2023-11-14:
It is a single xeon 2630v4 with 8gb DDR4 ECC [image]

Here's the video with the spectre flag https://youtube.com/shorts/Sx7n3R0gCYY?si=FLnu6MMog_xzJyBp

PartialVolume, 2023-11-14:
Can you try this version, I've added Intel-Firmware that fixes errors in the CPU. Could be a long shot but worth a try.

https://drive.google.com/file/d/1JW75XjTM0dAW2HGAb4dGgAuQundDe-w2/view?usp=sharing

Makowo, 2023-11-14:
Unfortunately not. https://youtu.be/bAP-L6rrSUU?si=YnzWg4HX0vn_FmBt

PartialVolume, 2023-11-14:
There are quite a few examples of other Linux distros hanging at those last two lines, i.e the `locksource switched to clocksource tsc` with many different ways of solving it. One of which was putting `noapic` on the Linux command line in both grub files. Resetting the bios back to factory default plus many others. If you up for trying these I can put together a list of what worked for others.

Makowo, 2023-11-14:
I would be available to test them tomorrow.

Makowo, 2023-11-15:
Noapic seems to have worked! https://youtu.be/vX9jhlJCQ5w?si=IZfXqjb8bnYQoiS3

PartialVolume, 2023-11-15:
That's excellent! Don't forget to change the loglevel back to 3 as leaving it at 7 can cause kernel messages to mess up the nwipe GUI.

Makowo, 2023-11-15:
Thank you very much for your help! I've set the log level back down but get some kernel messages in the gui thanks to hpsa but nothing closing and reopening can't fix.

PartialVolume, 2023-11-15:
No problem, glad you got it working. 👍

closed 2023-12-13 by PartialVolume