PartialVolume/shredos.x86_64 #173: Feature request: Support for Nwipe configuration file, /etc/nwipe/nwipe.conf

Aterfax, 2023-11-21:
https://github.com/martijnvanbrummelen/nwipe/releases/tag/v0.35

> Nwipe now has a configuration file, /etc/nwipe/nwipe.conf. This is currently used to store PDF parameters but will be enhanced over the next versions to include default wipe parameters amongst other additions.

As part of generating erasure certificates, some configuration options are stored in the `/etc/nwipe/nwipe.conf` file. Similarly to `nwipe_options=""`, we could do with a method for providing settings directly from the boot `APPEND` line so settings can be applied during netboots.

It may make sense for this configuration file to be supplied as a URL to CURL/wget download a configuration and use it, or simply as some form of inline.

Incidentally, the documented `lftp=""` lines can also be amended to copy these certificates back to an FTP server with the additional command `mput *.pdf`

e.g.

`lftp="open 192.168.1.250; user shredosuser password12345; cd upload; mput nwipe_*.txt; mput *.pdf"`

PartialVolume, 2023-11-21:
Thanks for the reminder re lftp, I'll make the changes for that.

I might need some clarification regarding the boot append line. Do you mean have a typical grub boot menu where you can edit the command line from the grub menu?

PartialVolume, 2023-11-22:
>Incidentally, the documented lftp="" lines can also be amended to copy these certificates back to an FTP server with the additional command mput *.pdf
>e.g.
lftp="open 192.168.1.250; user shredosuser password12345; cd upload; mput nwipe_*.txt; mput *.pdf"

Committed https://github.com/PartialVolume/shredos.x86_64/commit/a08811f60e34ca61c2309fd2ea62d169dfebab25

Aterfax, 2023-11-22:
Apologies, this wasn't clear.

The APPEND line I refer to would be in the PXE boot configs, e.g.

```
 LABEL linux-http
 LINUX http://boot-server/boot/mykernel
 APPEND initrd=http://boot-server/boot/myinitrd
 ```
 
 Or in the ipxe parlance from Netboot.xyz
 
 ```ipxe
 :shredos_options
clear menu
clear shredos_version
iseq ${os_arch} i686 && set kernel_url ${live_endpoint}/asset-mirror/releases/download/0.34_32-bit_20221231-570c24e6/shredos ||
set shredos_version 0.34_32-bit_20221231
iseq ${os_arch} x86_64 && set kernel_url ${live_endpoint}/asset-mirror/releases/download/0.34_20221231-212ce2e7/shredos ||
set shredos_version 0.3420221231
menu ShredOS ${shredos_version}
item --gap Choose a wipe method:
item dodshort ${space} Wipe all disks with the short DoD 5220.22-M method
item dod522022m ${space} Wipe all disks with the DoD 5220.22-M method
item dod3pass ${space} Wipe all disks with the DoD 5220.22-M method (3 pass)
item ops2 ${space} Wipe all disks with the RCMP TSSIT OPS-II method
item gutmann ${space} Wipe all disks with the Gutmann method
item prng ${space} Wipe all disks with the PRNG Stream method
choose nuke_method || goto shredos_exit
goto shredos_boot

:shredos_boot
imgfree
kernel ${kernel_url} console=tty3 loglevel=3 nwipe_options="--method=${nuke_method}" ${cmdline}
boot
```

I was just meaning to get at the fact we're going to need to pass this info during the netboot process as part of a command, so we're going to need another ``kernel`` line compatible argument like ``nwipe_options=`` to be passable.

PartialVolume, 2023-11-24:
Thanks for the clarification. I'll add this into the next point release 25.1 with the other changes/additions I'm currently making.