PartialVolume/shredos.x86_64 #174: ShredOS Not supporting all Intel RST RapidStore Drivers

p1r473, 2023-11-21:
Hello,
I have 2 machines that ShredOS is not supporting the HD drivers so I cant shred these drives.
I cant remember which set of drivers it is but its either f6vmdflpy-x64.zip or RST_V19.1.0.1001_PV.zip (theres multple versions of these drivers and not all versions work with these drivers)
Ive uploaded the files here
[f6vmdflpy-x64.zip](https://github.com/PartialVolume/shredos.x86_64/files/13430470/f6vmdflpy-x64.zip)
[RST_V19.1.0.1001_PV.zip](https://github.com/PartialVolume/shredos.x86_64/files/13430471/RST_V19.1.0.1001_PV.zip)

p1r473, 2023-11-21:
Darn , thank you

closed 2023-11-21 by p1r473

p1r473, 2023-12-15:
@PartialVolume so the bios is currently set to AHCI
So these are just two SSDs that can be normally wiped with shredos but can't be seen when raided
However, I believe I should still be able to see the two drive separately to wipe them separate and ignore the raid array
Any idea?

PartialVolume, 2023-12-15:
If you login to a virtual terminal in ShredOS and check if there is a /etc/mdadm/mdadm.conf ?

If you type `sudo mdadm` is the program installed?

PartialVolume, 2023-12-15:
I just checked, mdadm is not installed. Thing is I don't know whether it is required to see the Intel RST drives as normal drives or not. I wouldn't have thought so as it's normally used to configure raid which we don't want. In your bios is there a third option, AHCI, RAID and ?? or only those two?. Maybe mdadm needs to be installed and configured in a non raid state. Unfortunately I don't know enough about these drives. Maybe somebody else can offer some advice.

PartialVolume, 2023-12-15:
Just wondering but do you have something like this in your bios?
[image]

p1r473, 2024-01-27:
Nope no option in bios for changing or modifying the raid

Firminator, 2024-01-28:
What laptop model or PC motherboard do you have? BIOS or UEFI?
There *should* be an option to disable the RST and just use AHCI/non-RAID mode like PartialVolume said. Also you can invoke the RST menu seen on the screenshot by pressing CTRL+I or CTRL+R during POST if I recall correctly (it's been over a decade when I last seen an RST enabled device). It might be hidden in one of the submenus of the BIOS.