PartialVolume/shredos.x86_64 #175: Feature Request: Allow reboot option / CTRL+ALT+DEL at end of wiping process

Chewie9999, 2023-11-22:
I have just used shredos-2023.08.2_25.0_x86-64_0.35_20231110.img on an HP Proliant DL385p G8 (AMD) server.
It booted up and found the drives on both the 420i and 420 SCSI cards, and the wipe completed successfully.

At the end of the wipe process (or even when you quit without running a wipe), the only option is to restart the nwipe program.
CTRL+ALT+DEL does nothing, and there's no other obvious option to get it to reboot.
On this server, the only option is to turn it off and on again if you want it to reboot.

Chewie9999, 2023-11-22:
Yes, I could use a virtual terminal and issue a reboot/shutdown I guess (not tried yet), but this is not obvious to most users.

PartialVolume, 2023-11-22:
I'll change the current option to restart nwipe only and provide the following three options.

`R=Reboot, S=Shutdown, Spacebar=Restart Nwipe`

PartialVolume, 2024-03-06:
Implemented `Paused, press r to reboot, s to shutdown, spacebar to restart nwipe` by commit #230

closed 2024-03-06 by PartialVolume