PartialVolume/shredos.x86_64 #177: Small Cosmetic Notes

fieldofgreen, 2023-11-22:
I noticed while in the Config menu that the instructions at the bottom of the screen show that you press Space to select which just takes you back to the drive selection screen. You need to use the Enter key to select the setting you wish to change. Related to that if you enter one of the setting edit screens and then exit back to the config menu, the instructions now show the previous screens instructions. I'm not sure if it is just not updating the instructions to the proper set or if it is something else going on. 
I did also notice that alot of the menus have j and k as up and down when the arrow keys also work, is that on purpose? 

Again all of this is just things I noticed while using the latest build, I'm super happy with the program. 

One last quick question, Is there a way to save the PDF's if using Ventoy to run shredos? it seems to have an issue finding the USB to store them on.

PartialVolume, 2023-11-22:
> I noticed while in the Config menu that the instructions at the bottom of the screen show that you press Space to select which just takes you back to the drive selection screen. You need to use the [...]

Yes, I need to tidy up some of those menus in the next update as I add more options.

>I did also notice that alot of the menus have j and k as up and down when the arrow keys also work, is that on purpose?

The j &k are historical left overs from the early days, it's possible that's what DBAN used. I would imagine most people use the arrow keys. Was there ever keyboards that didn't have arrow keys? I honestly don't remember. But in theory the j & k stuff could be removed.

>One last quick question, Is there a way to save the PDF's if using Ventoy to run shredos? it seems to have an issue finding the USB to store them on.

In theory if ventoy is using disk formatted and identified exactly as FAT32 then it should work. Here's what shredOS USB would look like. Currently ShredOS will only mount a USB stick if it's identified as FAT32 as below.
```
Disk /dev/sda: 1.88 GiB, 2013265920 bytes, 3932160 sectors
/dev/sda1        1263 509075  507813  248M  c W95 FAT32 (LBA)
```

So I would suggest you boot up ShredOS using Ventoy then ALT F2 and run the following command to see what the format is for the USB Ventoy stick
```
fdisk -l 
```

Paste the output here and if it's different to FAT32 I will add it's format to the script so ShredOS will recognise the Ventoy stick.

I presume because it's not saving the PDF's it's also not saving & retrieveing the /etc/nwipe/nwipe.conf & customers files.

PartialVolume, 2023-11-22:
I just checked the ventoy disk format.
```
Disk /dev/sda: 1.88 GiB, 2013265920 bytes, 3932160 sectors
/dev/sda1  *       2048 3866623 3864576  1.8G  7 HPFS/NTFS/exFAT
/dev/sda2       3866624 3932159   65536   32M ef EFI (FAT-12/16/32)

```

I'll make the changes to ShredOS so it recognises a ventoy formatted stick.

fieldofgreen, 2023-11-22:
oh yeah I was just about to post my log but yeah thats the same output minus a different drive size.

PartialVolume, 2023-11-24:
Do you just copy the file "shredos-2023.08.2_25.0_x86-64_0.35_20231114.img" or "shredos-2023.08.2_25.0_x86-64_0.35_20231114.iso" etc to the ventoy root directory without changing it's name?

Reason I ask is that I need identify a exFAT drive from other exFAT drives that could be plugged into the system and searching for "shredos-2023.08.2_25.0_x86-64_0.35_20231114.*" in the root or whatever the booted version happens to be would be a good way of identifying the boot disc. You won't believe how difficult it is to identify which drive the OS booted from after the fact and when it's not predetermined! You would think the O.S knew that but it doesn't seem to. So I have to try to figure out what's the most likely drive that was the boot disc.

If you change it's name then that particular way of identifying the boot disc would not work.

I could of course make it a requirement that if you are using ventoy you must use the original filename, that's the version recorded in version.txt or from the download page or if buillding ShredOS yourself you match the filename details against the version in ../board/shredos/version.txt i.e. 2023.08.2_25.0_x86-64_0.35

fieldofgreen, 2023-11-24:
Yeah I just copy over the file directly. I've been wary of changing the name in case something in the file needed to reference the name.

PartialVolume, 2023-11-24:
That's good. I'll write the code with that in mind.

fieldofgreen, 2023-11-28:
So this may be my lack of knowledge in Github but is that added to the release that can be downloaded or is that just added as a change that will show up when the next release happens, I know I could compile the code myself and all that, Just curious if it already shows up in there.

fieldofgreen, 2023-11-28:
oh ok cool, that makes sense. Thanks for the update :)

PartialVolume, 2023-11-29:
> I noticed while in the Config menu that the instructions at the bottom of the screen show that you press Space to select which just takes you back to the drive selection screen. You need to use the [...]

Fixed by commit to nwipe https://github.com/martijnvanbrummelen/nwipe/commit/32321f49fa7d940f6356ff6255e4d9d4ed7b3f7b

closed 2026-02-14 by PartialVolume