PartialVolume/shredos.x86_64 #178: AutoPowerOff not working with certificate

gorbiWTF, 2023-11-23:
I've been running the new version with the PDF certificates and so far I like it a lot, but I noticed a couple of small issues or rather things that could be improved upon:

1. On the first page of the certificate it says "Page 1 - Erasure Status", on the second "Smart Data - Page 2". I think the order should be the same, i.e. "Page 2 - Smart Data" and "Page 3 - Smart Data".

3. I have `--autopoweroff` in my grub.cfg, however it waited for me to press a button before it shutdown. I had the screen off, so pressing B made it shutdown. I don't know why it waited, but I assume because of the certificate creation / saving to USB?

5. I think drives don't get powered down after finishing their wipe process. All my drives were pretty warm, despite finishing at different times (2 hours vs 5 hours vs 12 hours). I think spinning or powering every drive down as soon as it is finished wiping would be a good idea.

Thanks everyone involved in this project!

PartialVolume, 2023-11-23:
>On the first page of the certificate it says "Page 1 - Erasure Status", on the second "Smart Data - Page 2". I think the order should be the same, i.e. "Page 2 - Smart Data" and "Page 3 - Smart Data".

I'll make that change.

>I have --autopoweroff in my grub.cfg, however it waited for me to press a button before it shutdown. I had the screen off, so pressing B made it shutdown. I don't know why it waited, but I assume because of the certificate creation / saving to USB?

Thanks for reporting that, I'll look into that and let you know.

>I think drives don't get powered down after finishing their wipe process. All my drives were pretty warm, despite finishing at different times (2 hours vs 5 hours vs 12 hours). I think spinning or powering every drive down as soon as it is finished wiping would be a good idea.

No problem, I can issue a low level standby command to the drive which parks the heads and shuts down the platter motor as soon as a given drive's wipe thread completes. Accessing the drive by restarting nwipe powers it back up again. Just tested this on a Seagate SATA drive. No idea whether this works on SAS but will check it out.

PartialVolume, 2023-11-23:
@gorbiWTF Can you double check both your grub.cfg files. At the moment I can't see why anything should have changed with autopoweroff, even with the PDF certificates, should fail with a message to the console only if it can't access any USB stick, there is no waiting for user input in the event of failure.
```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 nwipe_options="--autopoweroff"
}
```

PartialVolume, 2023-11-24:
@gorbiWTF Forget checking the grub.cfg file. I can confirm that there is a situation where autopoweroff does not work. Thanks for such a detailed description of what you were doing. That would have been impossible to find without you mentioning that you blanked the screen.

It turns out --autopoweroff, which is converted to a --nowait by ShredOS, does indeed wait once the drives have completed wiping but only if you blanked the screen from blue/white to grey to blank. If you press b to unblank then that's when nwipe exits.

I'll take a look at the code & fix that

PartialVolume, 2023-11-24:
--autopoweroff issue Fixed by https://github.com/martijnvanbrummelen/nwipe/pull/529/commits/7f39d81548309faaed195ced07222fc42315181f

PartialVolume, 2023-11-28:
>On the first page of the certificate it says "Page 1 - Erasure Status", on the second "Smart Data - Page 2". I think the order should be the same, i.e. "Page 2 - Smart Data" and "Page 3 - Smart Data".

Fixed by commit in nwipe https://github.com/martijnvanbrummelen/nwipe/commit/789aa134a0281f073fb92fee52648dc0d453760f

PartialVolume, 2023-12-13:
I'm closing this as complete.

With reference to item 3 in your first comment.
>I think drives don't get powered down after finishing their wipe process. All my drives were pretty warm, despite finishing at different times (2 hours vs 5 hours vs 12 hours). I think spinning or powering every drive down as soon as it is finished wiping would be a good idea.

I will open that as a specific issue.

closed 2023-12-13 by PartialVolume