PartialVolume/shredos.x86_64 #182: Network boot: persist PDF options

cyberm8, 2023-11-30:
I really like the idea of a PDF report.

I understand that the `/etc/nwipe/nwipe.conf` and `/etc/nwipe/nwipe_customers.csv` files are persisted when a USB key is used.

However, I boot ShredOS via the network. As a result, I have to systematically redefine the PDF fields (customers, organisation, technician name). In fact, I always enter the same thing.

Is it possible to define a default value, which can be changed as required (typically the technician's name is likely to change)?

If this isn't already possible, is it something that could be considered?

PartialVolume, 2023-11-30:
I'm currently working on how settings are saved when network booting.

The three possible enhancements are:

1. Write config files to a FTP server. This feature already exists within ShredOS, I would just need to add the code that reads and writes the nwipe config and customers file.
2. An option similar to option 1 but using http. The URL would be specified as a nwipe option.
3. An option that reads/writes any USB stick it finds looking for the config and saving the config. You could then use any exact/FAT32 formatted USB as your config store, it wouldn't need to be the device ShredOS booted from.

Any comments on pros & cons or other options you feel are better are most welcome.

PartialVolume, 2023-11-30:
Also see #173

cyberm8, 2023-11-30:
I think option 1 is preferable.

An FTP server is probably already set up as the PDF reports have to be sent somewhere. What about using the TFTP set up for the PXE boot?

Option 2 is acceptable, but maybe it's not a good idea to force the deployment of a web server just for a few text files?

Option 3 seems the least practical. It would use up a USB stick, and you'd have to juggle with it if you had to wipe several machines at once (or have several USB sticks). Not to mention the likelihood of purging the key by mistake.

PartialVolume, 2023-12-01:
>Help regarding PXE config needed!
This is where I need some help with the PXE configuration, I can't figure out how you pass command line options when booting pxe using EFI. I'm editing pxelinux.cfg and appending the above option to the APPEND line but when I boot ShredOS and type more /proc/cmdline I just see the default values, but not the extra option I added. Do I need to run some command after editing pxeconfig.cfg before the values are recognised?

Forget that help, I needed a nice strong cup of coffee and then realised where I was going wrong.

So if the following changes to the PXE server configuration sound reasonable, then I'll go ahead and make the additions to ShredOS's scripts.

i.e changes to the PXE server
```
mkdir  /srv/tftp/nwipe_data
chown tftp:tftp /srv/nwipe_data
chmod 774 /srv/nwipe_data

then
vi /etc/init.d/tftpd-hpa 
and add --create to the start command

finally
Add "tftp_server=192.168.1.241" to the kernel command line in the ShredOS grub files etc
```

Assuming /srv/nwipe_data/nwipe.conf & /srv/nwipe_data/nwipe_customers.csv exist they will be read on nwipe start. On nwipe exit, /srv/nwipe_data/nwipe.conf & /srv/nwipe_data/nwipe_customers.csv will be written back to the tftp server. In addition the dmesg.txt nwipe_logs and PDF certificates will also be written to /srv/nwipe_data/.

PartialVolume, 2023-12-01:
@Aterfax Would the ability to restore and save of nwipe.conf & nwipe_customers.csv and save log, pdfs etc to the pxe tftp server work in your PXE setup work for you? See previous comment for the changes you would need to make.

cyberm8, 2023-12-01:
_It makes me think about posting my Ansible playbook to setup a PXE server for ShredOS..._

What you're proposing sounds interesting.
You just need to create a "nwipe_data" subfolder and do the appropriate chown/chmod.

Sending PDFs to this location via TFTP would make SFTP unnecessary. As I send myself the latest reports and logs by e-mail before deleting them, it wouldn't particularly affect me a priori.

PartialVolume, 2023-12-02:
>Ultimately I think the important considerations will be ensuring whatever methods implemented will allow the person erasing to ensure that their correct erasure details (nwipe.conf) are or can be selected and the output can be copied to storage of their choice, be it TFTP / (S)FTP.

Agreed, a command line option, something like config_location=tftp, ftp, http, usb. If the user doesn't provide this option then ShredOS will make the most logical decision based on what it determines was the boot method or what other options might be on the command line.

The config_location will take precedence so if you boot by USB but want your config and files saved to FTP then config_location would allow that to happen. I Might have to have two options to separate pdfs and logs from configs. So in addition a  results_location=. So configs would be read/written to/from USB as you are booting from USB and certificates/logs go to FTP or tftp or http etc.

kristianronningen, 2024-02-25:
I'm involved in a project to reinstall old laptops donated to us, and hand them off to people in need. Part of this effort is to wipe/shred the disks of those laptops and provide reports back to donors. We use pxe/netboot for this, to make it easier to handle a large amount of laptops. We would very much like to see a way to pre-populate the customer information on the netboot server, and then just simply select the appropriate ShredOS boot option for each donor during PXE boot of each laptop. Having the donor name in the log/PDF report makes these things easier to track. (We're also looking at adding a custom incrementing serial number for each laptop, but I suspect this may be too specific to make sense in a netboot environment.)

The repository containing the ansible playbook/roles we use to set up our netboot host is publicly available at https://git.nuug.no/nuug-public/skolelinux-netboot if anyone is interested.

closed 2024-07-16 by cyberm8