PartialVolume/shredos.x86_64 #183: Inability to boot with latest version on dual Xeon L5640 server

wolfbeast, 2023-12-10:
I tried using ShredOS v2023.08.2_25.0_x86-64_0.35 on an HP ProLiant DL380 Gen 6 with 2x Xeon L5640 and 60GB RAM.
Unfortunately when it was supposed to boot and load the wiper application, the screen just went black and the system hung (hard).

I then put an older version on the same boot medium (v2021.08.2_21_x86-64_0.32.023) and that booted up just fine.

Due to time constraints I have not tried any other versions at the moment.

PartialVolume, 2023-12-10:
I would edit the two grub files on the USB stick and add noapic and/or nomodeset. Try each in turn.

PartialVolume, 2023-12-10:
The older (v2021.08.2_21_x86-64_0.32.023) release does not include drm drivers so adding nomodeset to the kernel command line in both grub.cfg files will disable DRM and use frame buffer mode.