PartialVolume/shredos.x86_64 #184: Does building ShredOS also generate a .iso image? If not, then how?

RDGR8, 2023-12-12:
I want to be able to use the latest ShredOS version on my Ventoy USB, though the .img image didn't work, and the latest iso file I can find is quite old, so I am planning to build ShredOS. Building takes a lot of resources so I want to make sure that an iso file would be generated in the building process, and if it won't be, then how could I build one? and how much time would it approximately take? (Ryzen 5 3600, 16gb ram 3000Mhz)

PartialVolume, 2023-12-12:
When you say it didn't work and you explain that in more detail.

I'm running the latest .IMG on Ventoy and it's working on my system with nomodeset appended to the kernel command line in both grub.cfg files. I'm running on a Dell desktop with Nvidia graphics.

PartialVolume, 2023-12-12:
>Building takes a lot of resources so I want to make sure that an iso file would be generated in the building process, and if it won't be, then how could I build one? 

A .iso won't be build without making a few changes to the configuration. I'll build a .iso and in the process document the changes that need to be made. However putting an .iso on there instead, I don't think will solve your problem. 

What's the hardware you are running on? Especially the graphics card?

RDGR8, 2023-12-12:
> A .iso won't be build without making a few changes to the configuration. I'll build a .iso and in the process document the changes that need to be made. However putting an .iso on there instead, I [...]

I am using an NVIDIA gpu

PartialVolume, 2023-12-12:
Appending 'nomodeset' to the end of the kernel command line may fix the problem. I'm assuming you get the percentage status reaching 100% then a blank screen?

To edit the .IMG, you will need to mount it as a drive, make the edit and recreate the .IMG

I'll see if I can find a procedure for that.

PartialVolume, 2023-12-12:
@RDGR8 Here's the procedure for editing the two grub.cfg files inside the Shredos .img file then re creating a new modified .img file that you can copy to the root of the Ventoy USB stick. It will be far easier to do this than build ShredOS from source.

https://github.com/PartialVolume/shredos.x86_64/blob/master/README.md#how-to-edit-the-shredos-efibootgrubcfg-and-bootgrubcfg-files-when-using-ventoy-with-shredos-img-files

PartialVolume, 2023-12-12:
>Anytime I would run the .img file it would just give me a command line in grub

If you are using the vanilla  shredos-2023.08.2_25.0_x86-64_0.35_20231110.img, once copied to the Ventoy stick I would mount the ventoy stick and run sha1sum on shredos-2023.08.2_25.0_x86-64_0.35_20231110.img to determine it's checksum matches. It could be corrupt.

```
> sha1sum shredos-2023.08.2_25.0_x86-64_0.35_20231110.img 
c0e7c6a1260979280543b95d6faca6addf7195ac  shredos-2023.08.2_25.0_x86-64_0.35_20231110.img
```

PartialVolume, 2023-12-13:
Video of re-packed ShredOS .img with modified grub.cfg files inside the .img, booting in Ventoy. Using  [this](https://github.com/PartialVolume/shredos.x86_64/blob/master/README.md#how-to-edit-the-shredos-efibootgrubcfg-and-bootgrubcfg-files-when-using-ventoy-with-shredos-img-files) procedure for re-packing the ShredOS .img file.

[Video](https://github.com/PartialVolume/shredos.x86_64/assets/22084881/ab7277ec-e5f7-4d25-bf39-2cf0a4e7aa72)

RDGR8, 2023-12-13:
> Video of re-packed ShredOS .img with modified grub.cfg files inside the .img, booting in Ventoy. Using [this](https://github.com/PartialVolume/shredos.x86_64/blob/master/README.md#how-to-edit-the-shredos-efibootgrubcfg-and-bootgrubcfg-files-when-using-ventoy-with-shredos-img-files) procedure for re-packing the ShredOS .img file. Using Ventoy v1.0.96
> 
>  VID_20231212234131_Ventoy_with_ShredOS.2.mp4

So, an update:
The sha1sum is identical
After editing the shredos I am getting the same error, though it differ if I am booting Ventoy in UEFI or non-UEFI (just like before)

UEFI: 
```
shrim_lock protocol not found
...
failed to boot both default and fallback entries
```

non-UEFI:

Usually when I open ShredOS at the top right corner it shows me the progress, here the progress is only shows for 1 frame, then it gives me a grub command line

PartialVolume, 2023-12-13:
Is secure boot enabled in the bios for UEFI?  What's the make/model of the computer?

RDGR8, 2023-12-13:
> Is secure boot enabled in the bios for UEFI? What's the make/model of the computer? Sorry just saw your comment above.
> 
> Will add that as a third solution. Thanks.

So, my question wasn't answered, but the core problem was solved, so should I close this issue?

PartialVolume, 2023-12-13:
No, leave it open for the time being as I do need to document in the README.md how you modify the buildroot configuration to create .iso images for CD/DVD.

PartialVolume, 2023-12-13:
>Opening BIOS settings and then going to Secure Boot and changing OS type from Windows to Other OS fixed the issue, now booting into the original ShredOS works. Please add this as issue No.3 workaround.

Workaround No.3 for Ventoy added to [release notes](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2023.08.2_25.0_x86-64_0.35).

RDGR8, 2023-12-14:
> Workaround No.3 for Ventoy added to [release notes](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2023.08.2_25.0_x86-64_0.35).

> If the boot fails before grub loads the kernel, i.e you don't see the kernel loading to 100% then look for a bios related issue.
 
 Actually no, I can see the ShredOS loading to 100% from the top right corner, after that I get the shim error

PartialVolume, 2023-12-14:
Thanks for letting me know, I'll correct that statement.

PartialVolume, 2023-12-14:
@RDGR8 I made workaround no.3 more succinct. [Release Note](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2023.08.2_25.0_x86-64_0.35)

RDGR8, 2023-12-15:
> @RDGR8 I made workaround no.3 more succinct. [Release Note](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2023.08.2_25.0_x86-64_0.35)

Thanks, I am really happy to participate in such a project, even if it was a small workaround.Thanks for being responsive and I hope to see this project grow up.

korpx, 2024-01-04:
https://github.com/PartialVolume/shredos.x86_64/issues/89#issuecomment-1876659864