PartialVolume/shredos.x86_64 #186: Set IP address on ShredOS

flhoest, 2023-12-14:
I need to setup an IP address on my SHredOS box.
I booted the SHredOS IMG remotely using IPMI to completely shred a 7 HDDs server before shipping. I have the reports on the filesystem and I need to scp them to a remote location for reference.
I used the 2nd terminal (ALT+F2) and I'm abe to type some commands. I've tried to add an IP using ip address add command, but I can't seem to have it working.
Anyone if facing similar issues ?
Thanks.

PartialVolume, 2023-12-14:
I assume you mean it's saved to the non persistent filesystem that held in memory. If so what ever you do don't switch it off or reboot ShredOS other wise you will lose any logs/pdfs.

If you just want to get the logs/pdfs off I would plug a USB stick in and get them off that way by control C on nwipe which will save to the first FAT32 formatted USB  stick, preferably a ShredOS stick.You can then start setting static IP addresses.

I've never tested with scp or booted via IPMI? So can't help much.

PartialVolume, 2023-12-14:
What version of ShredOS?

flhoest, 2023-12-14:
Yes you are right, I'm in the process of wiping a fes disks and it runs for another 40 hours ... so in // I'm trying access the file system from the outside world to expatriate the files in a safe place. I'm running the latest version I guess ? This is v2023.08.2_25.0_x86-64_0.35

PartialVolume, 2023-12-14:
Isn't the dynamic IP any good for you?

flhoest, 2023-12-14:
That's the issue, we do not use DHCP in our datacenter. We have fixed IP assigned on demand and open in the firewall.

PartialVolume, 2023-12-14:
I'm on the road today so may not be able to help much. So how are you accessing the system at the moment, can you access virtual terminals, are you telneted in?

flhoest, 2023-12-14:
I'm using the remote console in HTML5 provided by the Supermicro BMC. I have mounted the ShredOS IMG on a virtual drive. I'm in a virtual terminal using CLI trying to set an IP on the machine, I have 4 interfaces and trying to find the one with the link and correct VLAN ...

PartialVolume, 2023-12-14:
Ok, so are you running these sort of commands?
```
auto eth0
allow-hotplug eth0
iface eth0 inet static
address 192.168.0.1
netmask 255.255.255.0
```

flhoest, 2023-12-14:
No, I'm playing with ip addr add dev eth1 ..... up link show, etc ...

flhoest, 2023-12-14:
It looks like I need to setup a VLAN ... that's my issue ;)

flhoest, 2023-12-14:
vconfig add says package not installed ...

PartialVolume, 2023-12-14:
Have you got or can setup a FTP server on your network?big so you could use lftp to send.

flhoest, 2023-12-14:
@PartialVolume message crossing ;)

closed 2023-12-14 by flhoest

PartialVolume, 2023-12-14:
>Having a network configuration UI in the menu would be awesome ;)

I can add that option.