PartialVolume/shredos.x86_64 #191: HP DL360 G10 Not detecting drives.

Mathew021, 2023-12-19:
Hi there! Ive had some issues with the latest version of shredos detecting the drives in my HP server, they are SAS drives... I've tried setting each drive to raid 0, or deleting the raid configuration entirely to ensure its in a HBA type mode, the controller is a Smart Array P408i-a which when theres no raid config, it should go into a HBA mode when theres no config as far as i know. Is there anything i need to do with Shredos to get it to detect my drives? drivers? Bios settings to look out for?  Thanks!

PartialVolume, 2023-12-19:
Hi, for what it's worth both HP raid controllers hpsa and smartpqi are installed in ShredOS.

>Ive had some issues with the latest version of shredos detecting the drives in my HP server, they are SAS drives
Just to clarify, are you saying it worked with a different version of ShredOS or SATA drives?

I don't know if this is true but I came across a comment that you can't put Smart Array P408i into HBA mode. Is that true for these controllers?

https://groups.google.com/g/mpc.lists.freebsd.hardware/c/LiLFIY2O7X4/m/LmbS4feCAAAJ
```
I'm not sure what your expectation was but the Smart Array hardware RAID
controllers I've seen from HP are the same as IBM / Dell in the sense
that the controller presents a logical disk to the host for all RAID
configurations, RAID 0 or otherwise. Host access to a disk where the
controller acts as just a HBA isn't possible. It's a purposeful design
difference between a RAID controller and a HBA.
```
```
CONFIG_SCSI_HPSA:                                                                                                                                     │  
  │                                                                                                                                                       │  
  │ This driver supports HP Smart Array Controllers (circa 2009).                                                                                         │  
  │ It is a SCSI alternative to the cciss driver, which is a block                                                                                        │  
  │ driver.  Anyone wishing to use HP Smart Array controllers who                                                                                         │  
  │ would prefer the devices be presented to linux as SCSI devices,                                                                                       │  
  │ rather than as generic block devices should say Y here.                                                                                               │  
  │                                                                                                                                                       │  
  │ Symbol: SCSI_HPSA [=y]                                                                                                                                │  
  │ Type  : tristate                                                                                                                                      │  
  │ Defined at drivers/scsi/Kconfig:348                                                                                                                   │  
  │   Prompt: HP Smart Array SCSI driver                                                                                                                  │  
  │   Depends on: SCSI_LOWLEVEL [=y] && PCI [=y] && SCSI [=y]                                                                                             │  
  │   Location:                                                                                                                                           │  
  │     -> Device Drivers                                                                                                                                 │  
  │       -> SCSI device support
```
```
CONFIG_SCSI_SMARTPQI:                                                                                                                                 │  
  │                                                                                                                                                       │  
  │ This driver supports Microchip PQI controllers.                                                                                                       │  
  │                                                                                                                                                       │  
  │ <http://www.microchip.com>                                                                                                                            │  
  │                                                                                                                                                       │  
  │ To compile this driver as a module, choose M here: the                                                                                                │  
  │ module will be called smartpqi.                                                                                                                       │  
  │                                                                                                                                                       │  
  │ Note: the aacraid driver will not manage a smartpqi                                                                                                   │  
  │       controller. You need to enable smartpqi for smartpqi                                                                                            │  
  │       controllers. For more information, please see                                                                                                   │  
  │       Documentation/scsi/smartpqi.rst                                                                                                                 │  
[... cut ...]
  │       -> SCSI device support                                                                                                                          │  
  │         -> SCSI low-level drivers (SCSI_LOWLEVEL [=y])                                                                                                │  
  │           -> Microchip PQI Driver (SCSI_SMARTPQI [=y])                                                                                                │  
  │ Selects: SCSI_SAS_ATTRS [=y] && RAID_ATTRS [=y]
```

Mathew021, 2023-12-19:
Hey, answer to your first question. No I haven't had this work with SATA drives or shredos, but I have experience using dban. I tried using that and it didn't work so I was hoping shredos. might. 

The second question you had regarding how I set the raid up. I did this through the raid controller from system utilities during startup. I tried both setting raid 0 on each drive individually and also clearing the configuration entirely. After each attempt, the drives wouldn't show up in Shredos

PartialVolume, 2023-12-19:
This is the spec sheet, it looks like it should operate in HBA or mixed mode.
[HPE Smart Array SR Gen10 Controllers for HPE ProLiant DL, ML and Apollo Servers-a00047736enw.pdf](https://github.com/PartialVolume/shredos.x86_64/files/13715699/HPE.Smart.Array.SR.Gen10.Controllers.for.HPE.ProLiant.DL.ML.and.Apollo.Servers-a00047736enw.pdf)

PartialVolume, 2023-12-19:
> I tried both setting raid 0 on each drive individually and also clearing the configuration entirely. After each attempt, the drives wouldn't show up in Shredos

I'm not sure raid 0 is what you want, as doesn't that just strip the data across all the drives with no redundant drive. Wasn't there any other options? It looks like you would need to configure HBA for each drive.

PartialVolume, 2023-12-22:
If you get a chance can you post the output of `lspci`. I want have a more accurate record of problematic hardware. Thanks.

Firminator, 2024-01-18:
The PDF linked by PartialVolume states that these controller do support Mixed Mode, so they can be set to HBA/JBOD mode:
[image]

> which when theres no raid config, it should go into a HBA mode when theres no config as far as i know.

I think that's the issue. Apparently - reading between the lines here - it's still set to RAID as you assume it goes to HBA mode automatically. It doesn't or shouldn't from my experience. There should be a setting in the SmartArray BIOS/UEFI Config toll for the controller where you explicitely set it to from RAID to HBA/JBOD mode. Then the drives should be detected by ShredOS.

There is also the HP SSA (Smart Storage Admin Tool) which supports setting it to HBA mode:
[image]

ExaneServerTeam, 2024-02-23:
Hello,

My two cents on this.
Removing the Raid0 does not set the Controller in HBAMode.

Sample usage of `ssacli` to set HPE P Type controler in `hbamode` on my end
```
ssacli ctrl slot=0 modify hbamode=on
``` 

Algo you give a look to `hpe-kill-raid.py` in the following archive. That's how I do it so far (Going to Console 2, downloading the thing with `wget`, exec the python script).
[hpe-cleandisk.tar.gz](https://github.com/PartialVolume/shredos.x86_64/files/14384318/hpe-cleandisk.tar.gz)

flhoest, 2024-06-17:
Hi there, is there any news around this since then ?

jansiee, 2024-08-12:
I have the same issue. Online is still read that the HP Smart array P408i should automatically do HBA mode when no array is created. Because we can also see the disks in ESXI when nothing is created, but SredOS cant see them.

We are doing this on HP ML380 GEN10

Source: https://community.hpe.com/t5/proliant-servers-ml-dl-sl/p408i-a-sr-gen10-configure-hba-mode/td-p/7113580