PartialVolume/shredos.x86_64 #193: Telnet not enabling after editing grub.cfg

mdomanic, 2023-12-20:
After editing both grub.cfg files to add telnetd=enable (using the method within linux of mounting the original shredos.img and editing the grub cfgs then rewriting the image, and using VENTOY), telnet port is not open after successfully booting into shredos.

I have tried this on multiple motherboards, and multiple networks (making sure telnet/port 23 is not being blocked) and I still don't see port 23 open on any of these instances.

Using latest build (v2023.08.2_25.0_x86-64_0.35)

PartialVolume, 2023-12-20:
Using the Ventoy flash drive you created can you boot ShredOS on a system with keyboard & monitor, then ALT F2 for the virtual terminal and type `more /proc/cmdline` and post the output.

Then with a RJ45 network plugged in, can you type 'ifconfig' and post the output. Thanks.

mdomanic, 2023-12-20:
BOOT_IMAGE=/boot/shredos console=tty3 loglevel=3 telnetd=enable nomodeset noapic

eth0      Link encap:Ethernet  HWaddr B0:6E:BF:2A:FA:50  
          inet addr:10.22.22.6  Bcast:10.22.22.255  Mask:255.255.255.0
          inet6 addr: fe80::b26e:bfff:fe2a:fa50/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3561 errors:0 dropped:12 overruns:0 frame:0
          TX packets:2324 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:231768 (226.3 KiB)  TX bytes:151068 (147.5 KiB)
          Interrupt:11 Memory:df100000-df120000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:64 errors:0 dropped:0 overruns:0 frame:0
          TX packets:64 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:4736 (4.6 KiB)  TX bytes:4736 (4.6 KiB)

PartialVolume, 2023-12-20:
Thanks, it looks like the telnetd package in busybox some how got deselected in the last update as the S50telnet file is missing from /etc/init.d/ 

 I'll do another update as soon as I can, I'm currently building a new version which I'll publish as a pre-release. If it all goes ok that may be in the next few days, if not it won't be until after Christmas.

In the meantime you can always go back to the previous release if you don't need the PDF support.

mdomanic, 2023-12-20:
No problem! I will roll back for now, thanks!

PartialVolume, 2024-03-02:
Re-introduced telnetd as part of commit  https://github.com/PartialVolume/shredos.x86_64/commit/c9346c1f16ebd429ba738accce4cba2b88d901e0

closed 2026-02-14 by PartialVolume