PartialVolume/shredos.x86_64 #194: RS720-E10: System completely hangs after booting ShredOS and nwipe gui is loaded

mdomanic, 2023-12-21:
I am attempting to wipe a RS720-E10 system, can get ShredOS to boot just fine and nwipe gui starts, but the whole system hangs immediately after.  Not sure yet if this is a specific issue with RS720 servers as I have not seen any previous issues brought up with this system. Not able to switch to virtual terminal (thus proving the system being hung/halted for unknown reason)

The good news is that nwipe sees the entire raid 👍 

Not seeing a dmesg output being left on USB drive (I am assuming that the system is halting before this process gets completed)

Also, have tried this on latest release both stock and with noapic flag

I will edit this if I can get specific debug output to help determine root cause of the issue

PartialVolume, 2023-12-21:
You could try changing the log level to 7, start the wipe and see if you get any clues from the kernel. The nwipe display will get messed up due to the kernel messages but maybe it will provide a clue as to why it hangs.

mdomanic, 2023-12-22:
This log/error is being spammed by system

[image]

PartialVolume, 2023-12-22:
Assuming I've understood that correctly that looks like the i40e (ethernet driver having requested a interrupt but not shared it and megasas0 requesting the same interrupt but not getting it. Does that happen right from boot up or does it happen after a wipe started?

mdomanic, 2023-12-22:
This starts as soon as the system has booted into ShredOS

PartialVolume, 2023-12-22:
I don't think that's the cause of the hang, if you leave that running then ALT F2 and in the virtual terminal, type nwipe and start a wipe. Then once it's started switch back to ALT F1 where the kernel is printing all that stuff then wait. If the system hangs I'm hoping it will print something useful just before it hangs.

PartialVolume, 2023-12-22:
Can you post the output of lspci for this system. Thanks.

mdomanic, 2023-12-22:
here is the lspci output

0000:00:00.0 System peripheral: Intel Corporation Device 09a2 (rev 04)
0000:00:00.1 System peripheral: Intel Corporation Device 09a4 (rev 04)
0000:00:00.2 System peripheral: Intel Corporation Device 09a3 (rev 04)
0000:00:00.4 Host bridge: Intel Corporation Device 0998
0000:00:01.0 System peripheral: Intel Corporation Device 0b00
0000:00:01.1 System peripheral: Intel Corporation Device 0b00
0000:00:01.2 System peripheral: Intel Corporation Device 0b00
0000:00:01.3 System peripheral: Intel Corporation Device 0b00
0000:00:01.4 System peripheral: Intel Corporation Device 0b00
0000:00:01.5 System peripheral: Intel Corporation Device 0b00
0000:00:01.6 System peripheral: Intel Corporation Device 0b00
0000:00:01.7 System peripheral: Intel Corporation Device 0b00
0000:00:02.0 System peripheral: Intel Corporation Device 09a6
0000:00:02.1 System peripheral: Intel Corporation Device 09a7
0000:00:02.4 Non-Essential Instrumentation [1300]: Intel Corporation Device 3456 (rev 01)
0000:00:11.0 Unassigned class [ff00]: Intel Corporation C620 Series Chipset Family MROM 0 (rev 0a)
0000:00:11.5 SATA controller: Intel Corporation C620 Series Chipset Family SSATA Controller [AHCI mode] (rev 0a)
0000:00:14.0 USB controller: Intel Corporation C620 Series Chipset Family USB 3.0 xHCI Controller (rev 0a)
0000:00:14.2 Signal processing controller: Intel Corporation C620 Series Chipset Family Thermal Subsystem (rev 0a)
0000:00:16.0 Communication controller: Intel Corporation C620 Series Chipset Family MEI Controller #1 (rev 0a)
0000:00:16.1 Communication controller: Intel Corporation C620 Series Chipset Family MEI Controller #2 (rev 0a)
0000:00:16.4 Communication controller: Intel Corporation C620 Series Chipset Family MEI Controller #3 (rev 0a)
0000:00:17.0 SATA controller: Intel Corporation C620 Series Chipset Family SATA Controller [AHCI mode] (rev 0a)
0000:00:1c.0 PCI bridge: Intel Corporation C620 Series Chipset Family PCI Express Root Port #1 (rev fa)
0000:00:1c.4 PCI bridge: Intel Corporation C620 Series Chipset Family PCI Express Root Port #5 (rev fa)
0000:00:1c.5 PCI bridge: Intel Corporation C620 Series Chipset Family PCI Express Root Port #6 (rev fa)
0000:00:1f.0 ISA bridge: Intel Corporation Device a1cb (rev 0a)
0000:00:1f.2 Memory controller: Intel Corporation C620 Series Chipset Family Power Management Controller (rev 0a)
0000:00:1f.4 SMBus: Intel Corporation C620 Series Chipset Family SMBus (rev 0a)
0000:00:1f.5 Serial bus controller [0c80]: Intel Corporation C620 Series Chipset Family SPI Controller (rev 0a)
0000:03:00.0 PCI bridge: ASPEED Technology, Inc. AST1150 PCI-to-PCI Bridge (rev 06)
0000:04:00.0 VGA compatible controller: ASPEED Technology, Inc. ASPEED Graphics Family (rev 52)
0000:16:00.0 System peripheral: Intel Corporation Device 09a2 (rev 04)
0000:16:00.1 System peripheral: Intel Corporation Device 09a4 (rev 04)
0000:16:00.2 System peripheral: Intel Corporation Device 09a3 (rev 04)
0000:16:00.4 Host bridge: Intel Corporation Device 0998
0000:16:02.0 PCI bridge: Intel Corporation Device 347a (rev 04)
0000:16:04.0 PCI bridge: Intel Corporation Device 347c (rev 04)
0000:17:00.0 Ethernet controller: Intel Corporation Device 15ff (rev 02)
0000:17:00.1 Ethernet controller: Intel Corporation Device 15ff (rev 02)
0000:18:00.0 RAID bus controller: LSI Logic / Symbios Logic Device 10e2
0000:30:00.0 System peripheral: Intel Corporation Device 09a2 (rev 04)
0000:30:00.1 System peripheral: Intel Corporation Device 09a4 (rev 04)
0000:30:00.2 System peripheral: Intel Corporation Device 09a3 (rev 04)
0000:30:00.4 Host bridge: Intel Corporation Device 0998
0000:4a:00.0 System peripheral: Intel Corporation Device 09a2 (rev 04)
0000:4a:00.1 System peripheral: Intel Corporation Device 09a4 (rev 04)
0000:4a:00.2 System peripheral: Intel Corporation Device 09a3 (rev 04)
0000:4a:00.4 Host bridge: Intel Corporation Device 0998
0000:64:00.0 System peripheral: Intel Corporation Device 09a2 (rev 04)
0000:64:00.1 System peripheral: Intel Corporation Device 09a4 (rev 04)
0000:64:00.2 System peripheral: Intel Corporation Device 09a3 (rev 04)
0000:64:00.4 Host bridge: Intel Corporation Device 0998
0000:64:00.5 RAID bus controller: Intel Corporation Device 28c0 (rev 04)
0000:64:02.0 System peripheral: Intel Corporation Device 09ab
0000:64:03.0 System peripheral: Intel Corporation Device 09ab
0000:64:04.0 System peripheral: Intel Corporation Device 09ab
0000:64:05.0 System peripheral: Intel Corporation Device 09ab
0000:7e:00.0 System peripheral: Intel Corporation Device 3450
0000:7e:00.1 System peripheral: Intel Corporation Device 3451
0000:7e:00.2 System peripheral: Intel Corporation Device 3452
0000:7e:00.3 Host bridge: Intel Corporation Device 0998
0000:7e:00.5 System peripheral: Intel Corporation Device 3455
0000:7e:02.0 System peripheral: Intel Corporation Device 3440
0000:7e:02.1 System peripheral: Intel Corporation Device 3441
0000:7e:02.2 System peripheral: Intel Corporation Device 3442
0000:7e:03.0 System peripheral: Intel Corporation Device 3440
0000:7e:03.1 System peripheral: Intel Corporation Device 3441
[... 266 lines cut ...]
0000:ff:1d.0 System peripheral: Intel Corporation Device 344f
0000:ff:1d.1 System peripheral: Intel Corporation Device 3457
0000:ff:1e.0 System peripheral: Intel Corporation Device 3458 (rev 06)
0000:ff:1e.1 System peripheral: Intel Corporation Device 3459 (rev 06)
0000:ff:1e.2 System peripheral: Intel Corporation Device 345a (rev 06)
[... cut ...]
10001:02:02.0 PCI bridge: PMC-Sierra Inc. Device 4068
10001:02:03.0 PCI bridge: PMC-Sierra Inc. Device 4068
10001:02:04.0 PCI bridge: PMC-Sierra Inc. Device 4068
10001:02:05.0 PCI bridge: PMC-Sierra Inc. Device 4068
10001:02:06.0 PCI bridge: PMC-Sierra Inc. Device 4068
10001:02:07.0 PCI bridge: PMC-Sierra Inc. Device 4068
10001:02:08.0 PCI bridge: PMC-Sierra Inc. Device 4068
10001:02:09.0 PCI bridge: PMC-Sierra Inc. Device 4068
10001:02:0a.0 PCI bridge: PMC-Sierra Inc. Device 4068
10001:02:0b.0 PCI bridge: PMC-Sierra Inc. Device 4068

PartialVolume, 2023-12-23:
That is a very busy lspci output, I don't think I've ever come across so many repeated devices.

Can you post the output of `lspci -k` which will show me what drivers have been loaded. Certainly has some chips I don't see very often, like the `VGA compatible controller: ASPEED Technology, Inc. ASPEED Graphics Family (rev 52)`.

As this is a niche graphics controller, can you try one thing for me. Assuming you can't get through a single wipe without it hanging?.. can you boot shredos but run it in non gui mode with autonuke? This may not be something you want to do if you have any discs on this server you don't want to wipe, however if it consistently gets through the wipe it may mean there is an issue with the graphics and the use of ncurses causing the graphics to hang.

After booting shredos and nwipe having appeared, control C to the point where it says press any key to restart nwipe, however don't restart nwipe. Leave it as it is and switch to the second virtual terminal ALT F2 then run the following command which will nuke ALL discs without displaying a GUI.

```
nwipe --nogui --autonuke --method=dodshort  &
```

To find nwipe process ID
```
ps -ef | grep -i nwipe
```

Then to view the progress you need to send SIGUSR1 to the nwipe process. You can type this command as many times as you want to view progress.
```
kill -SIGUSR1 nwipes_processID

You should get a response similar to:
nwipe: info: /dev/sdb: 00.42%, round 1 of 1, pass 1 of 3 [writing]
and it continues wiping ..
```

mdomanic, 2023-12-23:
So the problem I still have is that the request_irq error happens so frequently (almost like an infinite loop), it is impossible to interact with the system.  So in essence, the 'hang' im describing is the system being bogged down by this error reporting to stdout 

The lspci output I was able to get was from booting a different liveOS.

I'm wondering, is there a way to alter the kernel to ignore this error?  Forgive me as I am not too familiar with debugging Linux kernel stuff.

PartialVolume, 2023-12-23:
We could try blacklisting the drivers for i40e or/and megasas0 from the kernel command line. However we would need to know the correct driver name, hence the `lspci -k` and to get that to run in ShredOS you will have to switch back to loglevel=3.

PartialVolume, 2023-12-23:
Forget what I just said about blacklisting the drivers, that's not possible as the drivers are built into the kernel in ShredOS. You can only blacklist modules. You can only disable a built-in driver if the driver provides that option. However it will still be useful to know what drivers are being used for all the devices by running lspci -k from within ShredOS having set loglevel back to 3.

closed 2026-02-14 by PartialVolume