PartialVolume/shredos.x86_64 #200: Protocol serial number

ITTTJ, 2024-01-07:
Shredos is an excellent program that suits me perfectly with one small thing missing. I often come into contact with Dell computers. Is there any way to ensure that the serial number of the device (called Service Tag in Dell) in which the disk is plugged into is also written to the resulting erasure protocol?

PartialVolume, 2024-01-07:
That's great, glad it's useful.

The Dell service tag is written to the nwipe_log as baseboard-serial-number. At least it is on this Dell Optiplex 9010 I'm testing on.

I'm not sure what you mean when you say 'written to resulting erasure protocol'

ITTTJ, 2024-01-09:
Sorry, I misspoke, I meant for the information to be written to the PDF certificate. Would it be possible to write the baseboard-serial-number, system-serial-number and chassis-serial-number to the certificate? The express service tag is not that important, I can do without it.

PartialVolume, 2024-01-09:
Yes, that could be added. Because there will be some that don't want that information on the report it would need to be enabled in the config so it can be switched on/off depending on your preference.

PartialVolume, 2024-08-16:
I have not coded this yet, but this is my first draft of the host erasure certificate which will be generated along with the current disk erasure certificate. The difference between the two certificates being that the disk erasure certificate contains details of a single disk and no host information while the host erasure certificate contains details of the host and ALL the discs on the system. This host specific certificate will work with a laptop with a single disc to a 50 disc server.  With more than about six discs on a system the disk info will flow onto subsequent pages.

Please feel free to comment if you feel I've missed anything, like or don't like anything. Like I said, it's just a mock up at the  moment so now is the best time to make any changes.

[image]

PartialVolume, 2024-08-16:
@ITTTJ start end times and duration are already recorded on the second line of the drive details, for each drive, however I could put a more prominent start and finish time that records the finish time of the last drive to complete in the header or maybe the wipe section or is that too much?

I forgot to mention the smart data pages, this will be a user enabled option. As the smart data can sometimes run to 3 pages per disk wiping a server with 40 discs could generate a report containing  approximately 124 pages as opposed to 4 pages without the extra smart data pages.

Nwipe will be able to generate both types of report simultaneously so if you wanted you could have the disk orientated certificate with smart data pages and the host orientated certificate with or without smart data pages as per your preference.

ITTTJ, 2024-08-16:
I missed that. In that case it is alright this way and there is no need to change it ;)

Understand, no problem, it could be this way ;)

Thank you for your dedication, I appreciate it ;)

gossman, 2024-10-08:
Following this, looks great, any ideas when this updated version of the PDF will be available? 

Cheers :-)

BwalyaBwalya, 2025-03-25:
Hello @PartialVolume , following up on this wonderful additions. When can we expect a release for this?

MagicCarpetRider, 2025-06-06:
+1
Would be phenomenal to have!

iPat1337, 2025-07-31:
Device Info is only missing info, would be awesome to get it released soon.