PartialVolume/shredos.x86_64 #207: Boots to gray screen

kampelmuehler, 2024-01-25:
I tried booting into shredos on my machine with the `shredos-2023.08.2_25.0_x86-64_0.35_20231110` image, but all I'm getting is a gray screen (narrow light gray bar on top, rest of the screen darker gray). Modifying `grub.conf` to include nomodeset, which helped me previously with booting into Ubuntu live, doesn't change anything.

I'm on and AMD R5 2600 CPU and nvidia GPU - might there be an issue with nvidia?

PartialVolume, 2024-01-25:
Just double checking but there are two grub.cfg files, one for legacy boot, one for UEFI boot, so depending on which boot method you use you need to edit the correct one /EFI/BOOT/grub.cfg or /boot/grub.cfg

Edit whichever is supported by your system. If unsure edit both.

There is a problem with some configuration of Nvidia card, however nomodeset will normally fix the problem. Not all Nvidia cards are affected and it may only be display port output connector rather than Nvidia so could affect AMD too. Does your system use display port or DVI or VGA?

kampelmuehler, 2024-01-25:
Thanks for getting back to me on such short notice.

Indeed I must have edited the non EFI grub.cfg, editing the correct one resolved the issue - another big thanks!

And yes, running output through DP.

closed 2024-01-25 by kampelmuehler