PartialVolume/shredos.x86_64 #210: ShredOS freezes up

p1r473, 2024-01-27:
Hi!
Im having a weird issue where ShredOS freezes up,  often within first 10 minutes, though last time it froze up at around 2 hours mark
Ive tried multiple USB Drives. Im running it on a Lenvo Thinkpad

The whole PC seems to freeze, the screen freezes.

How can I best get you logs or debug info?

Firminator, 2024-01-27:
Is your USB controller on your Thinkpad overheating due to being taxed at 100% for hours? Wiping 3TB over USB 2.0 will take several days! I recommend to wipe two >1TB drives through a different interface preferably a PC's M.2 or SATA port. Or connecting one drive directly to the laptop's internal drive interface (M.2 most likely on newer laptops).

p1r473, 2024-01-27:
So the ThinkPad has 2 USB C and 1 USB A. I've used both 1 USB C and 1 USB A with a m2 nvme usb enclosure. I'm using USB 3 enclosures. The wiping starts at 400 mb/s but drops down to crap speeds within the first 10 seconds. But I am using usb 3 cables, ports, enclosures, and nvme drives

PartialVolume, 2024-01-27:
>The wiping starts at 400 mb/s but drops down to crap speeds within the first 10 seconds. But I am using usb 3 cables, ports, and enclosures. I'm not sure why the speeds drop so much

The initial burst of speed is because nwipe is writing to disk via the Linux disk cache. The burst of speed is due to the disc cache (CPU RAM) being filled rapidly at the start, once it is filled and all RAM is in use the speed drops to represent the real disk I/O speed as nwipe can't write into cache until some space has been made by writing to disc. Your speed of 40MB/s per disk is more like the true speed. Like @Firminator says for large drives you will achieve much faster wipe speeds. Depending on the drives you may see a four fold increase in speed, with M.2 you may see a ten fold increase in speed.

As regards the hang, it does sound like a hardware fault. Are the fans working, heatsinks blocked with dust, which might require opening up the laptop to properly check. Could just be a bad graphics chip. I had a laptop a couple of years ago that had a similar fault. You seem to have a fair amount of load on the USB bus with two NvMe drives and charging as well? I would probably just try a single drive and see if that works.

You could post dmesg.txt that's found on the usb stick and the output of lspci (ALT F2 for the second virtual terminal and type lspci and also lsusb.

closed 2024-01-27 by p1r473

Trump125532, 2024-02-01:
Im having the same issue here with multiple different Laptop models. Currently im trying to whipe the 512GB M.2 disk of a Fujitsu Lifebook U7410 and Shred OS just freezes after a couple of minutes (3-5m). The same issue was observerd on HP Elitebook 830 G6/G8. I tried a couple of different Shred OS releases but none seem to resolve this issue. All Laptops have an M.2 SSD. The Shred OS USB-Stick is the only thing connected to the Laptop and it doesnt matter which USB-Port I plug it in. The whole wiping process should take less than an hour, according to Shred OS's estimation and considering it crashing so early I dont think it's the USB-Controller overheating. I also opened one Laptop and there was no dust clogging the fan / heatpipes.