PartialVolume/shredos.x86_64 #211: Certificate not creating

ITTTJ, 2024-01-30:
I have a problem that a certificate is not created. Unable to create qla2xxx trace instance, instance logging will be disabled. During the booting is shown this information: Unable to create qla2xxx trace instance, instance logging will be disabled.
After that Shredos run. Doesnt matter how i set it, certificate is not created with info: No FAT32 formatted drive found, unable to archive nwipe log file.
USB drive is FAT32 formatted. I even tried to make new partition on same USB drive with FAT32 file system, but it doesnt help. I tested Venoty, Rufus and also unetbootin, but same result. Also tried two diferent USB flashdisks.

Dont you please know where could be the problem?

[image]
[image]

PartialVolume, 2024-01-30:
>Unable to create qla2xxx trace instance, instance logging will be disabled. During the booting is shown this information: Unable to create qla2xxx trace instance, instance logging will be disabled.

Ignore that, it's a benign error that always occurs and isn't a problem.

The ability to copy PDF's & logs to a non vanilla USB drive, i.e one that wasn't created with dd is restricted in the current version v2023.08.02_25.0_x86-64_0.35.

The updated code in the master will now write to Ventoy created USB drives as well as dd created USB drives,  however this hasn't been officially released yet.

You have three options with the current version to copy PDFs to USB:
- Use dd to create the USB drive by copying the .img file to it as described [here](https://github.com/PartialVolume/shredos.x86_64?tab=readme-ov-file#linux-and-mac-users)
- Manually copy the PDFs using mount & cp described [here](https://github.com/PartialVolume/shredos.x86_64?tab=readme-ov-file#reading-and-saving-nwipes-log-files---via-usb-manually-or-ftp-manually--automatically)
- Build the master copy from source (something that's maybe not worthwhile unless you are series about contributing as a developer long term)

Alternatively you will need to wait for the next release that will support that functionality for Ventoy.

One other favour to ask, if you used Rufus and also unetbootin to create the USB stick can you boot each in turn and switch to the second virtual terminal in ShredOS using ALT F2 then run the command `fdisk -l`     (-l is -lower case L not 1) I want to see how the partition on the USB stick is named, it's probably exFAT but for both RuFus & unetbootin I want to makesure that is the case. This will allow we to be sure all four methods are supported in upcoming release. Thanks.

PartialVolume, 2024-01-30:
To all Windows users with Linux experience. Is there a nice simple block copy command or windows application that just does a simple block copy like dd without messing with the drive formatting or partition table. It's so simple under Linux using dd that I can't believe somebody hasn't written a Windows application that just does a straight block by block copy. Does not Rufus or any other Windows program support a raw block copy?

Nebuli1, 2024-01-30:
https://dmde.com/download.html

free version can copy and save to file per sector

manually edit mbr and other advanced functions.

after starting the programme alt+w

PartialVolume, 2024-01-30:
> https://dmde.com/download.html
> 
> free version can copy and save to file per sector

Nice ! , Thanks for the info.

Firminator, 2024-01-30:
Well there is more to it and Pete from Rufus has the why coverered in his FAQ @

- https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_doesnt_Rufus_recommend_DD_mode_over_ISO_mode_for_ISOHybrid_images_Surely_DD_is_better
- https://github.com/pbatard/rufus/wiki/FAQ#user-content-Using_an_UEFI_bootable_ISO_based_on_grub_all_I_get_is_the_grub_prompt
- https://github.com/pbatard/rufus/wiki/FAQ#why-doesnt-rufus-create-a-windows-installation-usb-that-can-be-booted-in-dual-biosuefi-mode
- and to a more extent https://github.com/pbatard/rufus/wiki/FAQ#user-content-Blah_UEFI_Blah_FAT32_therefore_Rufus_should_Blah

PartialVolume, 2024-01-31:
Thanks @Firminator some bedtime reading 👍

Firminator, 2024-01-31:
Btw the qla2xxx driver is most likely a QLOGIC FC driver enabling support to connect to a FC SAN which is probably not the core idea of ShredOS. I've seen other wipers boasting with the ability to wipe FC drives, but I rather not have a software capable - on purpose or by accident - have access to a backend SAN in order to wipe it :)
Also I would assume enabling WoFC (Wiping over Fiber Channel; I might just coined a new term :headscratch:) would require a bunch of new additional code. Feature creep in my mind. So in other words it could probably be removed unless I missed the discussion on how it should be added. In my environment we have QLE2xxx FC adapters so that driver could also be removed.

PartialVolume, 2024-01-31:
I have no problem taking those drivers out, I do have some fibre channel kit but to be honest I've got my work cut out with other things that need doing in ShredOS and nwipe so can't see me ever getting round to looking at it any time soon if ever. So they will be removed in the next release.

ITTTJ, 2024-01-31:
> > Unable to create qla2xxx trace instance, instance logging will be disabled. During the booting is shown this information: Unable to create qla2xxx trace instance, instance logging will be disabled.
> 
> Ignore that, it's a benign error that always occurs and isn't a problem.
> 
> The ability to copy PDF's & logs to a non vanilla USB drive, i.e one that wasn't created with dd is restricted in the current version v2023.08.02_25.0_x86-64_0.35.
> 
> The updated code in the master will now write to Ventoy created USB drives as well as dd created USB drives, however this hasn't been officially released yet.
> 
> You have three options with the current version to copy PDFs to USB:
> 
> * Use dd or dmde (see [comment](https://github.com/PartialVolume/shredos.x86_64/issues/211#issuecomment-1917935151) below) to create the USB drive by copying the .img file to it as described [here](https://github.com/PartialVolume/shredos.x86_64?tab=readme-ov-file#linux-and-mac-users)
> * Manually copy the PDFs using mount & cp described [here](https://github.com/PartialVolume/shredos.x86_64?tab=readme-ov-file#reading-and-saving-nwipes-log-files---via-usb-manually-or-ftp-manually--automatically)
> * Build the master copy from source (something that's maybe not worthwhile unless you are serious about contributing as a developer long term)
> 
> Alternatively you will need to wait for the next release that will support that functionality for Ventoy.
> 
> One other favour to ask, if you used Rufus and also unetbootin to create the USB stick can you boot each in turn and switch to the second virtual terminal in ShredOS using ALT F2 then run the command `fdisk -l` (-l is -lower case L not 1) I want to see how the partition on the USB stick is named, it's probably exFAT but for both RuFus & unetbootin I want to makesure that is the case. This will allow we to be sure all four methods are supported in upcoming release. Thanks.
> 
> Example:
> 
> ```
> Disk /dev/sda: 1.88 GiB, 2013265920 bytes, 3932160 sectors
> Disk model: UDisk           
> Units: sectors of 1 * 512 = 512 bytes
> Sector size (logical/physical): 512 bytes / 512 bytes
> I/O size (minimum/optimal): 512 bytes / 512 bytes
> Disklabel type: dos
> Disk identifier: 0x00000000
> 
> Device     Boot Start    End Sectors  Size Id Type
> /dev/sda1        1263 509075  507813  248M  c W95 FAT32 (LBA)
> ```

ITTTJ, 2024-01-31:
To the Rufus and Unetbootin question. I dont see USB drive in Shredos at all, only main computers harddrive. I can show you info about the USB drive from windows. 

This is for Rufus:
[image]
[image]
[image]

PartialVolume, 2024-01-31:
Thanks, for Windows info isn't useful as I need to see exactly how fdisk lists the drive.

Is there something wrong with your hardware or is there something in the bios that powers down the USB hardware due to inactivity? Does unplugging it and plugging back in make it appear in fdisk.

That's a problem. ShredOS can't mount a drive if Linux can't see it in fdisk.

Is anybody else able to provide the fdisk output of a Rufus created ShredOS stick preferably from with the current version of ShredOS.

PartialVolume, 2024-01-31:
@ITTTJ Can you post the output of `lspci`, I want to see what USB controller you have. Also Make/model of system and or motherboard make/model if home built.

I'm seriously starting to think ShredOS needs some telemetry to start recording basic non-identifiable hardware details, like the details that lspci and /proc/cpuinfo produce.

PartialVolume, 2024-01-31:
@ITTTJ BTW that **disabling interrupt #16** is something I've not seen before, could that error coincide with fdisk -l not finding the USB stick? If you reboot multiple times does that message  **disabling interrupt #16** come up everytime? If it doesn't come up everytime, then on a occasion it doesn't appear does fdisk -l see the USB flash drive. Seems to be an issue on some other distros from  16 years ago. What vintage is this equipment?

ITTTJ, 2024-02-01:
There is nothing wrong with the hardware as far as I know, it behaves the same way with other flash drives and on other computers. Specifically in this case I am testing on a Dell Latitude E5550 laptop. Removing and reinserting the drive doesn't help, it's still not visible.

The flash drive is Kingston brand.

Even if I you reboot multiple times interrupt 16 come up everytime.

[image]

PartialVolume, 2024-02-01:
As I've never actually created a ShredOS stick with Rufus, as I normally just use dd as I'm on Linux the majority of the time, I thought I'd give it a go.

Downloaded the latest Refus & ShredOS and created the stick, booted it, USB drive appears in the nwipe selection screen as it should, as I've not enable nousb in the options. Jumped over to the second virtual window ALT F2 and run fdisk.

```
Disk /dev/sda: 1.88 GiB, 2013265920 bytes, 3932160 sectors
Disk model: UDisk           
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device     Boot Start    End Sectors  Size Id Type
/dev/sda1        1263 509075  507813  248M  c W95 FAT32 (LBA)
```

The filesystem **Type** string contains FAT32 so the current release will write to USB upon completion of wipes when you hit the return key for both Refus & dd created sticks. The next release will also write to Ventoy USB sticks which have the string exFAT in the **Type** field.

So, I'm thinking your Interrupt 16 may well be why your USB stick can't be seen, I see your USB hardware is `Intel Wildcat Point-LP xHCI & EHCI` which I've come across before but can't remember in what context.

> it behaves the same way with other flash drives and on other computers. 

Can you clarify that, other systems also don't show the USB stick in fdisk? Are they also Intel Wildcat USB ?

PartialVolume, 2024-02-01:
Apparently on some Dell systems you can enable/disable the external USB ports in the BIOS. Do these laptops have that feature? Or anything related to USB configuration?

PartialVolume, 2024-02-01:
From the Dell E5550 manual:
```
USB Configuration
This field configures the integrated USB controller. If Boot
Support is enabled, the system is allowed to boot any type of USB
Mass Storage Devices (HDD, memory key, floppy).
If USB port is enabled, device attached to this port is enabled and available for
OS.

If USB port is disabled, the OS cannot see any device attached
to this port.

Enable Boot Support

Enable External USB Port

Enable USB3.0 Controller
```

https://www.dell.com/support/manuals/en-uk/latitude-e5550-laptop/dell_latitude_e5550_bdw/system-setup-options?guid=guid-a0f58acf-0e8b-40a2-97f9-f67f38643d65&lang=en-us

ITTTJ, 2024-02-04:
I have all these settings turned on and had them before. I've tried practically every possible and impossible combination, but nothing has worked. I've now tried Shredos on a Latitude 5510 and everything works without a problem, so I have no idea why the older E5550 is not working...

[image]

PartialVolume, 2024-02-04:
Can you post the dmesg.txt file. One way to access it by doing headless login from another system as described [here](https://github.com/PartialVolume/shredos.x86_64#how-to-wipe-drives-on-headless-systems-or-systems-with-faulty-display-hardware-for-use-on-secure-lans-only)

PartialVolume, 2024-02-23:
Same as #224 Wildcat Point-LP USB

ITTTJ, 2024-02-24:
> Can you post the dmesg.txt file. One way to access it by doing headless login from another system as described [here](https://github.com/PartialVolume/shredos.x86_64#how-to-wipe-drives-on-headless-systems-or-systems-with-faulty-display-hardware-for-use-on-secure-lans-only)

I tried, but it was not working. I connected laptop inti local network trough cable, booted the computer into shredos, but remote connection was not working. I founs IP address of device, but even i tried 3 different software to telnet connection no one worked. I there please anything else i can do?

PartialVolume, 2024-02-24:
Unfortunately, with the current version headless login doesn't work as the telnetd software is missing. It will be present in the next version. If your bios is up to date you may have to wait for the next version of ShredOS which will have updated kernel & drivers.

PartialVolume, 2024-02-25:
With non functional USB the only other way to get the certificates off  is to FTP them as described [here](https://github.com/PartialVolume/shredos.x86_64#transferring-nwipe-log-files-to-a-ftp-server)

closed 2026-02-14 by PartialVolume