PartialVolume/shredos.x86_64 #212: Unable to boot

MinecollYT, 2024-01-31:
Hi, sadly I'm unable to use shredos (again). 

Might be releated to my first issue, but I'm not so sure about that. I've tried the normal and ventoy methodes:

Rufus + shredos-2023.08.2_25.0_x86-64_0.35_20231110.img + UEFI
> Stops (the _ is still blinking) after shredos reaches 100% while booting and displays "qla2xxx [0000:00:00.0]-0001:  : Unable to create qla2xxx trace instance, instace logging will be disabled."

Rufus + shredos-2023.08.2_25.0_x86-64_0.35_20231110.img + UEFI + nomodeset
> Freezes when shredos reaches 100% while booting

Ventoy + shredos-2021.08.2_23_x86-64_0.34_20221231**.iso** + UEFI
> Stops (the _ is still blinking) after shredos reaches 100% while booting and displays "qla2xxx [0000:00:00.0]-0001:  : Unable to create qla2xxx trace instance, instace logging will be disabled."

Ventoy + shredos-2023.08.2_25.0_x86-64_0.35_20231110.img + UEFI
> Screen goes blank after shredos reaches 100% while booting

PartialVolume, 2024-01-31:
>Might be releated to my first issue, but I'm not so sure about that.

You would have to remind me what your first issue was.

What's you hardware, make/model of system, make/model motherboard, lspci output, more /proc/cpuinfo. Method of connecting display to monitor, VGA, HDMI, DVI, Display Port?

Sounds like a framebuffer/DRM/Connector issue, however I see you did use nomodeset on your second Refus attempt. Did you apply nomodeset to both grub.cfg's /EFI/BOOT/grub.cfg and /boot/grub.cfg ?

PartialVolume, 2024-02-01:
Is a DP connection involved in any of the setup then that could be the issue. I believe DRM and the code involved in connecting a DP output doesn't work. On my Dell system with a Nvidia DP output you don't get any output. In my particular case adding nomodeset to the kernel stops it switch into DRM from simple framebuffer and then DP works correctly. I will shortly be updating buildroot and the kernel to the next release where I'm hoping this graphics issues got fixed.

Although the screen has frozen at 100% I suspect ShredOS hasn't frozen and is still running behind the scenes. Following the headless login on the ShredOS readme.md would prove that.

If you remove it from the docking station so you have no other monitors attached does it then work?

PartialVolume, 2024-02-01:
In addition can you post the output of lspci, I'm just wondering whether it reported as VGA Intel UHD

PartialVolume, 2024-02-01:
Sorry for the excessive posting, one last comment. I just googled it, looks like it has AMD integrated graphics with both HDMI and VGA outputs. If there is nothing coming up in the screen after it hits 100% you might want to see if it's coming out the VGA or HDMI connections. Like the Dell is there any setting in the Bios for how video switching is handled between VGA/HDMI and the docking station?

MinecollYT, 2024-02-02:
> If you remove it from the docking station so you have no other monitors attached does it then work?
During the test above, I only used the internal screen, no docking station or external monitor

The docking station support 2x dp and 1x hdmi, the notebook only supports 1x hdmi output. I can try using the onboard hdmi output

MinecollYT, 2024-02-02:
Nope, when using the normal rufus version it freezes at the qla2xxx point and when using nomodeset it freezes at 100 % shredos. Even when using the direct hdmi output

PartialVolume, 2024-02-02:
Are you able to retrieve the lspci output by booting another distro like gparted or do a headless login to ShredOS?

PartialVolume, 2024-02-02:
I see from your original issue #140 that this system has both Intel UHD and Nvidia graphics. Maybe it's the Intel UHD that's the issue has others have reported the same issue who also had Intel UHD graphics.

Kolli96, 2024-02-11:
I have the same problem with an HP EliteDesk 705 MINI G4 , AMD Ryzen 5 Pro 2400G , 16GB DDR4 machine. Shredos startup freezes in a black box at 100%.

PartialVolume, 2024-02-11:
@Kolli96 its noticeable how often HP systems have an issue. Does nomodeset in the two grub files help? How do you create your boot stick, dd, refuse or Ventoy?

Kolli96, 2024-02-11:
> @Kolli96On huomattava, kuinka usein HP-järjestelmissä on ongelmia. Auttaako nomodeset kahdessa grub-tiedostossa? Kuinka luot boot stick, dd, waste tai Ventoy?

I have tried all usb versions with the Rufus program.

Kolli96, 2024-02-11:
New information. If I disable the Legacy boot support and also the secure boot feature, Shredos startup normally. But then I have trouble getting into the HP bios again. HP boot screen stops "entering boot menu".

PartialVolume, 2024-02-11:
Have you tried the latest version with both /boot/grub.cfg and /EFI/BOOT/grub.cfg having had  `nomodeset` appended to the kernel command line using the latest version of ShredOS? And if that doesn't work try `nomodeset noapic`. Try all these variations in both legacy and UEFI boot.

Also, what type of connection do you have to the monitor. Is it VGA, DVI or Display Port?

Kolli96, 2024-02-11:
> Have you tried the latest version with both /boot/grub.cfg and /EFI/BOOT/grub.cfg having had `nomodeset` appended to the kernel command line using the latest version of ShredOS? And if that doesn't [...]

I haven't because I don't understand what they mean and how to change them. The monitor is connected with an HDMI cable. There is no other connection.

DaveMcCloud, 2024-04-30:
I had a similar problem trying to wipe a laptop today.  It stuck at 100%.  The laptop was connected to a USB-C dock so reading this thread I tried it again disconnected from the dock and it launched fine - had that qla error mentioned but not an issue and it went past.  Once booted I was able to plug the dock back in (for charging purposes).  It is happily wiping away.

UEFI was enabled, Secure boot not if anyone was interested.

NMExact, 2024-11-22:
Hi did this ever get a solution? I am having a similer issue with a hp server trying to boot shredos, I have downloaded and tried all versions however it hangs on 100% or it goes to a black screen and nothing happens. Any help on this would be great.

Thanks

PartialVolume, 2024-11-22:
> Hi did this ever get a solution? I am having a similer issue with a hp server trying to boot shredos, I have downloaded and tried all versions however it hangs on 100% or it goes to a black screen and nothing happens. Any help on this would be great.
> 
> Thanks

What's the output of dmesg.txt that you will find in the usb stick?

marsnebulasoup, 2024-12-19:
Have the same issue on a Lenovo P14s Gen 5 (Intel/Nvidia)

This is the dmesg.txt

```
[21031.970074] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21036.522079] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21041.094083] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21045.682086] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21050.286079] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21054.870080] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21059.438074] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21064.014068] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21068.590070] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21073.162066] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21077.765885] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21082.378083] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21086.994079] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21091.590082] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21096.170094] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21100.746075] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21105.325521] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21109.926068] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21114.513640] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21119.098072] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21123.670079] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21128.254080] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21132.838121] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21137.406119] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21141.982078] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21146.562066] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21151.130073] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21155.710078] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21160.290075] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21164.877645] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21169.466080] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21174.070090] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21178.686077] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21183.302071] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21187.862088] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21192.438087] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21197.006077] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21201.574093] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21206.158087] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21210.742073] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21215.306072] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21219.886076] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21224.478087] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21229.050513] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21233.638071] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21238.234076] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21242.818082] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21247.378072] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21251.966750] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21256.525949] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21261.114082] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21265.689399] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21270.270093] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21274.878078] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21279.432713] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21284.014132] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21288.581677] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21293.178334] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21297.722088] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21302.314081] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21306.914078] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21311.498082] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21316.089657] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21320.682077] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[21325.258089] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[... 2251 lines cut ...]
[31651.526379] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31656.118341] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31660.714343] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31665.318339] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31669.906335] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31674.502396] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31679.094382] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31683.678343] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31688.266355] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31692.837913] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31697.418359] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31701.998709] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31706.586346] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31711.178382] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31715.754346] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31720.342643] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31724.922296] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31729.481924] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31734.050385] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31738.646457] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31743.246359] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31747.862347] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31752.482378] e1000e 0000:00:1f.6 eth0: NIC Link is Down
[31757.036799] e1000e 0000:00:1f.6 eth0: NIC Link is Down
```