PartialVolume/shredos.x86_64 #220: Is it possible to have an ShredOS .img i586_32bit

CreaTech56, 2024-02-19:
I have an old 32bits machine to shred disks
In a 64 bits img file,  i successly modify grub.cfg  : loadkeys and nwipe options (method)
I want to do the same with a 32bits img
I try with the iso file, with compiling but without success
Thanks for advance

PartialVolume, 2024-02-19:
I'll build the x86_32bit .img file for you. I'll start the build later today and upload later today or tomorrow.

I'll let you know when it ready to download.

CreaTech56, 2024-02-19:
Thanks a lot
It's not hurry at all

PartialVolume, 2024-02-19:
i586_32bit now available:
Discussion:https://github.com/PartialVolume/shredos.x86_64/discussions/221

CreaTech56, 2024-02-20:
Thanks
I'll modifiy it soon with your instructions
ShredOS is very useful and the pdf is perfect to securise machine's donator

closed 2024-02-25 by PartialVolume