PartialVolume/shredos.x86_64 #225: Option to reboot after wipe is finished (--autoreboot)

kristianronningen, 2024-02-25:
Similar to the current `--autopoweroff` option to nwipe, I'd love to have an option such as `--autoreboot`. We use netboot to prepare laptops (wipe + install). Our netboot default boot is the fully automated installation of an OS. In the same menu we have the ShredOS --autonuke option.

Our goal is to manually select the correct wipe option during netboot, have the disk wiped, reports uploaded, and then an automatic reboot, causing the default OS install boot option to run right away.

PartialVolume, 2024-02-25:
Presumably if the pdf upload failed you would also want it to abort the auto reboot so you could still manually retrieve the reports.

Are you using FTP to upload the pdfs?

PartialVolume, 2024-05-26:
@kristianronningen The command `shredos_autoreboot=enable` can now be added to the kernel command line PR #256 as from version v2024.02.2_26.0_x86-64_0.37 onwards, which will be release shortly.

closed 2026-02-14 by PartialVolume