PartialVolume/shredos.x86_64 #235: HPE Smart Array P816i-a SR on Gen10 DL 380 server cannot find hard disk

kkkelvinkk, 2024-03-27:
We tried to boot ShredOS using an HPE DL 380 Gen10 server with HPE Smart Array P816i-a SR, but ShredOS is not able to find any hard disk. In addition, seems there is no option to enable HBA mode on HPE Smart Array P816i-a SR. Not sure the driver for HPE Smart Array P816i-a SR is inside ShredOS. Thanks a lot !

Firminator, 2024-04-13:
Looks like the P816i-A SR does have HBA/JBOD mode accd. to https://www.hpe.com/psnow/doc/a00047736enw.pdf
[image]

So I would think it's just a matter of finding where to disable RIAD mode and/or enable HBA. Usually you can do this in SSA or if the RAID controller GUI is exposed pre-boot (by pressing CTRL+R or similar).

PartialVolume, 2026-02-14:
Please reopen if still an issue with https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2025.11_28_x86-64_0.40 
Thanks.