PartialVolume/shredos.x86_64 #242: nwipe.conf and nwipe_customers.csv files overwriten by empty files

tviikari, 2024-04-10:
We are using shredos with pxe boot and shredos_config parameter to get nwipe.conf and nwipe_customers.csv from ftp server.
Some computer seems to take some time to get network up and running.
This results case that pinging server  fails and files are not get from ftp-server.
nwipe starts with default files in ram image.
After nwipe completes, network is up and running so default files are writen to ftp server, replasing files.

PartialVolume, 2024-04-10:
Looks like I may have the put a check in the nwipe launch script to delay the launch of nwipe until after the specified server is responding to a ping.

I'll see if I can fit that into the next release.

tviikari, 2024-04-11:
I made quick fix to prevent files in ftp server being overwriten. 
--- nwipe_launcher.orig 2024-04-11 10:03:16.148810188 +0300
+++ nwipe_launcher      2024-04-11 10:02:09.043929935 +0300
@@ -273,6 +273,7 @@
                fi
        else
                printf "[`date`] Pinging $config_ip FAILED, Check RJ45 network connection\n" 2>&1 | tee -a transfer.log
+               shredos_config_cmd_exists="no"
        fi
 else
        # if the shredos_config=".." doesn't exist on the kernel cmdline then we have to assume we booted from USB

tviikari, 2024-04-11:
For the network part....
ping command fails because the network connection is not ready. So there has to be delay.
For me "sleep 3" works. May be that less is ok.
So far I fixed it as follows:

--- nwipe_launcher      2024-04-11 10:02:09.043929935 +0300
+++ nwipe_launcher.net  2024-04-11 15:32:56.072775329 +0300
@@ -170,7 +170,8 @@
        printf "[`date`] Remote Server IP = $config_ip\n" 2>&1 | tee -a transfer.log
        printf "[`date`] Remote Server path = $config_path\n" 2>&1 | tee -a transfer.log

-       ping -c1 $config_ip  2>&1 | tee -a transfer.log
+       #ping -c1 $config_ip  2>&1 | tee -a transfer.log
+       until ping -c1 $config_ip  2>&1 1>/dev/null; do sleep 1;echo "Waitting network connection to server $config_ip";done
        if test ${PIPESTATUS[0]} -eq 0
        then
                # ***** FTP TRANSFER *****

PR #255 (2024-05-21): Various fixes to do with tftp/ftp transfers
1. Debug tftp transfers for restoring config files and outputing pdfs and logs.

2. Added a ping status delay for the ftp/tftp that only proceeds with launching nwipe if ftp/tftp servers have been configured on the kernel command line. The delay has a 30 second timeout upon which nwipe will launch. This fixes a problem where nwipe launches before the ethernet hardware is active and a IP address had not been obtained. This caused nwipe to not be able to read the config files from the ftp/tftp server.

3. Removed 4 second countdown on restarting nwipe. [... cut]
commit c78a7329 (2024-05-21): Merge pull request #255 from PartialVolume/Ping_server_and_wait

Various fixes to do with tftp/ftp transfers.
Fixes #242
PR #256 (2024-05-24): Merge pull request #255 from PartialVolume/Ping_server_and_wait
Various fixes to do with tftp/ftp transfers.
Fixes #242 Pauses for upto 30 seconds before launching nwipe if ftp/tftp transfers are being used. This is because if the user specified shredos_config="..." on the kernel command line in order to retrieve the nwipe.conf and nwipe_customers.csv files from a local ftp/tftp server, the server needs to be online before nwipe is launched. On some systems/networks the ethernet hardware initialisation and then DHCP requests can be slower than nwipe launching hence why we now ping the server t [... cut]
commit 3b94392d (2024-05-24): Merge pull request #255 from PartialVolume/Ping_server_and_wait

Various fixes to do with tftp/ftp transfers.
Fixes #242 Pauses for upto 30 seconds before launching nwipe
if ftp/tftp transfers are being used. This is because if the
user specified shredos_config="..." on the kernel command line
in order to retrieve the nwipe.conf and nwipe_customers.csv
files from a local ftp/tftp server, the server needs to be
online before nwipe is launched. On some systems/networks the
ethernet hardware initialisation and then DHCP requests can be
slower than nwipe launching hence why we now ping the server
 [... cut]
closed 2024-05-21 by PartialVolume