PartialVolume/shredos.x86_64 #245: Can it detect windows partitions?

Jeandu23, 2024-04-18:
Probably not but still asking.

PartialVolume, 2024-04-19:
Well yes, however this feature is not documented as far as I'm aware. And we also don't test this feature routinely.

So, if you are asking can it wipe the contents of a partition, then yes, it can, but you cannot initiate a partition wipe from the gui. You have to do this from the command line. i.e `sudo nwipe /dev/sdb1`. You can then view progress in the GUI.

It been a very long time since I used this, so experiment with the utmost caution and if the data on the disc is important even if in a different partition make sure you have backups. Like I say we don't test this, however hopefully it should work fine.

Now, if you are asking about detecting windows partitions because you don't want nwipe to wipe any discs that might contain windows partitions, we are working on a feature where nwipe will avoid enumerating certain drives, this could include discs that contain Linux or windows. At the moment you would have to manually exclude those drives using the exclude option on the command line.

Jeandu23, 2025-03-10:
Almost a year later, the computer I was trying to work on had bitlocker protection enabled, so the whole drive was encrypted from the get go. Thank you for the information thought.

closed 2025-03-10 by Jeandu23