PartialVolume/shredos.x86_64 #246: Stuck on 100%, GUI won’t open

cudeata, 2024-04-21:
The UEFI has no legacy and i’m booting from a USB created using Rufus with the 64 bit .img file from the shredOS github, I’ve disabled secureboot as well and the UEFI is updated.
The PC does boot shredOS from the flash drive but stays stuck on a loading screen at 100%, the GUI never shows up, the “_” at the top left does not blink either.
The line at the top of the screen is as follows (without the quotation marks of course): “ [ shredos  218.11MiB  100%  48.81MiB/s  ] “

Computer is an HP Omen 30L desktop GT13, Intel Core i7-10700k with 16gb of memory.

PartialVolume, 2024-04-22:
Does adding nomodeset or noapic to the kernel command line solve the problem?

PartialVolume, 2024-04-22:
This motherboard may have Intel integrated graphics on the motherboard, if it has a graphics card as well, try removing the graphics card and boot using the intel integrated graphics.

Mazolei, 2024-05-02:
I seem to be having this same issue, on an omen 40L GT21 desktop. This board doesn't have integrated graphics, and adding nomodeset and noapic to the command line doesn't seem to have helped, assuming I did it correctly (I will admit that my knowledge of linux is limited, so I may very well have done it wrong)

PartialVolume, 2024-05-02:
> I seem to be having this same issue, on an omen 40L GT21 desktop. This board doesn't have integrated graphics, and adding nomodeset and noapic to the command line doesn't seem to have helped, [...]

Feel free to post a photo of grub.cfg and I'll check it's correct.

Mazolei, 2024-05-02:
> Feel free to post a photo of grub.cfg and I'll check it's correct.

This is how I've got it now, and I've changed grub.cfg in both locations
[image]

PartialVolume, 2024-05-02:
> > Feel free to post a photo of grub.cfg and I'll check it's correct.
> 
> This is how I've got it now, and I've changed grub.cfg in both locations ![Screenshot 2024-05-02 155256](https://private-user-images.githubusercontent.com/11491164/327570276-c8b81ec7-44ab-49c6-8854-a55f89b6dd38.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ2ODUzMTQsIm5iZiI6MTcxNDY4NTAxNCwicGF0aCI6Ii8xMTQ5MTE2NC8zMjc1NzAyNzYtYzhiODFlYzctNDRhYi00OWM2LTg4NTQtYTU1Zjg5YjZkZDM4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTAyVDIxMjMzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQxNDAxYTIxMjg2MjFlNTFmMjlmMzk4MTc0Y2YzMmZhYmE2NjI3YTljNjdjNDk4NTg2MGI1ZGI1MzIxMDFiNGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.h3aSJt3_M7uxHh2Uup9Q_EEXS_lMeFu-DXiWuPwReQM)

No, you have the two commands in the wrong place. It should look like this:

```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 nomodeset noapic
}
```

Mazolei, 2024-05-02:
> No, you have the two commands in the wrong place. It should look like this:
> 
> ```
> set default="0"
> set timeout="0"
> 
> menuentry "shredos" {
> 	linux /boot/shredos console=tty3 loglevel=3 nomodeset noapic
> }
> ```

Haha, gotcha, that was my first instinct, I should have went with it. I'll try that and let you know how it goes

PartialVolume, 2024-05-02:
If it doesn't work with both then try it with one then the other:

```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 nomodeset
}
```
or
```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 noapic
}
```

Mazolei, 2024-05-02:
Running with both seems to have worked! I got some error about a qla2xxx trace instance, but the GUI loads and it seems functional

PartialVolume, 2024-05-02:
That's great. Don't worry about the qla2xxx trace instance, it's a benign error related to a fibre driver that's not needed.

I just had a quick look at the spec and it looks like you have a display port and HDMI output on that system. There is some issue sometimes with display port so if it hadn't worked I was going to ask you to try HDMI but maybe you are already using that to connect to the monitor.

Glad it's all working :+1:

Mazolei, 2024-05-02:
> That's great. Don't worry about the qla2xxx trace instance, it's a benign error related to a fibre driver that's not needed.
>
> I just had a quick look at the spec and it looks like you have a display [...]

Well, I seem to have run into some other issue; The GUI boots, but the drive I'm trying to clear doesn't appear, and the UI doesn't respond to any of my inputs. I was using a display port before, but swapping to my HDMI monitor alone doesn't seem to have helped.

PartialVolume, 2024-05-02:
Do you have noapic on the Kernel command line? If you do try removing it so you've just got nomodeset.

Mazolei, 2024-05-02:
> Do you have noapic on the Kernel command line? If you do try removing it so you've just got nomodeset.

That seems to have done it for real this time! Thank you very much for the help

PartialVolume, 2024-05-02:
> > Do you have noapic on the Kernel command line? If you do try removing it so you've just got nomodeset.
> 
> That seems to have done it for real this time! Thank you very much for the help

No problem, hopefully with the updated graphics drivers in the next release, adding nomodeset won't be necessary with some of these systems.

boruto79, 2024-07-23:
I'm having the same issue but it is not working with Dell Precision 3490 laptop. i tried all 3 methods. it works fine with the older laptop Dell Latitude 5440.

Toby75, 2025-06-27:
Hello, I observed the following fix for the issue. Initially I had an NVidia GT730 installed, after replacing it with an ATI HD6450, the issue was gone. Maybe this helps.

PartialVolume, 2025-10-31:
Which grub.cfg did you add `nomodeset` to? There are two grub.cfg files. One for legacy boot and one for UEFI boot, in different directories. If you have added `nomodeset` to the correct grub file does it look like this?
```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 nomodeset
}
```

nomodeset should immediately follow loglevel=3 with a space no newline. Reason I mention that, is that when the above text is viewed on my phone the text wraps and so nomodeset looks like it's on athe line below, it's not, it must be on the same line.

DaanSelen, 2025-10-31:
I also encounter this.

ADDING THE `nodemodeset` and `noapic` works.

Therefor my question, why not do this by default?

khvlids, 2025-11-05:
@PartialVolume From that link, I tried switching terminals by doing `ALT-F2` and I am unable to switch terminals. 

Yea so it loads up to 100% and then just stops 
`_                                                                   [shredos 268.71MiB 100%  18.48MiB/s] `

So it doesn't create a dmesg for this unfortunately.