PartialVolume/shredos.x86_64 #251: Can't Ignore Boot Disc

zlehnherr1, 2024-05-14:
Maybe I am doing something wrong, but even after adding the string to the 2 grub.cfg files, it still shows the USB stick. I also tried method 2, but the /etc/shredos directory doesn't exist by default. After making the directory and adding the blank file, USB also showed up.

PartialVolume, 2024-05-14:
A few questions

- I'm assuming you are referring to the instructions here #228 in a recent commit (not yet released in .iso/.img form).
- Are you building from source or burning a .img or .iso? If you are not building from source the instructions won't work on any of the released.img or .isos.

zlehnherr1, 2024-05-15:
I am using the .img, which would explain why this isn't working. Is that planned to be added to the future?

PartialVolume, 2024-05-15:
Yes, it will be in the next release which is imminent.