PartialVolume/shredos.x86_64 #252: nwipe v0.37

barnabyc, 2024-05-15:
Is it possible to create a new release now that nwipe v0.37 is available?

PartialVolume, 2024-05-15:
Yes, that's the plan.

barnabyc, 2024-05-15:
Wonderful, thank you! The nwipe README somewhat over-promises by saying "ShredOS Always has the latest nwipe release." but I appreciate your efforts. 😁

PartialVolume, 2024-05-18:
> Wonderful, thank you! The nwipe README somewhat over-promises by saying "ShredOS Always has the latest nwipe release." but I appreciate your efforts. 😁

I'm full of good intentions however other things in life often take priority, so can't always spend as much time as I would like on ShredOS and nwipe. While I really appreciate anybody that has bought me a coffee, my plumber gets paid more for a days work than I have ever received in contributions in the last couple of years. My 'real' job pays the bills and therefore gets the priority.

Unless there are any showstoppers I plan on releasing a new version based on buildroot 2024.2.2 and nwipe 0.37 sometime this coming week. I'm currently testing the latest x86_64 img version and it's looking good but need to build the i686 and .iso builds as I plan on releasing all versions simultaneously.

PartialVolume, 2024-06-10:
> > Unless there are any showstoppers I plan on releasing a new version based on buildroot 2024.2.2 and nwipe 0.37 sometime this coming week. I'm currently testing the latest x86_64 img version and it's looking good but need to build the i686 and .iso builds as I plan on releasing all versions simultaneously.
> 
> If you release it within the next day or two I will give you $15AUD. I will give you $10 if you give me a working image file right now. I will give $5 whenever its released, because the program is neat.

I like a challenge, I'll see if I can release it by tomorrow night for £15. 😁

Nebuli1, 2024-06-11:
it worked :D congratulations.

https://github.com/PartialVolume/shredos.x86_64/discussions/258

In fact, maybe a collection option could be considered? what do you think?

Firminator, 2024-10-20:
Reminder to leverage 'Sponsors': https://docs.github.com/en/sponsors/receiving-sponsorships-through-github-sponsors