PartialVolume/shredos.x86_64 #260: Asus laptop HDD not recognized

Aljs52, 2024-07-08:
Hello. I'm currently trying to wipe my laptop's main HDDusing schredos in a ventoy USB. However, It only shows the usb stick which I used to boot and not the other drive.

Laptop is an Asus Rog Gl753VD with lastest bios version and I'm also using schredos lastest version (v2024.02.2_26.0_x86-64_0.37). I tried dissabling fast startup and secure boot and making sure AHCI is enabled. I'm I missing something? Thanks.

PartialVolume, 2024-07-08:
In shredos, can you ALT F2 and run `lspci` and post the output so I can see what PCI hardware is installed. Does sound like a BIOS setting, but you've enabled AHCI which is often the cause.

Aljs52, 2024-07-08:
lspci returns the following output.

[image]

Also, AHCI is actually the only SATA config I can select in BIOS. Is this expected in a laptop?

PartialVolume, 2024-07-08:
Apologies, can you rerun the command as follows: `lspci -nn -k` This will show me the device IDs and which drivers are loaded. Thanks

Aljs52, 2024-07-09:
Sure thing.

[image]

PartialVolume, 2024-07-09:
Interesting I would have expected the HM170/QM170 to be bound to the ahci driver, unfortunately it doesn't appear to be bound to any driver.
[image]

Can you try binding the ahci driver to the HM170  device with the following command
`echo 0000:00:17.0 >  /sys/bus/pci/drivers/ahci/bind`

Run `fdisk -l`. Does fdisk now see the missing drive? Alternatively ALT F1 back to nwipe, do a control C to abort, at the prompt, press spacebar to restart nwipe. Does the drive appear?  Don't reboot ShredOS as the bind command isn't saved across reboots, at least I don't think it is, but I don't know that for certain.

PartialVolume, 2024-07-09:
Also can you post the output of dmesg.txt that you should find on the ventoy flash drive. That may hopefully show an attempt to bind the ahci driver to the SATA controller.

Aljs52, 2024-07-09:
The echo command returns `write error: Transport endpoint is not connected` and no extra driver is recognized afterwards.

PartialVolume, 2024-07-09:
Does dmesg.txt also contain this error? Do any live distros like ubuntu see the disk? What's the history of this laptop, I'm just wondering if it's a hardware fault or a buggy driver with this specific hardware? Was this running Windows or Linux before? Is the disc recognised in the bios?

PartialVolume, 2024-07-09:
Can you also confirm what storage you have, the specs say a 1TB HDD and a 128 or 256GB SSD M.2/AHCI, is that correct?

Aljs52, 2024-07-09:
The 1TB HDD is recognized by the bios and when live booting ubuntu 24.04. Right now it has a windows 10 system installed with the usual partitions. However, I do not have any SSD installed, although the motherboard has one slot for it.

I'll post the dmesg.txt file so you can have a look into it. I believe there is information related to the command at the end of it, but I'm not tech savvy enough to know what's relevant.

[dmesg.txt](https://github.com/user-attachments/files/16147918/dmesg.txt)

PartialVolume, 2024-07-09:
Thanks. I wondering if this could be a bug in the 6.6.x kernel, does either of the previous two releases (kernels 6.4.16 and 5.13.19 detect the drive?

Aljs52, 2024-07-09:
I tried with shredos 0.35 and 0.34 and remains the same.

Aljs52, 2024-07-10:
When booting schredos 0.34 the following prompt is given.

[image]

PartialVolume, 2024-07-11:
Good to know that Ubuntu live detects the drive. Can you boot Ubuntu live again and run `lspci -nn -k` and also `uname -a`

I can then see if I should include some extra drivers, like the Intel management driver.

Thanks.

Firminator, 2024-10-19:
The HM175 chipset might come with Intel Rapid Storage Technology (RAID). And it might be enabled preventing any OS to address and talk to the drive directly. I recall you can set the RST to be in RAID mode even though just having 1 disk.