PartialVolume/shredos.x86_64 #262: Throughput very low on HP Probook 450 G10

cyberm8, 2024-07-16:
Hello,

I've tried shredding disks on computers of a more recent model than usual, and although it works it takes a long time (several hundred hours). Throughput is very low.

[image]

Also, ShredOS takes a long time to arrive on the home screen (about 15 minutes), the message "Disabling IRQ #16" is displayed during this time.

I tried it with the latest version of ShredOS (very good features btw), no change.

The PCs that are slow are HP Probook 450 G10s, I have no problems on G7, G8 and G9.

[image]

I have swapped disks between a G9 and G10 to rule out disk problems. It seems to be specific to the G10.
I've tried updating the BIOS, with no improvement. I should point out that I have several G10s, so this isn't an isolated case, and the PCs also work very well.

I can manage by shredding from an older PC honestly, but maybe other people with recent hardware will have similar issues?

Thank you

PartialVolume, 2024-07-17:
Can you boot ShredOS and post the output of `lspci -nn -k` on both the G10 and G9 to see what drivers are loading. Thanks.

PartialVolume, 2024-07-17:
It has the symptom of a bad SSD, but you said you moved a good Samsung drive to the G10 and that was very slow also?

I came across this comment on Reddit about extremely high failure rates on Hynix  drives. What does smartmontools say about the condition of the Hynix drive? 

https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://www.reddit.com/r/Dell/comments/f6y9vf/extremely_high_failure_rate_on_brand_new_hynix/&ved=2ahUKEwiHhuLI46yHAxVjQ0EAHR-6A3kQjjh6BAgiEAE&usg=AOvVaw1Bc1wBcMdc3z4ESVidTNEn

cyberm8, 2024-07-17:
The tests have been performed on two G10. Both were slow.
Computers work well as I reinstalled Debian afterwards, without issue. They have been already used for several months without complaint.

I put a G10 disk into a G9 => boot is fast and shred speed is normal.
I put a G9 disk into a G10 => boot is slow and shred speed is slow.

Below, you can find LSPCI outputs (g9 and g10), dmesg output (g10) and `smartctl -a /dev/nvme0n1` output (g10).

[g9_lspci.txt](https://github.com/user-attachments/files/16262802/g9_lspci.txt)
[g10_lscpi.txt](https://github.com/user-attachments/files/16262810/g10_lscpi.txt)
[image]
[dmesg.txt](https://github.com/user-attachments/files/16262928/dmesg.txt)

PartialVolume, 2024-07-17:
When you installed Debian, what was the kernel version? `uname -a`. I'm just wondering whether to bump to the latest buildroot 24.05 which will give me a updated kernel and drivers.

cyberm8, 2024-07-18:
Debian 12 is using kernel 6.1.99-1.

cyberm8, 2024-08-04:
I was just wondering. You're implying that you need to update buildroot to get a newer kernel.

I don't know anything about buildroot, but I do know a little about Yocto.
With Yocto, you can specify the kernel version you want.

It seems possible [to choose the version](https://vocal.com/resources/development/using-a-custom-linux-kernel-in-buildroot/).
I'll try to test this on my own: build ShredOS with a 6.1.x kernel.

PartialVolume, 2024-08-16:
> Compiling an image with an old kernel is not conclusive, I get a black screen once the image is loaded. The compilation ends without error, strange. I'll keep on digging...
> 
> Also, I've tried the previous version of ShredOS ( v2023.08.2_25.0_x86-64_0.35), and it's no better (very slow shreding).

Regarding the black screen, is that with and without nomodeset on the kernel command line in grub.cfg?

cyberm8, 2024-08-18:
`nomodeset` is not set indeed, but I don't need it usually, it works without. I'l will give it a try.

Also, how do you generate an ISO ? I would like to run ShredOS on virtual machines.

PartialVolume, 2024-11-28:
@cyberm8 good to know, I'll do a rebuild and release as soon as I can.