PartialVolume/shredos.x86_64 #263: Trying to build a custom iso but it fails ..

sreusr, 2024-07-17:
... with output "DISK FULL" when creating the iso image. I changed only grub.cfg slightly with nwipe options. any idea how to fix this?

PartialVolume, 2024-07-17:
What procedure are you following when building the .iso?

sreusr, 2024-07-17:
like it's written in the readme, just had to activate the iso option in config.

sreusr, 2024-07-17:
also making a squashfs or enabling compression doesn't help, I thought this could spare some space.

sreusr, 2024-07-17:
on ubuntu 20.04

sreusr, 2024-07-17:
or can I use the . img for pxe booting too?

PartialVolume, 2024-07-17:
When I do it, there's a few extra things you need to do. I don't currently rely upon just selecting .iso in the menus.

Here's a list of what I do to change the .img build to a .iso build. You can probably ignore items 1,2 and 11.

**I've highlighted part of item3 which may be what your problem is.** The 'disk being full' is reference to the disk that is created for the USB image (.img). This disk can also be increased in size from it's current 280,000,000 bytes by editing the following entry in ../board/shredos/genimage.cfg as shown below

[image]

Building a legacy boot .iso image (not UEFI) 

Using make menuconfig..
1. IMPORTANT ! Check the directory in build options, change to the appropriate location of the buildroot you are actually working on. This is relevant for me when I'm copying the .img build to the .iso build in a different directory. 
2. Select Build Options, then change the directory from /home/bloggs/Software/shredos/Shredos.2020.05/shredos.2020.02/configs/shredos_defconf  to /home/bloggs/Software/shredos/Shredos.2020.05.iso/shredos.2020.02/configs/shredos_defconf , i.e. the only difference is appending .iso so the path becomes  ../Shredos.2020.05.iso/..
3. Select bootloaders->grub2-> i386-pc (old bios boot method) **and deselect EFI 64 as this causes a disk full error.**
4. Change Boot Partition from 'hd0,msdos1' to 'cd'
5. Select bootloaders->builtin modules include biosdisk & iso9660
6. Select Filesystem Images->iso image-> grub2
7. Save and then exit make menuconfig.
8. make savedefconfig
9. cp board/shredos/grub.cfg  fs/iso9660/grub.cfg
10. Edit fs/iso9660/grub.cfg and replace /boot/shredos with /boot/bzImage (IMPORTANT) Get the case correct, that's a capital 'I'  not 'i' in bzImage.
11. Make sure the 32 bit futex fix is added on 32 bit architecture. CONFIG_COMPAT_32BIT_TIME see keep note.  To determine whether its installed run make linux-menuconfig and search using '/' for CONFIG_COMPAT_32BIT_TIME , if its not set as 'Y' then goto (-> General architecture-dependent options) and select it.
12. make savedefconfig
13. make clean
14. make shredos_defconfig
15. make
16. Once completed and with no errors. cd output/images
17.  cp rootfs.iso9660 to the name of the .img file but instead of .img use the .iso extension.
18.  Makesure you are building for the specific architecture that you require, for instance if you were building for x86_i586
Run make menuconfig and change to archtecture from  x86_64 to i386 with the variant being changed from nocona to i586. Target Options->Target Architecture->i386, then Target Options->Target Architecture Variant->

PartialVolume, 2024-07-17:
With reference to **deselect EFI 64 as this causes a disk full error.**. It doesn't matter if EFI 64 is deselected as ShredOS uses it's own separately build EFI software, not the EFI build by buildroot. This is probably why you are running out of space because you are creating two sets of EFI software and why you need to deselect EFI 64 bootloaders->grub2->

sreusr, 2024-07-17:
thanks a lot for the clear details, I'll try it out and let you know.