PartialVolume/shredos.x86_64 #268: System serial number into PDF report

ITTTJ, 2024-07-26:
Hello,
is possible to put system serial number into PDF report? I can see that number in log, but not in PDF report.

Thank you!

cyberm8, 2024-07-26:
Hello,

This is a duplicate of #200

ITTTJ, 2024-07-26:
> Hello,
> 
> This is a duplicate of #200

Oh, thank you. I can see there is no solution :(

PartialVolume, 2024-07-26:
@cyberm8 Not yet, but it's just a matter of time. A second PDF format will be introduced that's system centric, i.e records details of the system and the drives. The user can then select whether they want the system centric or disk centric or both.

I was wondering whether to write the code for this extra PDF report or start working on the ATA secure erase but I think the new certificate code will be much faster to implement. So may get that out of the way first.

ITTTJ, 2024-07-28:
That would be great. I think more users would appreciate that feature.

bruce1949, 2024-08-01:
> @cyberm8 Not yet, but it's just a matter of time. A second PDF format will be introduced that's system centric, i.e records details of the system and the drives. The user can then select whether [...]

@PartialVolume 

   Thank you for your sharing this great tool with us and it is meaningful to have SN and other information in PDF certificate report. It would be great if we can modify it by ourselves via an easy way.

fthobe, 2024-08-26:
@PartialVolume Given that there's already work happening around the report for systems, I could crosscheck the current template against NIST and ISO requierements. what do you think?

PartialVolume, 2024-08-26:
Yes, that would be great. Thanks.

fthobe, 2024-08-27:
Can you add a full template with placeholder to the repo?

PartialVolume, 2024-08-27:
> Can you add a full template with placeholder to the repo like an example PDF?
> 
> I am not in reach of anything to test it out right now.

See #200 there's an example snapshot in the comments.