PartialVolume/shredos.x86_64 #270: disk not found in PowerEdge 710

johnyoungatwork, 2024-08-01:
I am using the latest ShredOS version v2024.02.2_26.0_x86-64_0.37 [ShredOS .img x86_64bit for USB Vanilla] to erase harddisks configured in RAID 1 and RAID 10. I added the "nomodeset noapic" to the /boot/grub.cfg and /EFI/boot/grub.cfg. The raid controller is Dell PERC H710P Mini and the raid bus controller is Broadcom/ LSI MegaRaid SAS 2208.

When I am booting the ShredOS, I see the following message:

Booting the kernel (entry_offset:)
Cant locate firmware cbfw-3.2.5.1.bin
Cant locate firmware cbfw-3.2.5.1.bin
Disabling IRQ #15

After booted to the GUI, only the ShredOS USB can be found under the "disks and partitions" section.
Switched to Alt + F3, the following messages were shown:

SELinux: Could not open policy file ‹= /etc/selinux/targeted/policy/policy.33
No such file or directory
INIT: version booting
INIT: No inittab.d directory found 
INIT: Entering runlevel: 3 
Saving 256 bits of creditable seed for next boot 
Starting syslogd: 0K 
Starting klogd: 0K 
Starting restorecond: 0K 
Running sysctl: 0K 
Starting thd: Device /dev/input/eventi not suitable. 
OK
mkdir: cannot create directory ' /var/lock': File exists 
Starting system message bus: done 
touch: cannot touch ' /var/lock/subsys/dbus-daemon': No such file or directory 
Starting rpcbind: 0K 
Starting iptables: 0K 
Starting network with shredos net.sh (hot plug): OK 
nohup: appending output to 'nohup.out' 
Starting ntpd: 0K 
Starting HPA's titpd: done
Starting collecta: OK
Welcome to ShredOS https://github.com/PartialVolume/shredos.x86_64 
shredos login:

Any hints on what I am missing?

Thanks in advance!

PartialVolume, 2024-08-02:
I would think the missing firmware file might have something to do with it. Can you post the output of `lspci -k -nn`. I want to check the controller ID. I'm having a bit of trouble locating that firmware in buildroot. Thanks.

johnyoungatwork, 2024-08-02:
Here is the lspci output, 

00:00.0 Host bridge [0600]: Intel Corporation Xeon E5/Core i7 DMI2 [8086:3c00] (rev 07)
	Subsystem: Dell Device [1028:048c]
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E5/Core i7 IIO PCI Express Root Port 1a [8086:3c02] (rev 07)
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: pcieport
00:01.1 PCI bridge [0604]: Intel Corporation Xeon E5/Core i7 IIO PCI Express Root Port 1b [8086:3c03] (rev 07)
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: pcieport
00:02.0 PCI bridge [0604]: Intel Corporation Xeon E5/Core i7 IIO PCI Express Root Port 2a [8086:3c04] (rev 07)
	DeviceName: SLOT 5
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: pcieport
00:02.2 PCI bridge [0604]: Intel Corporation Xeon E5/Core i7 IIO PCI Express Root Port 2c [8086:3c06] (rev 07)
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: pcieport
00:03.0 PCI bridge [0604]: Intel Corporation Xeon E5/Core i7 IIO PCI Express Root Port 3a in PCI Express Mode [8086:3c08] (rev 07)
	DeviceName: SLOT 6
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: pcieport
00:03.2 PCI bridge [0604]: Intel Corporation Xeon E5/Core i7 IIO PCI Express Root Port 3c [8086:3c0a] (rev 07)
	DeviceName: SLOT 7
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: pcieport
00:05.0 System peripheral [0880]: Intel Corporation Xeon E5/Core i7 Address Map, VTd_Misc, System Management [8086:3c28] (rev 07)
	Subsystem: Dell Device [1028:048c]
00:05.2 System peripheral [0880]: Intel Corporation Xeon E5/Core i7 Control Status and Global Errors [8086:3c2a] (rev 07)
	Subsystem: Dell Device [1028:048c]
00:11.0 PCI bridge [0604]: Intel Corporation C600/X79 series chipset PCI Express Virtual Root Port [8086:1d3e] (rev 05)
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: pcieport
00:16.0 Communication controller [0780]: Intel Corporation C600/X79 series chipset MEI Controller #1 [8086:1d3a] (rev 05)
	Subsystem: Dell Device [1028:048c]
00:16.1 Communication controller [0780]: Intel Corporation C600/X79 series chipset MEI Controller #2 [8086:1d3b] (rev 05)
	Subsystem: Dell Device [1028:048c]
00:1a.0 USB controller [0c03]: Intel Corporation C600/X79 series chipset USB2 Enhanced Host Controller #2 [8086:1d2d] (rev 05)
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: ehci-pci
00:1c.0 PCI bridge [0604]: Intel Corporation C600/X79 series chipset PCI Express Root Port 1 [8086:1d10] (rev b5)
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: pcieport
00:1c.7 PCI bridge [0604]: Intel Corporation C600/X79 series chipset PCI Express Root Port 8 [8086:1d1e] (rev b5)
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: pcieport
00:1d.0 USB controller [0c03]: Intel Corporation C600/X79 series chipset USB2 Enhanced Host Controller #1 [8086:1d26] (rev 05)
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: ehci-pci
00:1e.0 PCI bridge [0604]: Intel Corporation 82801 PCI Bridge [8086:244e] (rev a5)
	Subsystem: Dell Device [1028:048c]
00:1f.0 ISA bridge [0601]: Intel Corporation C600/X79 series chipset LPC Controller [8086:1d41] (rev 05)
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: lpc_ich
00:1f.2 SATA controller [0106]: Intel Corporation C600/X79 series chipset 6-Port SATA AHCI Controller [8086:1d02] (rev 05)
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: ahci
01:00.0 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5720 Gigabit Ethernet PCIe [14e4:165f]
	DeviceName: NIC1
	Subsystem: Dell Device [1028:1f5b]
	Kernel driver in use: tg3
01:00.1 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5720 Gigabit Ethernet PCIe [14e4:165f]
	DeviceName: NIC2
	Subsystem: Dell Device [1028:1f5b]
	Kernel driver in use: tg3
02:00.0 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5720 Gigabit Ethernet PCIe [14e4:165f]
	DeviceName: NIC3
	Subsystem: Dell Device [1028:1f5b]
	Kernel driver in use: tg3
02:00.1 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5720 Gigabit Ethernet PCIe [14e4:165f]
	DeviceName: NIC4
[... 198 lines cut ...]
7f:10.4 System peripheral [0880]: Intel Corporation Xeon E5/Core i7 Integrated Memory Controller Channel 0-3 Thermal Control 2 [8086:3cb4] (rev 07)
	Subsystem: Intel Corporation Device [8086:0000]
	Kernel driver in use: snbep_uncore
7f:10.5 System peripheral [0880]: Intel Corporation Xeon E5/Core i7 Integrated Memory Controller Channel 0-3 Thermal Control 3 [8086:3cb5] (rev 07)
	Subsystem: Intel Corporation Device [8086:0000]
	Kernel driver in use: snbep_uncore
7f:10.6 System peripheral [0880]: Intel Corporation Xeon E5/Core i7 Integrated Memory Controller ERROR Registers 2 [8086:3cb6] (rev 07)
	Subsystem: Intel Corporation Device [8086:0000]
7f:10.7 System peripheral [0880]: Intel Corporation Xeon E5/Core i7 Integrated Memory Controller ERROR Registers 3 [8086:3cb7] (rev 07)
	Subsystem: Intel Corporation Device [8086:0000]
7f:11.0 System peripheral [0880]: Intel Corporation Xeon E5/Core i7 DDRIO [8086:3cb8] (rev 07)
	Subsystem: Intel Corporation Device [8086:0000]
7f:13.0 System peripheral [0880]: Intel Corporation Xeon E5/Core i7 R2PCIe [8086:3ce4] (rev 07)
	Subsystem: Intel Corporation Device [8086:0000]
7f:13.1 Performance counters [1101]: Intel Corporation Xeon E5/Core i7 Ring to PCI Express Performance Monitor [8086:3c43] (rev 07)
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: snbep_uncore
[... cut ...] rs [8086:3ce6] (rev 07)
	Subsystem: Dell Device [1028:048c]
7f:13.5 Performance counters [1101]: Intel Corporation Xeon E5/Core i7 Ring to QuickPath Interconnect Link 0 Performance Monitor [8086:3c44] (rev 07)
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: snbep_uncore
7f:13.6 System peripheral [0880]: Intel Corporation Xeon E5/Core i7 Ring to QuickPath Interconnect Link 1 Performance Monitor [8086:3c45] (rev 07)
	Subsystem: Dell Device [1028:048c]
	Kernel driver in use: snbep_uncore

PartialVolume, 2024-08-02:
What drives have you got in this system, fibre channel, SAS or SATA?

PartialVolume, 2024-08-02:
I removed a qlogic fibre channel driver from the latest version of ShredOS. Can you try the previous version of ShredOS and see if that recognises the drives. Just so I can rule that out as the problem.

johnyoungatwork, 2024-08-02:
> What drives have you got in this system, fibre channel, SAS or SATA?

All of the hard disks are SAS.

johnyoungatwork, 2024-08-02:
> I removed a qlogic fibre channel driver from the latest version of ShredOS. Can you try the previous version of ShredOS and see if that recognises the drives. Just so I can rule that out as the problem.

Originally I tried the version v2023.08.2_25_x86-64_0.35 [ShredOS .img x86_64bit for USB Vanilla DRM], having the same issue, then I tried the latest version v2024.02.2_26.0_x86-64_0.37. I upgraded the firmware version shortly before using ShredOS, could it be the reason?

PartialVolume, 2024-08-02:
The firmware for the bios? I don't know. I think the most likely reason at the moment is the missing cbfw-3.2.5.1.bin firmware file which is for QLogic Infiniband, SCSI, Fibre Channel/FCoE adapters and Converged Network Adapters supported by the bfa, bna, ip_iband, qed, qla1280 and qla2xxx drivers according to Debian.

If this file is  missing from buildroot I can still download it from Debian and manually incorporate into ShredOS. I think I would also enabled the qlogic driver I disabled last time. The reason I disabled it was to get rid of the benign warning it showed at boot time but maybe it's needed for this controller.

The Poweredge R720 was released 2012 so it's not like it's new hardware that drivers have been written for yet.

No promises as to when I might be able to do this as I'm really busy at the moment. Are you hanging onto this R720 for a while, i.e will it still be available to test in a week or two?

johnyoungatwork, 2024-08-02:
Yes, I upgraded the firmware for the BIOS. The R720 is always available for testing, just take your time. Thank you so much for the help and I am looking forward to your news!

Firminator, 2024-10-19:
> cbfw-3.2.5.1.bin firmware file which is for QLogic Infiniband, SCSI, Fibre Channel/FCoE adapters and Converged Network Adapters supported by the bfa, bna, ip_iband, qed, qla1280 and qla2xxx 

These are all FC devices. However the PERCH710 is a RAID controller so not controlled by the FC driver. In other words the QLogic driver that was removed shouldn't be the culprit and doesn't need to be added back. I would think a possibly missing megaraid driver needs to be added to buildroot since the H710 is based on the LSI MegaRaid SAS 2208.

> 03:00.0 RAID bus controller [0104]: Broadcom / LSI MegaRAID SAS 2208 [Thunderbolt] [**1000:005b**] (rev 05)
> DeviceName: Integrated RAID
> 

According to https://wiki.debian.org/LinuxRaidForAdmins#megaraid_sas megaraid_sas supports 1000:005B

> megaraid_sas
> 
> Supported Devices
> 
> The following list is based on the alias fields of modinfo megaraid_sas in Debian 3.16 (3.16.7-ckt9-3deb8u1) kernel images.
> 
>     PCI: 1000:002F LSI Logic / Symbios Logic MegaRAID SAS 2208 IOV [Thunderbolt]
>     PCI: **1000:005B** LSI Logic / Symbios Logic **MegaRAID SAS 2208** [Thunderbolt]
>     PCI: 1000:005D LSI Logic / Symbios Logic MegaRAID SAS-3 3108 [Invader]
>     PCI: 1000:005F LSI Logic / Symbios Logic MegaRAID SAS-3 3008 [Fury]
>     PCI: 1000:0060 LSI Logic / Symbios Logic MegaRAID SAS 1078
>     PCI: 1000:0071 LSI Logic / Symbios Logic MR SAS HBA 2004
>     PCI: 1000:0073 LSI Logic / Symbios Logic MegaRAID SAS 2008 [Falcon]
>     PCI: 1000:0078 LSI Logic / Symbios Logic (Device name unknown)
>     PCI: 1000:0079 LSI Logic / Symbios Logic MegaRAID SAS 2108 [Liberator]
>     PCI: 1000:007C LSI Logic / Symbios Logic MegaRAID SAS 1078DE
>     PCI: 1000:0411 LSI Logic / Symbios Logic MegaRAID SAS 1068
>     PCI: 1000:0413 LSI Logic / Symbios Logic MegaRAID SAS 1068 [Verde ZCR]
>     PCI: 1028:0015 Dell PowerEdge Expandable RAID controller 5

Firminator, 2024-10-19:
This issue is linked to https://github.com/users/PartialVolume/projects/1/views/1?pane=issue&itemId=73347777

PartialVolume, 2024-10-19:
@Firminator `megaraid_sas` has been installed since `v2020.05.014_x86-64_0.31`. 

Regarding the message that appears during boot:
```
Cant locate firmware cbfw-3.2.5.1.bin
Cant locate firmware cbfw-3.2.5.1.bin
```
Seems to be for the Qlogic BR series cards which is for FC and NIC.

@johnyoungatwork does seem to have fibre channel in this system, hence why it's probably complaining about the missing firmware. However that doesn't explain why we are not seeing SAS, unless the raid controller is not operating in HBA mode.
```
43:00.0 Fibre Channel [0c04]: Cavium QLogic 415/815/41B/81B 4Gbps/8Gbps PCIe single port FC HBA [1657:0017] (rev 01)
Subsystem: Cavium QLogic 415/815 4Gbps/8Gbps single port PCIe FC HBA [1657:0014]
43:00.1 Fibre Channel [0c04]: Cavium QLogic 415/815/41B/81B 4Gbps/8Gbps PCIe single port FC HBA [1657:0017] (rev 01)
Subsystem: Cavium QLogic 415/815 4Gbps/8Gbps single port PCIe FC HBA [1657:0014]
44:00.0 Fibre Channel [0c04]: Cavium QLogic 415/815/41B/81B 4Gbps/8Gbps PCIe single port FC HBA [1657:0017] (rev 01)
Subsystem: Cavium QLogic 415/815 4Gbps/8Gbps single port PCIe FC HBA [1657:0014]
44:00.1 Fibre Channel [0c04]: Cavium QLogic 415/815/41B/81B 4Gbps/8Gbps PCIe single port FC HBA [1657:0017] (rev 01)
Subsystem: Cavium QLogic 415/815 4Gbps/8Gbps single port PCIe FC HBA [1657:0014]
```

PartialVolume, 2024-10-19:
@johnyoungatwork does state the SAS are configured as RAID 1 and 10 though, so maybe not in HBA mode.

@johnyoungatwork Can you switch the raid controller to HBA mode?

johnyoungatwork, 2024-12-11:
@PartialVolume Sorry for the late reply, HBA mode is not available for this raid controller, flashing the PERC H710P Mini is not allowed for some reasons. What else can I check?

PartialVolume, 2024-12-11:
I've ordered a PERC H710P so I'll get back to you once I've checked it out.

johnyoungatwork, 2024-12-13:
Thanks for the checking and helps!