PartialVolume/shredos.x86_64 #271: HP ML350 GEN10 doesn't see disks

jansiee, 2024-08-12:
On the HP ML350 Gen10 with controlelr P408I the disk don't show up.

They don't show up without creating any array/raid and not when i create a raid 0 for every disk.
Disk shoud be automatically be HBA mode on this controller.

What can i do?

PartialVolume, 2024-08-14:
I could be completely wrong here... but it could be that a special driver is required for the P408I, HP only provide binary drivers for Red Hat and Opensuse as far as I can see from their website and maybe the opensource megaraid driver doesn't work with this controller assuming you don't need to do anything to get the drives into HBA mode.

So if it was me I'd probably load up Fedora 39, 40 or rawhide onto a disc and install nwipe from their respository which is using the latest nwipe 0.37 and wipe the discs like that. I'm guessing nwipe is available in Redhat but don't know which version they are at.

However, somebody that has this hardware may know better workaround.

ps-pinky, 2024-11-15:
Hello @jansiee and @PartialVolume ,
I hope, i can give this al little more attention and hopefully ai've come over this project, cause I am in need to wipe some disks in HPE branded Servers with Smart Array Controllers only. 
As from my point of view, changing the configured array to pass the disks thru might be an option, if you don't  have serveral dozens of Servers or an ability to automatize that process. So this might be a quick and dirty workaround, for single servers, but not, if you got more and don't have plenty of thime.

So, no nerdy moaning, without a solution.
For the two servers i was able to test, I was able to do my nwipe jobs using the gparted livecd from https://downloads.sourceforge.net/gparted/gparted-live-1.6.0-10-amd64.iso which worked for the HPE storage controllers and nwipe.
So, from my investigation, from which I hope they are right, the gparted livecd has installed the hpsa driver under /lib/modules/$(uname -r)/kernel/drivers/scsi/hpsa.ko.xz from which I think is the thing, that does the trick with HPE based Smart Array Controller
and which I couldn't find in the shredos structure.

I think, that the integration of the hpsa module should therefore be integrated, as it is Open Source, from my understanding
(https://github.com/torvalds/linux/blob/master/drivers/scsi/hpsa.c (Line 8 to 10)), if I get that right.

So there starts my inability to build livecds, inlude modules in iat and so on and so forth and where I kindly need to ask to integrate this module in the Image.
What I can help with, is testing it on Hardware, if i get a specific iso file

Here are some more information, of what i found on a quick search
https://docs.kernel.org/scsi/hpsa.html
https://manpages.ubuntu.com/manpages/focal/en/man4/hpsa.4.html

It would be REEEEEAAAAL Great, if this module finds the way in the image.
Let me know, if I can help with that issue :-)

berst regards pinky

PartialVolume, 2024-11-15:
@ps-pinky All of ShredOS's drivers are built into the kernel rather than being .ko files you will find in the directory structure.

To view all builtin drivers you run the command `cat /lib/modules/$(uname -r)/modules.builtin`. There are about 1,430 drivers builtin so to look for the hpsa driver specifically, `cat /lib/modules/$(uname -r)/modules.builtin | grep -i hpsa`

As you will see the hpsa driver is already included in the latest ShredOS.

Boot the server that doesn't show the discs with the latest version of ShredOS. ALT-F2 to switch to the 2nd virtual terminal and type `lspci -k` and post the output here. This will list all the hardware on your system and show which modules are loaded and running and which hardware hasn't got a module loaded.

PartialVolume, 2024-11-15:
@ps-pinky Can you also boot up gparted on your server and run the `lspci -k` command and post the output here. When can then compare the ShredOS lspci -k output and the gparted lspci -k output and see what differs.

ps-pinky, 2024-11-15:
@PartialVolume well i think copying out of the ilo session aint possible, as this doesn't behave like a ssh session and I'm not quite sure, if im allowed to post internal stuff on a public platform, it's just some wird compliance stuff. I'm going to try to ship around that problem and check, if i'm going to get some old hardware, for that purpose to provide yo with informations as best, as i can.

PartialVolume, 2024-11-15:
@ps-pinky You don't need to copy anything out, just take a photo with your phone. There's nothing identifiable in the output of lspci -k, e.g. no UUIDs, serial numbers or anything, just the general names of the hardware. Tends to be all very common well known stuff. However if you don't want to post it publicly you can always email it to shredos.eraser@gmail.com

ps-pinky, 2024-11-15:
@PartialVolume hey there, i had some investigations on a hardware i could use for a couple of minutes, just to recognise, that shredos does not do uefi boots (yet) and i wasn't allowed to reconfigure it for the test.
what i found out from booting a gparted livecd, that the controller (a Smart Array P408I), if you do a ls -k uses:
Kernel driver in use: SmartPQI
Kernel Modulles: SmartPQI
the controller seems to be a serial attached scsi controller and the subsystem  then is the Hewlett Packard Company Smart array Controller.

from what i read and hopefully understand here: https://man7.org/linux/man-pages/man4/smartpqi.4.html
i think, that these modules replace the cciss and hpsa modules for those controllers.
what i can do is, that i grab myself another hardware, which i can reconfigure to Bios boot and test this against a gparted and a shredos instance and check, what lspci -k gives me in the comming week and give you some feedback here?!
thanks for being so quick in responding! :-)

PartialVolume, 2024-11-16:
@ps-pinky ShredOS does UEFI boot however it's not signed so you would have to switch off secure boot, however it still boots via UEFI.

The smartpqi driver is already installed in the latest ShredOS.
```
Symbol: SCSI_SMARTPQI [=y]                                                                                                                       │  
  │ Type  : tristate                                                                                                                                 │  
  │ Defined at drivers/scsi/smartpqi/Kconfig:40                                                                                                      │  
  │   Prompt: Microchip PQI Driver                                                                                                                   │  
  │   Depends on: SCSI_LOWLEVEL [=y] && PCI [=y] && SCSI [=y] && !S390                                                                               │  
  │   Location:                                                                                                                                      │  
  │     -> Device Drivers                                                                                                                            │  
  │       -> SCSI device support                                                                                                                     │  
  │         -> SCSI low-level drivers (SCSI_LOWLEVEL [=y])                                                                                           │  
  │ (1)       -> Microchip PQI Driver (SCSI_SMARTPQI [=y])                                                                                           │  
  │ Selects: SCSI_SAS_ATTRS [=y] && RAID_ATTRS [=y]
```

PartialVolume, 2024-11-19:
@ps-pinky Can you post dmesg.txt which you should find in the root of the USB flash drive. Thanks.

ps-pinky, 2024-11-20:
@PartialVolume hmmm, I'm sorry, I didn't mention, I'm booting from an iso file, which explains, that uefi boot might not work yet. ;-) does the iso file create a dmsg.txt too or is that a usb drive only thing?

ps-pinky, 2024-11-20:
@PartialVolume I'll do that, once I'm back on site in the coming days, will take a while. I'm going to come back here then! :-)

EmmV33, 2024-11-27:
I'm working on this myself so we can wipe some client drives and the lab HP we have would make this a lot quicker. That said have found the Driver for smartpqi in shredos is 2.1.24.046 and none of the drives come through. and in the ubuntu 2404 live cd it is version 2.1.26-030 which is able to mount all those drives. In dmesg on shredos I only get an error about MSI-X init failing with error -28. I'm currently teaching myself how to pull shredos down so I can try the updated driver in it to see if that is all it takes. Gonna be slow tomorrow so I'll let you know if I'm able to sort out buildroot and get this built proper.

ps-pinky, 2024-11-28:
@EmmV33 that would be great, if you have the opportunity to grab some hardware from the lab then me, cause I gotta take the long run until I see a lab or a data center again to grab myself some stuff to test. Would be really interesting to see, if it works with the newer driver or what that -28 error is. Keep fingers crossed it works with an updated driver and will follow this. I think @PartialVolume surely appreciates that input. I'll go and put in my help and output as well, as soon, as I can lay my hands on some metal again :-) 
Cheers

EmmV33, 2024-12-02:
Yeah my Holiday ended up being a busy work day because too many people decided to work on the holiday. go figure. I'm back at it now, though again have to give me patience, have not used buildroot before, and feels like doing it in WSL has some fun pitfalls I have to navigate first.

EmmV33, 2024-12-05:
Alright. I have been playing with this for a while and I'll admit this is a purely lack of knowledge issue, but @PartialVolume @ps-pinky or anyone else, where am I adding this new driver to the make directory? I have been able to add other programs and build those without issues but I have no idea where these drivers are loaded from, as the only config I can find is in the defconfig saying yes install smartpqi. Do I add these drivers via the patching mechanism that I haven't figured out yet? or do I turn smartpqi off in the linux-menuconfig and then add it as a package with the updated stuff? Trying to give an answer here but getting stuck so figured it be quicker to just ask here for any ideas.

ps-pinky, 2024-12-05:
@EmmV33  Well, I'm so sorry to say, but unfortunately I guess my knowledge of this it's even more limited than yours and I'm not a good in giving advice in adding a driver and building a livecd with that, but I guess and hope @PartialVolume has a better view on that.

PartialVolume, 2024-12-05:
Unfortunately I can't help you with that, I've added new packages/applications but not a driver other than via selection through make linux-menuconfig. I'm bumping the buildroot version shortly so other than a learning experience you could be wasting your time as it's all about to change.

EmmV33, 2025-03-12:
Sorry our company had a lot of big projects come through but we finally got back to this and wanted to update everyone that the updated ShredOS build DOES work with the Gen 10 HP servers that we have here and the controllers they have on board. so the driver must have updated and this problem at least on our end can be closed. Just wanted to let @PartialVolume know so he can let people know this problem should be resolved now. We have successfully wiped 48 drives with it.

PartialVolume, 2025-03-12:
> Sorry our company had a lot of big projects come through but we finally got back to this and wanted to update everyone that the updated ShredOS build DOES work with the Gen 10 HP servers that we [...]

That great, thanks for letting us know and glad it's proving useful.

ps-pinky, 2025-03-17:
Hey there I must confess, times an asshole and after all that Christmas stuff, vacation, sick leave and managing to handle HPE servers with redfish and trying to save a server for me to test, that went in the bin against my will, I read, that  @EmmV33 tested the new release, @PartialVolume threw out a while ago.
Today I just pulled myself a server, that is going to be in  production this week and tested it. Besides the uncomfortable side quest, that the ISO doesn't boot in uefi mode (i know, that the usb image does, but I can't access servers in a datacenter physically that fast and don't feel the need to) I can say, that after booting it in legacy mode, shredos and nwipe are now able to see the controller and the logical drive and I guess, if I wanted to, I could format the disks. Therefore I'd suggest @jansiee  now is a good chance to test and to close this issue 😊
Cheers