PartialVolume/shredos.x86_64 #273: shredos_output seems undocumented

hygri, 2024-08-20:
Hello! I've been trying to work out how to use the shredos_output kernel command line option to send reports to a TFTP server but, try as I might, I can't figure it out - have I missed something or is documenting this feature on the to-do list?

I'd love some pointers to get it going in the meantime. I've scrabbled about trying to work out the syntax but haven't got anywhere!

Thanks very much for the awesome work on ShredOS, it's amazing. Cheers :smile:

PartialVolume, 2024-08-20:
I'll try and put some documentation together for shredos_output and shredos_config later today.

PartialVolume, 2024-08-20:
Before I update the documentation are you using the following syntax?

```
	# Search /proc/cmdline for a command such as shredos_output="ftp:192.168.0.2:/home/joe/ftpdata/:jo:488993d:d"
	# Format:
	# shredos_output="protcol:IP_address:path:username:password:debug"
	# where:
	# protocol = ftp, tftp
	# path     =
	# username = username to access remote server, enter nothing between the colon delimiter and shredos will
	#            interatively ask for a username at boot up. Enter 'no user' if the server does not require a
	#            username.
	# password = password to access remote server, enter nothing between the colon delimiter and shredos will
	#            interatively ask for a password at boot up if using a protocol that supports authentication
	#            such as ftp, stfp (but not tftp). Enter 'no password' if the server does not require a
	#            password.
	# debug    = enter 'd' to enable debug mode in some protocols. In ftp mode this enables a detailed log of
	#            communication between ShredOS and the server. Only enable if you are trying to diagnose a
	#            communication error. View the contents of /transfer.log for a transcript of communications.
	#
	# Example:
	# A command to access a ftp server that has no authentication, writes to the default directory and is in debug
	# mode and does not require a interative request for username/password would look like this.
	# shredos_output="ftp:192.168.0.2::no user:no password:d"
	#
```

This example sends the PDF and logs via ftp to 192.168.0.2 into a directory /home/archive/pdf (which must already exist, you can make this directory whatever you want), the server requires a username and password, appending d means send full debug info to the log file, always append the d if you want verbose output sent to the transfer.log. 
```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 shredos_output="ftp:192.168.0.2:/home/archive/pdf_docs/:joe:gju76-HpoibX:d"
}
```

hygri, 2024-08-20:
Wow, thanks - this is exactly what I needed. I was attempting to re-hash the lftp syntax using semicolons which obviously isn't correct.

I'll be using tftp instead of ftp, so can I just omit the username / password, or do I need to include some sort of null character in there for you to parse? If not, and based on the above, I imagine this will work:

`shredos_output="tftp:192.168.0.2:/home/joe/ftpdata/:d"`

I've already got tftpd_hpa configured, directories and permissions created etc so that part is up and running.

Many, many thanks!

PartialVolume, 2024-08-20:
For tftp you don't have login credentials so the syntax might look something like

```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 shredos_output="ftp:192.168.0.2:::no password:d"
}

Notice the three colons which represent fields 3, 4 and 5

field 3 empty for chrooted tftp server
no path (the files get written to the root of the tftp server, on the server tftp chrooted root might be /srv/tftp/

field 4 empty
no username, i.e an empty field, 

field 5 empty
no password required for tftp server. However you must enter `no password` in this field.

hygri, 2024-08-20:
Gotcha, that's brilliant. I'll try it in a bit and report back!

PartialVolume, 2024-08-20:
Syntax is the same for `shredos_config=`, which you might want to use in addition to shredos_output. If shredos_config= is found on the kernel command line shredos will read from and write back the nwipe.conf and customers.csv files to the tftp/ftp server. You would probably want to use shredos_config if you are booting via PXE rather than a USB stick or .iso.

PartialVolume, 2024-08-20:
Also for test purposes you can also add /dev/loop0 /dev/loop1 etc to the end of the kernel command line. ShredOS will create two dummy 1MB virtual discs, actual discs on the system won't be shown in nwipe just the loop devices. 

This makes it real fast to generate a report by wiping the loop device and saves the time having to wipe a real disc or of course you could start a wipe on a real disc and control C to abort the wipe which will also generate a report.

hygri, 2024-08-20:
Brilliant, all sorted. This rig is running off an internal ssd with shredos_exclude_disc; all the config live there but I might pull them from tftp in the future. I used the loopback devices for testing (much quicker!) and also managed to set a path inside the tftpserver - my config line looks like this now:

shredos_output="tftp:10.0.0.10:shredos/::no password:d"

A-mazing. Thank you so much for your help!!

PartialVolume, 2024-08-20:
No problem, glad it's useful.

closed 2024-08-20 by hygri