PartialVolume/shredos.x86_64 #274: Package Request: Add parallel to allow multiple sg_format commands at the same time

fthobe, 2024-08-24:
There are several sg_format or hdparm operations that make sense to be executed in parallel, for example the formating of a set of multiple disks at the same time as outlined here: [Remove Integrity Protection](https://github.com/gms-electronics/formatingguide/blob/main/removeintegrityprotection.md)

The inclusion of parallel would only add 324.4 KB to the iso as all required dependencies 

1. [perl](https://archlinux.org/packages/core/x86_64/perl/)
2. [procps-ng](https://archlinux.org/packages/core/x86_64/procps-ng/)

PartialVolume, 2024-08-24:
Yes, I can add parallel to ShredOS. Were to thinking of submitting a PR or do you want me to do it?

fthobe, 2024-08-24:
I never used arch, if you give me some hints I can do a pull request later today, I am in Asia right now so it's still early

PartialVolume, 2024-08-25:
Looks like all the correct files have been changed. Sounds like you manually made those changes, was parallel not in packages using make menuconfig or make busybox?

I'll download your branch and build it later today.

fthobe, 2024-08-25:
Hey, I am not deep into the Linux embedded world so it was a first for me. I pieced it together using other packages and a commit for a package you made in the past. If you have some useful links I am happy to take a look at them!

closed 2024-08-26 by fthobe

fthobe, 2024-08-28:
@PartialVolume I should have fixed literally everything and I was successful building it.

PR #275 (2024-08-31): #274 ATTENTION: FIRST COMMIT, TRIPLE CHECK FOR MISTAKES Added parallel package to buildroot
ATTENTION: First time I did this, verify with care!!! 

1. Added parallel package 20240822 to buildroot to close #274 including config, hashfile and .mk
2. Sorted package list configs/shredos_defconfig alphabetically
commit 1fe0ac88 (2024-08-31): #274 ATTENTION: FIRST COMMIT, TRIPLE CHECK FOR MISTAKES Added parallel package to buildroot (#275)

Added parallel package to buildroot to close #274
Added .DS_Store due for development on mac clients.
Updated readme.md regarding sg3_utils & parallel