PartialVolume/shredos.x86_64 #276: ShredO

Happy-Hardcore, 2024-08-26:
ENGLISH.

Hello ShredOS and Nwipe creators.

Have fun with your projects and good luck.

I have some questions.

If I have erased a data carrier by overwriting it with Nwipe
(possibly packaged in ShredOS)
and if that has succeeded,
is it still necessary to test the data carrier for:
Bad Sectors/Damaged Blocks/or if a part of the data carrier is broken,
or is it certain that the data carrier is not partially broken,
and is a check whether the data carrier is partially broken not necessary?

How and where can I see in a report that the data carrier is not broken from
Bad Sectors/Damaged Blocks/or if a part of the data carrier is not broken?

If you want to respond, would you please respond in easy and simple words?

Have a nice time.
Kind regards, username Happy-Hardcore.

Dutch/Nederlands

Hallo makers van ShredOS en Nwipe.

Veel plezier met jullie projecten en succes.

Ik heb wat vragen.

Als ik een gegevensdrager heb gewist door het te overschrijven met Nwipe
(mogelijk verpakt in ShredOS)
en als dat gelukt is,
is het dan nog nodig om de gegevensdrager te testen op:
Bad Sectors/Damaged blocks/of een gedeelte van de gegevensdrager kapot is,
of is het zeker dat die gegevensdrager dan niet gedeeltelijk kapot is,
en is een controle of de gegevensdrager gedeeltelijk kapot is niet nodig?

Hoe en waar kan ik in een rapport zien, dat de gegevensdrager niet kapot is van
Bad Sectors/Damaged Blocks/een gedeelte van de gegevensdrager niet kapot is?

Als u wilt reageren, wilt u dan in makkelijke en simpele woorden reageren?

Een Prettige tijd.
Met vriendelijke groeten, gebruikersnaam Happy-Hardcore.

fthobe, 2024-08-26:
Sooo... there are some issues here: 
nwipe is not searching for bad sectors, but might indicate you in multiple ways that there are some present on your disk: 
1) the pdf report created by nwipe lists reallocated_sector_ct (keep in mind that the values are not fully standardised), a value of 100 indicates usually that the disk itself doesn't believe that it has bad blocks, lower numbers usually indicate decay of the disk

2) if too many blocks can't be wiped it usually indicates a large quantity of bad blocks > indicates decay of the disk

If you have any of the problems listed above, a deep reformat using for example sg_format for SAS disks might help, otherwise decomission and destroy the disk physically as you can't warrant a full wipe.

fthobe, 2024-08-27:
@Happy-Hardcore Also a better title would be: can I check for bad sectors with nwipe / shredos?

Happy-Hardcore, 2024-09-16:
> Dus... er zijn hier wat problemen: nwipe zoekt niet naar slechte sectoren, maar kan op meerdere manieren aangeven dat er wel slechte sectoren op je schijf aanwezig zijn:
> 
> 1. het pdf-rapport dat door nwipe is gemaakt, vermeldt reallocated_sector_ct (houd er rekening mee dat de waarden niet volledig gestandaardiseerd zijn). Een waarde van 100 geeft doorgaans aan dat de schijf zelf niet gelooft dat er slechte blokken in zitten. Lagere getallen geven doorgaans aan dat de schijf is versleten.
> 2. Als er te veel blokken niet gewist kunnen worden, duidt dit meestal op een grote hoeveelheid slechte blokken > duidt op verval van de schijf
> 
> Als u een van de hierboven genoemde problemen ondervindt, kan een grondige herformattering, bijvoorbeeld met sg_format voor SAS-schijven, helpen. Anders moet u de schijf buiten gebruik stellen en fysiek vernietigen, omdat u geen volledige wissing kunt garanderen.
> 
> Daarnaast vindt u via [deze link een goede handleiding om met de juiste hulpmiddelen te controleren op slechte blokken](https://wiki.archlinux.org/title/Badblocks) .

Nederlands

Hallo,

bedankt voor uw reactie.
uw uitleg is wat ingewikkeld.

Als ik zeker wil weten of een opslagapparaat wel of geen bad sectors heeft, is het dan handig om dat te testen met andere software dan ShredOS en Nwipe?

Bedankt voor uw reactie.
heb een prettige dag.
Groetjes van gebruikersnaam: Happy-Hardcore

Engels

Hello,

Thanks for your response.
Your explanation is a bit complicated.

If I want to know for sure whether a storage device has bad sectors or not, is it useful to test that with software other than ShredOS and Nwipe?

Thanks for your response.
Have a nice day.
Greetings from username: Happy-Hardcore

PartialVolume, 2024-09-16:
For a hard disc, If ShredOS/nwipe wipes your drive without any I/O errors and in the smart data on page 2 it says reallocated sectors = 0 you don't have any bad sectors.

PartialVolume, 2024-09-16:
For clarification badblocks and ShredOS/nwipe detect badblocks in much the same way, by destructively writing to every block on the disc. However nwipe has many wipe methods so to detect badblocks it's recommended you use a wipe method that includes a prng, i.e you don't just do a zeros wipe. While not specifically a bad block I have had a drive that passed a zeros and verification but failed a prng and verification. The cause was a stuck bit in the data channel, permanently stuck at zero so not detected on a zeros wipe, but absolutely detected using a prng wipe. Sometimes you get unusual faults on drives.

As regards reformatting to fix bad blocks blocks, I wouldn't waste your time. If the drive is more than three years old and has thousands of reallocated sectors it should be wiped, assuming it can be wiped without errors and then physically destroyed.