PartialVolume/shredos.x86_64 #277: Package procps-ng inconsistencies

fthobe, 2024-08-26:
Packages that rely on procps-ng can not be added as .mk and config.in are inconsistent with procps-ng folder name.

fthobe, 2024-08-26:
Temporary fix pushed, needs to be reported upstream to rootbuild (see [commit related to adding packages that depended on procps-ng](https://github.com/PartialVolume/shredos.x86_64/pull/275/commits/a1e3b1d5fea5969a6804c1dfad68b13b56624501)

fthobe, 2024-08-27:
[Issue raised with rootbuild](https://gitlab.com/buildroot.org/buildroot/-/issues/31)

fthobe, 2024-08-27:
Hey, I posted the issue but they told me to go "upstream", what does that mean in this case? Package developer?

PartialVolume, 2024-08-27:
Upstream to me would mean the package developer but the only files you had to fix was the .mk file, which I wouldn't have thought was anything to do with the upstream developer.

Is the upstream developer responsible for the buildroot makefile? I wouldn't have thought so.

I'd probably ask them to clarify that statement after looking at the changes you made in the .mk diff and why that's a upstream issue?

https://github.com/PartialVolume/shredos.x86_64/pull/275/commits/a1e3b1d5fea5969a6804c1dfad68b13b56624501

PartialVolume, 2024-08-27:
I was just reading the reply at https://gitlab.com/buildroot.org/buildroot/-/issues/31

To be honest I wouldn't waste any time on it. You've reported it and he's asked you to report bugs on the master buildroot code not an earlier version. If you really wanted you could just git clone the master copy of buildroot, i.e the current development version and just run the basic make menuconfig commands and try and add parallel. If it doesn't work report that. Depends how much time you want to spend on is. You've fixed it in ShredOS and if it is indeed working in the later releases it will be taken care of in my update commit.

PartialVolume, 2024-08-27:
BTW, when I search for procps in ShredOS, I only see one procps, not two. I know you mentioned two in your buildroot report and one of the maintainers of buildroot searched and only found one procps upstream which may be why he dismissed the report. I don't know why you are seeing two, did you create a procps-ng or procps_ng entries in package/Config.in or maybe two procps folders under package?

[image]

fthobe, 2024-08-28:
> I was just reading the reply at https://gitlab.com/buildroot.org/buildroot/-/issues/31
> 
> To be honest I wouldn't waste any time on it. You've reported it and he's asked you to report bugs on the master buildroot code not an earlier version. If you really wanted you could just git clone the master copy of buildroot, i.e the current development version and just run the basic make menuconfig commands and try and add parallel. If it doesn't work report that. Depends how much time you want to spend on this. You've fixed it in ShredOS and if it is indeed working in the later releases of buildroot it will be taken care of in my update commit, if not it will break the enabling of parallel so we would have to patch the .mk file again. I'll try to remember to test parallel is present after the next buildroot update.
> 
> Of course, you may enjoy the challenge of proving that bug exists in the buildroot master code. 😉 I'll leave it up to you. 😊

Thomas was right not to humor my BS. 
Works as expected now, loads packages as expected and adds parallel as expected. 
https://github.com/PartialVolume/shredos.x86_64/pull/275/commits/c0ba685b10bd4c06673b0319b6e0457b239847d2

The issue was the following: 
I added 
`depends on BR2_PACKAGE_PROCPS-NG`
instead of 
`depends on BR2_PACKAGE_PROCPS_NG`
for a misunderstanding of the syntax because for me package names are the holy cow. 

In my feeling make menuconfig should throw me an error, instead it fetches your depends on and throws it out as a package without any further information (which I find tremendously wrong). 

Anyhow fixed,
Fabian

closed 2024-08-28 by fthobe