PartialVolume/shredos.x86_64 #278: Stuck on 100%, GUI won’t open #246 same issue

boruto79, 2024-08-26:
I tried everything on the #246 and still having issue. i think it might be a kernel issue. I have a Precision 3490 and it will not load. according to  dell site it works with kernel 6.8 and newer. Is there anyway to update the Kernel to 6.10.5 or anything after 6.8?

PartialVolume, 2024-08-26:
If you are building from source you can select a custom kernel however I've never done that so can't give you any pointers.

However I am moving ShredOS very shortly to the latest stable buildroot 2024.05.2 which will most likely have a newer kernel than 6.8

boruto79, 2024-08-26:
Latest release candidate: 2024.08-rc2 has the latest 6.10.5. I am still learning buildroot and just Linux as a whole. i copied the shredos from package, board and config folder. I don't know if that is all I need to grab so i can try to compile from source.

PartialVolume, 2024-08-26:
There's a bit more to it. There are custom packages, nwipe being one of them that also needs to be copied.

Updating can be a challenge but the way I've found works best for me through trial and error is to copy the new version of buildroot ontop of the old version. In this way all the buildroot files get updated and all the extra packages are still in place.

When you build it with make it will usually fail due to various patch files, you need to then delete the old patch file, look at the file date but keep the new patch files. There is other things like updating version.txt in board so the nwipe title has the correct version info. When you run make buildroot usually recognises you have updated and may ask you a question or two.

I guess I should really write a procedure that goes through the steps when I do the next update.

boruto79, 2024-08-26:
what about the kernel defconfig file? do i need to build a new one?