PartialVolume/shredos.x86_64 #283: ftp over ssh support (sftp)

tadvaas, 2024-09-13:
No documentation on how to use sftp to send pdf reports to remote server on port 22.

Tried shredos_output="protcol:IP_address:path:username:password:debug" does not work with protocol set as sftp, lftp seems to not work when it comes to sftp

PartialVolume, 2024-09-13:
sft support has not been implemented yet. It will be available in the upcoming release.

tadvaas, 2024-09-14:
For anyone looking sftp support right now to upload files on completion:

lftp="set sftp:auto-confirm yes; open sftp://user:pass@host; cd /home/shredos/data/; mput *.pdf; mput nwipe_*.txt"

tadvaas, 2024-09-14:
When it comes to shredos_config="", is there a SFTP support for that yet?
I can see that the command uses LFTP to fetch the config files from remote server? Any way to supply additional parameters?

PartialVolume, 2024-09-14:
I'm currently testing a sftp update for both sending pdfs/logs using shredos_output and importing .conf and customers files using shredos_config.

tadvaas, 2024-09-16:
That's great. Just want to thank you for developing this software, it really is great. If you have any alphas to run, please do let me know. Any ideas when is the new release coming out?