PartialVolume/shredos.x86_64 #285: Intermittent kernel bugs

lowland3r, 2024-09-13:
This may be a hardware problem, but I'm running into issues where nwipe fails partway through a wipe process with a Kernel bug.

I can try to capture the full, raw message at some point, but the gist is as follows:

```
kern.warn kernel: ------------[ cut here ]------------
kern.crit kernel: kernel BUG at include/linux/scatterlist.h:115!
kern.warn kernel: invalid opcode: 0000 [#1] PREEMPT SMP PTI
kern.warn kernel: CPU: 2 PID: 94 Comm: kworker/2:1H Tainted: G            W              N 6.6.22 #2 
kern.warn kernel: Hardware name: Gigabyte Technology Co., Ltd. H310M A 2.0/H310M
kern.warn kernel: Workqueue: kblockd blk_mq_run_work_fn
kern.warn kernel: RIP: 0010:__blk_rq_map_sg+0x314/0x470
```

PartialVolume, 2024-09-13:
I was reading that `PREEMPT SMP PTI` i.e  PTI(Page Table Isolation) is a Spectre/Meltdown correction at the kernel bios level. Whether that's correct, I don't know.

What's the output of lscpu

Should show something like this.
```
lscpu
Architecture:             x86_64
  CPU op-mode(s):         32-bit, 64-bit
  Address sizes:          39 bits physical, 48 bits virtual
  Byte Order:             Little Endian
CPU(s):                   20
  On-line CPU(s) list:    0-19
Vendor ID:                GenuineIntel
  Model name:             12th Gen Intel(R) Core(TM) i7-12700H
    CPU family:           6
    Model:                154
    Thread(s) per core:   2
    Core(s) per socket:   14
    Socket(s):            1
    Stepping:             3
    CPU max MHz:          4700.0000
    CPU min MHz:          400.0000
    BogoMIPS:             5376.00
    Flags:                fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc 
                          art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma c
                          x16 xtpr pdcm sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l2 cdp_l2 ssbd ibrs ibp
                          b stibp ibrs_enhanced tpr_shadow flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid rdt_a rdseed adx smap clflushopt clwb intel_pt 
                          sha_ni xsaveopt xsavec xgetbv1 xsaves split_lock_detect user_shstk avx_vnni dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp hwp_pkg_req hfi vnmi umi
                          p pku ospke waitpkg gfni vaes vpclmulqdq rdpid movdiri movdir64b fsrm md_clear serialize arch_lbr ibt flush_l1d arch_capabilities
Virtualization features:  
  Virtualization:         VT-x
Caches (sum of all):      
  L1d:                    544 KiB (14 instances)
  L1i:                    704 KiB (14 instances)
  L2:                     11.5 MiB (8 instances)
  L3:                     24 MiB (1 instance)
NUMA:                     
  NUMA node(s):           1
  NUMA node0 CPU(s):      0-19
Vulnerabilities:          
  Gather data sampling:   Not affected
  Itlb multihit:          Not affected
  L1tf:                   Not affected
  Mds:                    Not affected
  Meltdown:               Not affected
  Mmio stale data:        Not affected
  Reg file data sampling: Mitigation; Clear Register File
  Retbleed:               Not affected
  Spec rstack overflow:   Not affected
  Spec store bypass:      Mitigation; Speculative Store Bypass disabled via prctl
  Spectre v1:             Mitigation; usercopy/swapgs barriers and __user pointer sanitization
  Spectre v2:             Mitigation; Enhanced / Automatic IBRS; IBPB conditional; RSB filling; PBRSB-eIBRS SW sequence; BHI BHI_DIS_S
  Srbds:                  Not affected
  Tsx async abort:        Not affected
```

lowland3r, 2024-09-13:
```
Architecture:                         x86_64
CPU op-mode(s):                       32-bit, 64-bit
Address sizes:                        39 bits physical, 48 bits virtual
Byte Order:                           Little Endian
CPU(s):                               4
On-line CPU(s) list:                  0-3
Vendor ID:                            GenuineIntel
BIOS Vendor ID:                       Intel(R) Corporation
Model name:                           Intel(R) Core(TM) i3-8100 CPU @ 3.60GHz
BIOS Model name:                      Intel(R) Core(TM) i3-8100 CPU @ 3.60GHz To Be Filled By O.E.M. CPU @ 3.6GHz
BIOS CPU family:                      206
CPU family:                           6
Model:                                158
Thread(s) per core:                   1
Core(s) per socket:                   4
Socket(s):                            1
Stepping:                             11
CPU(s) scaling MHz:                   100%
CPU max MHz:                          3600.0000
CPU min MHz:                          800.0000
BogoMIPS:                             7200.00
Flags:                                fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault pti ssbd ibrs ibpb stibp tpr_shadow flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm arat pln pts hwp hwp_notify hwp_act_window hwp_epp vnmi flush_l1d
Virtualization:                       VT-x
L1d cache:                            128 KiB (4 instances)
L1i cache:                            128 KiB (4 instances)
L2 cache:                             1 MiB (4 instances)
L3 cache:                             6 MiB (1 instance)
NUMA node(s):                         1
NUMA node0 CPU(s):                    0-3
Vulnerability Gather data sampling:   Vulnerable: No microcode
Vulnerability Itlb multihit:          Processor vulnerable
Vulnerability L1tf:                   Mitigation; PTE Inversion
Vulnerability Mds:                    Vulnerable: Clear CPU buffers attempted, no microcode; SMT disabled
Vulnerability Meltdown:               Mitigation; PTI
Vulnerability Mmio stale data:        Vulnerable: Clear CPU buffers attempted, no microcode; SMT disabled
Vulnerability Reg file data sampling: Not affected
Vulnerability Retbleed:               Mitigation; IBRS
Vulnerability Spec rstack overflow:   Not affected
Vulnerability Spec store bypass:      Mitigation; Speculative Store Bypass disabled via prctl
Vulnerability Spectre v1:             Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:             Mitigation; IBRS, IBPB conditional, STIBP disabled, RSB filling, PBRSB-eIBRS Not affected
Vulnerability Srbds:                  Vulnerable: No microcode
Vulnerability Tsx async abort:        Not affected
```

PartialVolume, 2024-09-13:
That processor sure has a lot of vulnerabilities compared to my more recent I7, it also mentions no microcode in a few places. I don't if that means we are missing some microcode? I thought we included Intel microcode but I'll check on that.

lowland3r, 2024-09-13:
Yeah, I'm starting to wonder if that's related. I just pulled this box out of storage to start wiping old drives. I've applied the latest bios/microcode patches from the vendor and restarted shredos. I'll check back on Monday to see if it did any better.

lowland3r, 2024-09-17:
Hmm. Although updating the BIOS seems to have at least reduced the occurrence of errors, they're still happening:

```
Sep 16 18:04:32 shredos daemon.info ntpd[179]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
Sep 16 18:04:48 shredos kern.warn kernel: general protection fault, probably for non-canonical address 0xffff0881ec598078: 0000 [#1] PREEMPT SMP PTI
Sep 16 18:04:48 shredos kern.warn kernel: CPU: 2 PID: 26 Comm: kworker/2:0H Tainted: G        W        N 6.6.22 #2
Sep 16 18:04:48 shredos kern.warn kernel: Hardware name: Gigabyte Technology Co., Ltd. H310M A 2.0/H310M A 2.0-CF, BIOS F5 11/10/2021
Sep 16 18:04:48 shredos kern.warn kernel: Workqueue: kblockd blk_mq_run_work_fn
Sep 16 18:04:48 shredos kern.warn kernel: RIP: 0010:__blk_rq_map_sg+0x9e/0x470
Sep 16 18:04:48 shredos kern.warn kernel: Code: 28 45 8b 54 24 30 89 5c 24 38 85 ed 0f 84 01 01 00 00 4d 89 e5 44 89 d3 41 89 ec 8b 54 24 38 48 c1 e2 04 49 89 d1 49 03 55 60 <8b> 6a 08 44 8b 42 0c 48 8b 32 29 dd 44 39 e5 41 0f 47 ec 41 01 d8
Sep 16 18:04:48 shredos kern.warn kernel: RSP: 0018:ffffc90000153b88 EFLAGS: 00010282
Sep 16 18:04:48 shredos kern.warn kernel: RAX: ffff8881ab19e2c0 RBX: 0000000000000000 RCX: 0000000000000001
Sep 16 18:04:48 shredos kern.warn kernel: RDX: ffff0881ec598070 RSI: ffffea000578ee40 RDI: 0000000000002000
Sep 16 18:04:48 shredos kern.warn kernel: RBP: 0000000000001000 R08: 0000000000000000 R09: 0000000000010000
Sep 16 18:04:48 shredos kern.warn kernel: R10: 0000000000000000 R11: ffff88812bdd8000 R12: 0000000000001000
Sep 16 18:04:48 shredos kern.warn kernel: R13: ffff8881ec588000 R14: 0000000000001000 R15: ffffc90000153c28
Sep 16 18:04:48 shredos kern.warn kernel: FS:  0000000000000000(0000) GS:ffff88880ed00000(0000) knlGS:0000000000000000
Sep 16 18:04:48 shredos kern.warn kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Sep 16 18:04:48 shredos kern.warn kernel: CR2: 00007effc69e73e0 CR3: 00000003cfc42003 CR4: 00000000003706e0
Sep 16 18:04:48 shredos kern.warn kernel: Call Trace:
Sep 16 18:04:48 shredos kern.warn kernel:  <TASK>
Sep 16 18:04:48 shredos kern.warn kernel:  ? die_addr+0x31/0x80
Sep 16 18:04:48 shredos kern.warn kernel:  ? exc_general_protection+0x19e/0x3a0
Sep 16 18:04:48 shredos kern.warn kernel:  ? asm_exc_general_protection+0x26/0x30
Sep 16 18:04:48 shredos kern.warn kernel:  ? __blk_rq_map_sg+0x9e/0x470
Sep 16 18:04:48 shredos kern.warn kernel:  ? __blk_rq_map_sg+0xf5/0x470
Sep 16 18:04:48 shredos kern.warn kernel:  scsi_alloc_sgtables+0x9f/0x2c0
Sep 16 18:04:48 shredos kern.warn kernel:  sd_init_command+0x12d/0x8a0
Sep 16 18:04:48 shredos kern.warn kernel:  scsi_queue_rq+0x826/0xb70
Sep 16 18:04:48 shredos kern.warn kernel:  blk_mq_dispatch_rq_list+0x306/0x7e0
Sep 16 18:04:48 shredos kern.warn kernel:  ? __sbitmap_weight+0x2f/0x80
Sep 16 18:04:48 shredos kern.warn kernel:  __blk_mq_sched_dispatch_requests+0x3fb/0x5d0
Sep 16 18:04:48 shredos kern.warn kernel:  blk_mq_sched_dispatch_requests+0x32/0x60
Sep 16 18:04:48 shredos kern.warn kernel:  blk_mq_run_work_fn+0x5c/0x70
Sep 16 18:04:48 shredos kern.warn kernel:  process_one_work+0x12c/0x240
Sep 16 18:04:48 shredos kern.warn kernel:  worker_thread+0x2f0/0x410
Sep 16 18:04:48 shredos kern.warn kernel:  ? __pfx_worker_thread+0x10/0x10
Sep 16 18:04:48 shredos kern.warn kernel:  kthread+0xe0/0x110
Sep 16 18:04:48 shredos kern.warn kernel:  ? __pfx_kthread+0x10/0x10
Sep 16 18:04:48 shredos kern.warn kernel:  ret_from_fork+0x2c/0x50
Sep 16 18:04:48 shredos kern.warn kernel:  ? __pfx_kthread+0x10/0x10
Sep 16 18:04:48 shredos kern.warn kernel:  ret_from_fork_asm+0x1b/0x30
Sep 16 18:04:48 shredos kern.warn kernel:  </TASK>
Sep 16 18:04:48 shredos kern.warn kernel: Modules linked in:
Sep 16 18:04:48 shredos kern.warn kernel: ---[ end trace 0000000000000000 ]---
Sep 16 18:04:48 shredos kern.warn kernel: ------------[ cut here ]------------
Sep 16 18:04:48 shredos kern.crit kernel: kernel BUG at include/linux/scatterlist.h:115!
Sep 16 18:04:48 shredos kern.warn kernel: invalid opcode: 0000 [#2] PREEMPT SMP PTI
Sep 16 18:04:48 shredos kern.warn kernel: CPU: 0 PID: 589 Comm: nwipe Tainted: G      D W        N 6.6.22 #2
Sep 16 18:04:48 shredos kern.warn kernel: Hardware name: Gigabyte Technology Co., Ltd. H310M A 2.0/H310M A 2.0-CF, BIOS F5 11/10/2021
Sep 16 18:04:48 shredos kern.warn kernel: RIP: 0010:__blk_rq_map_sg+0x314/0x470
Sep 16 18:04:48 shredos kern.warn kernel: Code: 24 e8 b0 5e 04 00 4c 8b 5c 24 10 44 8b 44 24 08 4c 8b 14 24 48 89 c7 49 89 3a 48 8b 07 83 e0 03 41 f6 c5 03 0f 84 3c ff ff ff <0f> 0b 8b 86 a0 00 00 00 8b 96 a4 00 00 00 48 8b ae 98 00 00 00 48
Sep 16 18:04:48 shredos kern.warn kernel: RSP: 0018:ffffc9000219f7b0 EFLAGS: 00010202
Sep 16 18:04:48 shredos kern.warn kernel: RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000
Sep 16 18:04:48 shredos kern.warn kernel: RDX: 0000000000000000 RSI: 000000002b05e749 RDI: ffff8881731a8040
Sep 16 18:04:48 shredos kern.warn kernel: RBP: 0000000000001000 R08: 0000000000000749 R09: 0000000000080000
Sep 16 18:04:48 shredos kern.warn kernel: R10: ffffc9000219f850 R11: ffff88812bdd8000 R12: 0000000000001000
Sep 16 18:04:48 shredos kern.warn kernel: R13: bbf3ef2ceb93687d R14: 0000000000000000 R15: bbf3ef2ceae750fd
Sep 16 18:04:48 shredos kern.warn kernel: FS:  00007f4bdc8486c0(0000) GS:ffff88880ec00000(0000) knlGS:0000000000000000
Sep 16 18:04:48 shredos kern.warn kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[... cut ...]
Sep 16 18:04:48 shredos kern.warn kernel: R13: ffff8881ec588000 R14: 0000000000001000 R15: ffffc90000153c28
Sep 16 18:04:48 shredos kern.warn kernel: FS:  00007f4bdc8486c0(0000) GS:ffff88880ec00000(0000) knlGS:0000000000000000
Sep 16 18:04:48 shredos kern.warn kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Sep 16 18:04:48 shredos kern.warn kernel: CR2: 000055722b713420 CR3: 000000012ccaa001 CR4: 00000000003706f0
Sep 16 18:04:48 shredos kern.warn kernel: ------------[ cut here ]------------
```

PartialVolume, 2024-09-17:
What's the history of this machine?.

Bearing in mind ShredOS will absolutely make full use of all available memory I would run an overnight memtest64 as you may have a bad memory that might not show up if you just load an ordinary live DVD etc.

lowland3r, 2024-09-17:
Yep, it was bad RAM. Come to think of it, that probably explains a few problems I've had with this machine. Took one of the 16 GB sticks out and the other tested all clear. Just wiped three drives simultaneously without issue.

closed 2024-09-17 by lowland3r

PartialVolume, 2024-09-17:
That's great, glad you got it sorted.