PartialVolume/shredos.x86_64 #291: Issues with ShredOS Stalling During Gutmann Wipe

amiina-png, 2024-09-27:
Hello !
I’m currently facing an issue with ShredOS while trying to perform a Gutmann wipe on a hard drive. 
[image]
The process starts, but it stalls at around 2.27 without showing any error messages.

PartialVolume, 2024-09-27:
Could be a hardware problem. Before you start the wipe try switching to a virtual terminal ALT F2 and run `smartctl -a /dev/sda | more`

What does it say about the health of the drive?

How many reallocated sectors does the drive have?

Does it also hang if you only do a PRNG wipe, no blanking pass ( no second zeros pass), verification set to last pass only.

The gutmann method is a bit extreme in terms of being unnecessary on a modern non MFM drive, however one good thing going for it is that it really tests your hardware as it takes so long to complete. 70 hours in your case.

Your drive temperature looks ok, however that doesn't mean there isn't something wrong with the drive.

What's the history and age of this machine?

Also, just for reference while you are in the virtual terminal can you post the output of 'lspci -k` so I can see what hardware you have. Thanks. 

ALT F1 to get back to nwipe after you are done in the virtual terminal.

amiina-png, 2024-09-27:
Thank you for your help !
I am currently testing the PRNG method, and it seems to be working. It looks like the process will take around 3 hours and 40 minutes.
However, I still have to use the Gutmann wipe method in the end

[image]
[image]

amiina-png, 2024-09-27:
And now the PRNG method has stopped as well 
[image]

PartialVolume, 2024-09-27:
Not that it matters, as it has nothing to do with the problem but you didn't switch off the blanking pass. With the settings you chose, prng + blanking + last pass verification. It would have written the first prng pass by the time it reached 33% and nearly completed the second blanking (zeros) pass which would have completed at 66% (but never made it). Verification would have been the last 33%. So any data on the drive is gone with zero I/O errors upto the point it froze.

Next thing to rule it is that Intel i915 DRM video driver. There have been a load of bugs in this particular kernel in relation to video drivers.

There are two ways to determine whether the video driver is the cause. The first way is more involved and requires you have the computer on a network that you login to ShredOS via telnet and run nwipe in a terminal from a different computer on the network.

1. See [headless logins](https://github.com/PartialVolume/shredos.x86_64#how-to-wipe-drives-on-headless-systems-or-systems-with-faulty-display-hardware-for-use-on-secure-lans-only) 

1. Disable DRM video drivers and revert to simple framebuffer mode. This is the easier option as all you need to do is edit the two grub.cfg files on the USB stick,  /boot/grub/grub.cfg and /EFI/BOOT/grub.cfg by appending `nomodset` as shown in the example below. Once you have edited the files, boot ShredOS.  If you edited the grub files correctly running `lspci -k` should show the VGA compatible controller with the  driver in use line missing 
[image]

If no driver in use run the wipe. If it works it's a buggy DRM driver which hopefully will be fixed in the next release when we update the kernel and all the video drivers.

Example showing nomodeset appended to the kernel command line.
```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 nomodeset
}
```

PartialVolume, 2024-09-27:
Although its most likely a buggy i915 driver or a hardware related problem, I just want to confirm you are not inadvertently typing CONTROL S at any point during the wipe. CONTROL S will freeze the screen, while CONTROL Q will restore activity.