PartialVolume/shredos.x86_64 #292: ShredOS wipes hang, System Rescue distro works?

funk0id, 2024-10-01:
I've had persistent issues with various systems freezing when running nwipe via the current ShredOS download. 

Weirdly, on the same machines, with the same parameters, things run through without issues using the System Recovery live image which currently has nwipe-0.37?

It's a workaround for now, but ideally I want a workflow that boots, formats and uploads the output pdf to a common location.

PartialVolume, 2024-10-01:
Have you tried adding nomodeset to the kernel command line in both grub.cfg files on the USB drive? This disables the DRM video drivers and uses framebuffer mode rather than the video drivers.

If it works without freezing then the video driver is the problem. Please let me know whether it solves the problem or not.

```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 nomodeset
}
```

PartialVolume, 2024-10-02:
Can you also post the output of `lspci -k` so I can see what hardware and drivers are in use. Thanks.

NMExact, 2024-11-22:
> Have you tried adding nomodeset to the kernel command line in both grub.cfg files on the USB drive? This disables the DRM video drivers and uses framebuffer mode rather than the video drivers.
>
> If [...]

Tried this but doesn't seem to work I am trying to boot form a HP server however it gets stuck on 100% and just stays on that screen and nothing happens.

PartialVolume, 2024-11-23:
> Tried this but doesn't seem to work I am trying to boot form a HP server however it gets stuck on 100% and just stays on that screen and nothing happens.

Assuming you are booting from a usb stick or if not, plug any old usb stick in, does dmesg.txt get written to the USB stick. If it does can you post it's content. Thanks. If the dmesg.txt file exists it means ShredOS is running and it's a display issue. The content of dmesg.txt may give a clue as to why the graphics is not working.

NMExact, 2024-11-27:
> Assuming you are booting from a usb stick or if not, plug any old usb stick in, does dmesg.txt get written to the USB stick? If it does can you post it's content. Thanks. If the dmesg.txt file [...]

Hi thanks for getting back to me, It doesn't seem to. All I have on the usb sick are the following.

boot (Folder)
EFI (Folder)
autorun.inf
README.txt
Shredos.ico

I am try on a hp server HPE ProLiant DL360 Gen9 to try and wipe some old SAS drives.