PartialVolume/shredos.x86_64 #298: Kernel panic - not synching: write error on Dell PowerEdge Server SC1420

thracian2015, 2024-11-04:
Any idea how to fix this? ShredOS does through 100% loading and then crashes with this error (see full stack attached).
[image]

PartialVolume, 2024-11-04:
Could be a corrupt kernel image, maybe bad memory? I would run memtest, burn a new flash drive.

Are you booting from flash drive or over the network?

jbgross, 2025-07-01:
I had this exact same problem. I was trying to wipe my old ASUS eee PC. I tried different USBs and memtest passed just fine. 

I ended up using another tool, grml. 

I don't need help, but if someone else finds this, they may find it useful to try grml and just manually use shred from the command line.

PartialVolume, 2025-07-01:
I do have an old EEEPC 901 somewhere. I've never tried booting ShredOS on a Intel Atom running on an EEEPC so will give it a go & see what the issue is. First thought is that it could be lack of CPU RAM especially if below 1GB.

What model of EEEPC do you have, 701, 900, 901, 1000H, 1000HA, 1000HE, 1002HA, 1005HA, 1008HA or 1101HA

MichaelWarnecke, 2025-12-30:
I ran into the same problem with a fairly old device with 1GB of RAM. Using a rather old version ([2020.02.005_x86-64_0.30.001](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2020.02.005_x86-64_0.30.001)) shredOS started correctly. I didn’t check which version began to require more than 1 GB of RAM, though. Newer versions than the one mentioned above might work as well.