PartialVolume/shredos.x86_64 #302: lost IP address

ehn29, 2024-11-27:
Hi

Do you have any idea why when loading Shredos via an IPXE boot, I lose the network connection.
My client station responds to the ping as long as I am on the Ipxe boot menu.
This problem appears when my client is connected via a network adapter on a USB-C port.
If I use the internal network port, no loss of network.
And therefore at the end of the deletion, no more network to operate the LFTP transfer.

if I manually reconfigure an IP address from the console with ifconfig -a eth0 xxx.xxx.xxx.xxx, I can transfer the log to my server via FTP

thanks

PartialVolume, 2024-12-01:
If you plug a FAT formatted USB stick into the system & boot via IPXE, when ShredOS boots it will write the dmesg.txt file to the root of the USB stick. It also writes it again when you Control C to exit nwipe which will probably show more information. Maybe dmesg.txt will show why the USB network fails.

However, having said that, ShredOS will automatically attempt to get a DHCP IP address whenever the network lead is plugged in or out.

When the USB network is connected what's the output of ifconfig? The reason I ask is that ShredOS will only work with certain network device names, if I remember correctly. `ethx` being one and maybe `enpxxxx` being another device name. If your USB network has a name that differs from ethx or enpxx then the network scripts will fail to obtain a dynamic address from the DHCP server. If this is the case it is a fairly easy fix. I just need to know the first few characters of the network device name.

I've only tested ShredOS networking with DHCP so I've no idea how it will work with static IPs as the networking scripts will presumably still try to obtain a dynamic IP address.

ehn29, 2024-12-02:
la carte réseau semble bien être vue en eth0.

Voici le fichier dmesg.txt complet.
cela sera peut-être plus parlant pour toi
[dmesg.txt](https://github.com/user-attachments/files/17977057/dmesg.txt)

PartialVolume, 2024-12-04:
Can you post the output of a file called nohup.out. You will find it in the / directory in virtual memory not on the USB stick. This file is a log of network DHCP request. A working file should look something like this:
```
[OK] eth0 is down
[OK] eth1 is down
[OK] eth2 is down
more: read error: Invalid argument
udhcpc: started, v1.36.1
udhcpc: broadcasting discover
udhcpc: no lease, forking to background
[OK] eth0 is up
```

ehn29, 2024-12-05:
I have these messages but right after I have this one

[OK] eth0 is down

result of tail -f nohup.out

[image]
[nohup.out.log](https://github.com/user-attachments/files/18020727/nohup.out.log)

PartialVolume, 2024-12-11:
Have you got a working DHCP router or server on your network or is everything static?

PartialVolume, 2024-12-11:
I think you said it worked ok from the internal ethernet port it's just the USB ethernet adapter that's the problem so I'm guessing your DHCP server/router is fine.

PartialVolume, 2024-12-11:
According to nohub it looks like it's broadcasting a DHCP request but not getting a response, it then initialises the network port and trys again and again. 

What's the make/model of the USB adapter?

ehn29, 2024-12-11:
I have a dhcp server and it works.I tried with DA200, DA310, WD15 and WD19 adapters from Dell. The problem occurs for these 4 adapters and on several computer models

PartialVolume, 2024-12-11:
I'll order up a DA200 and get back to you.

PartialVolume, 2024-12-13:
Received the DA200 today and I can see the issue. Basically the network is initialised prior to the DA200 being available which is causing the USB ethernet to be missing from /etc/network/interfaces.

I'll work out the best way to resolve this while taking into account the USB wifi interface could be unplugged and plugged back in while ShredOS is running so needs to cope with such a scenario.

ehn29, 2025-01-28:
Hi

thanks for this update.
it works perfectly

have a nice day

closed 2025-01-28 by ehn29

PartialVolume, 2025-01-28:
Glad it works, however I still need to apply some updates to the code for hot plugging these sort of ethernet devices. Don't be surprised if this device is not seen if you plug it in after the system is already booted rather than the device being present at power up.

I'll get a permanent fix for this in the next release. I've reopened so I don't forget about it.

PartialVolume, 2026-02-26:
We're probably missing a driver for that device. Can you boot up shredOS, then  ALT F2 to switch to the terminal, makesure the HP USB/ethernet device is plugged in and type lsusb and post the output here. Thanks.

PartialVolume, 2026-04-08:
@Runningmedic Again with the USB ethernet adapter plugged in, boot up ShredOS, then ALT F2 to the virtual terminal and type `lshw -C network` and post the output here. This will tell us what driver is attached. Thanks.

PartialVolume, 2026-04-10:
Can you also post the contents of nohup.out i.e `more nohup.out` It should look something like this..

```
[OK] eth0 is down
udhcpc: started, v1.37.0
udhcpc: broadcasting discover 
udhcpc: no lease, forking to background
[OK] eth0 is up
```

Runningmedic, 2026-04-24:
Even though I got it working it was inconsistent at best. I ended compiling shredos and editing the kernel. In linux-menuconfig/network support/networking options - I included everything that started with IP. In linux-menuconfig/general setup - I included boot config support and force unconditional bootconfig processing. Now it recognizes the dongle right away with the following config

`set default="0"
set timeout="0"

menuentry "ShredOS" {
    linux /boot/bzImage console=tty3 loglevel=3 ip=:::::eth0:dhcp nwipe_options="--method=dod3pass --nousb --autonuke --verify=last" shredos_output="tftp:000.000.000.000:/"
}
`

Thank you again for your amazing notes! Even someone as new as me could follow.