PartialVolume/shredos.x86_64 #303: eMMC drive not detect

ntxg, 2024-12-01:
hi my eMMC drive on my Beelink Z83II x5-Z8350 2GB DDR3 32GB EMMC not get detected by shredos to wype

PartialVolume, 2024-12-01:
It would be useful if you could boot shredOS and once nwipe appears, ALT F2 to switch to the virtual terminal and post `lspci -k` so I can see what hardware is present and what drivers are loaded. Thanks.

PartialVolume, 2024-12-01:
There are quite a few eMMC, SD/MMC drivers that I don't have installed in ShredOS but are available so once I know what controller you have (from the output of `lspci -k`), I can make sure I load a driver for it in the next version.

paddymccdrc, 2026-09-25:
Did you ever get a resolution to this?

I've the same issue woth some Igel M350c thin clients and exact same output as the user above?

PartialVolume, 2026-09-25:
No, not so far, I need to identify the actual eMMC device. Can you boot up ShredOS and provide the output of 
`ls -l /sys/bus/acpi/devices/*/driver`

The previous post with the lspci -k did not show what I needed as eMMC devices are not PCI devices but are registered by the acpi. The above `ls` command should hopefully show the acpi devices even though a driver is missing.

PartialVolume, 2026-09-25:
@paddymccdrc Apparently the Igel M350c thin clients run a AMD Ryzen embedded R1505G CPU. It's on board eMMC host controller is routed through the AMD SoC's Secure Digital Interface  (SDHCI) using ACPI enumeration. It appears I have all the necessary configs loaded except for one that may or may not be required that's to do with power management quirks. CONFIG_AMD_PMC.

I can enable this and build you a .iso and/or .img if you want to test it? Or if you routinely build from source just add that to the build. Let me know if you want me to build it. It will take a couple of hours.

Solverz-0, 2026-10-14:
> 6\. [@paddymccdrc](https://github.com/paddymccdrc) Apparently the Igel M350c thin clients run a AMD Ryzen embedded R1505G CPU. It's on board eMMC host controller is routed through the AMD SoC's Secure Digital Interface  (SDHCI) using ACPI enumeration. It appears I have all the necessary configs loaded except for one that may or may not be required that's to do with power management quirks. CONFIG_AMD_PMC.
>         I can enable this and build you a .iso and/or .img if you want to test it? Or if you routinely build from source just add that to the build. Let me know if you want me to build it. It will take a couple of hours.

Hello, I have the same issue, the eMMC device is not detected in my DELL Wyse 3040. Please let me know which outputs you need and I will get them over to you!

PartialVolume, 2026-10-14:
> 7\. > 6. [@paddymccdrc](https://github.com/paddymccdrc) Apparently the Igel M350c thin clients run a AMD Ryzen embedded R1505G CPU. It's on board eMMC host controller is routed through the AMD SoC's Secure Digital Interface  (SDHCI) using ACPI enumeration. It appears I have all the necessary configs loaded except for one that may or may not be required that's to do with power management quirks. CONFIG_AMD_PMC.
>     > I can enable this and build you a .iso and/or .img if you want to test it? Or if you routinely build from source just add that to the build. Let me know if you want me to build it. It will take a couple of hours.
>     
>     
>     Hello, I have the same issue, the eMMC device is not detected in my DELL Wyse 3040. Please let me know which outputs you need and I will get them over to you!

I have a Dell Wyse 3040, I'll check it out.

PartialVolume, 2026-10-14:
> 8\. > 7. > 6. [@paddymccdrc](https://github.com/paddymccdrc) Apparently the Igel M350c thin clients run a AMD Ryzen embedded R1505G CPU. It's on board eMMC host controller is routed through the AMD SoC's Secure Digital Interface  (SDHCI) using ACPI enumeration. It appears I have all the necessary configs loaded except for one that may or may not be required that's to do with power management quirks. CONFIG_AMD_PMC.
>     > > I can enable this and build you a .iso and/or .img if you want to test it? Or if you routinely build from source just add that to the build. Let me know if you want me to build it. It will take a couple of hours.
>     > ```
>     > Hello, I have the same issue, the eMMC device is not detected in my DELL Wyse 3040. Please let me know which outputs you need and I will get them over to you!
>     > ```
>     
>     
>     I have a Dell Wyse 3040, I'll check it out.

Confirmed. eMMC not seen. I'll check it out but it looks like the `Intel Low Power Subsystem Support` is required. All the other MMC drivers are installed, I'm currently building with that support enabled, I'll let you know if that solves it.

Solverz-0, 2026-10-14:
Thank you very much!

paddymccdrc, 2026-10-14:
@PartialVolume sorry for being dumb but im using Igel M250c which have Intel Celeron CPU N3160.

When i run ls -l /sys/bus/acpi/devices/*/driver

I get the attached output:

sh-5.2# ls -l
total 156
lrwxrwxrwx   1 root root      7 Jan 23  2025 bin -> usr/bin
drwxr-xr-x   3 root root     60 Jan 23  2025 boot
drwxr-xr-x   2 root root     40 Oct 14 16:51 boot_tmp
----------   1 root root      0 Oct 14 16:51 crond.reboot
drwxr-xr-x   7 root root  12980 Oct 14 16:51 dev
-rw-r--r--   1 root root 116686 Oct 14 16:51 dmesg.txt
drwxr-xr-x  22 root root   1080 Oct 14 16:51 etc
drwxr-xr-x   2 root root     60 Oct 14 16:51 exported
drwxr-xr-x   2 root root     80 Oct 14 16:51 imported
-rwxr-xr-x   1 root root    462 Jan 23  2025 init
lrwxrwxrwx   1 root root      7 Jan 23  2025 lib -> usr/lib
lrwxrwxrwx   1 root root      3 Jan 23  2025 lib64 -> lib
-rwxr-xr-x   1 root root  14256 Jan 23  2025 linuxrc
drwxr-xr-x   2 root root     40 Dec  8  2024 media
drwxr-xr-x   2 root root     40 Dec  8  2024 mmt
-rw-------   1 root root    164 Oct 14 16:51 nohup.out
drwxr-xr-x   2 root root     40 Dec  8  2024 opt
dr-xr-xr-x 186 root root      0 Oct 14 16:51 proc
drux------   2 root root     40 Dec  8  2024 root
drwxr-xr-x   4 root root    400 Oct 14 16:51 run
lrwxrwxrwx   1 root root      8 Jan 23  2025 sbin -> usr/sbin
drwxr-xr-x   6 root root    120 Jan 23  2025 share
dr-xr-xr-x  12 root root      0 Oct 14 16:51 sys
-rw-r--r--   1 root root    196 Oct 14 16:51 tftp_status.txt
drwxrwxrwt   3 root root    100 Oct 14 16:51 tmp
-rw-r--r--   1 root root   5121 Oct 14 16:51 transfer.log
drwxr-xr-x   7 root root    160 Jan 23  2025 usr
drwxr-xr-x   5 root root    220 Jan 23  2025 var
-rw-r--r--   1 root root     31 Oct 14 16:51 wget.log
sh-5.2# ls -l /sys/bus/acpi/devices/*/driver
lrwxrwxrwx 1 root root 0 Oct 14 16:52 /sys/bus/acpi/devices/LNXPWRBN:00/driver -> ../../../bus/acpi/drivers/button
lrwxrwxrwx 1 root root 0 Oct 14 16:52 /sys/bus/acpi/devices/LNXTHERM:00/driver -> ../../../bus/acpi/drivers/thermal
lrwxrwxrwx 1 root root 0 Oct 14 16:52 /sys/bus/acpi/devices/LNXVIDEO:00/driver -> ../../../bus/acpi/drivers/video
lrwxrwxrwx 1 root root 0 Oct 14 16:52 /sys/bus/acpi/devices/PNP0C0C:01/driver -> ../../../bus/acpi/drivers/button
lrwxrwxrwx 1 root root 0 Oct 14 16:52 /sys/bus/acpi/devices/PNP0C0E:00/driver -> ../../../bus/acpi/drivers/button
sh-5.2#

[image]
[image]

PartialVolume, 2026-10-14:
> 10\. Thank you very much!

Adding the `Intel Low Power Subsystem Support` did the trick. I can bump the revision to 0.43.2 and put this .iso on Google drive if you want to test it out on your system. Because we only just released an official version (32_0.43) we won't be releasing 33_0.44 for another 3- 6 months so if you need this functionality now this is your chance, I'll post it here anyway just in case somebody else also has the same issue. The change will be committed to ShredOS this evening.

[image]

PartialVolume, 2026-10-14:
@paddymccdrc Can you also try the .iso I'm going to post in the next hour or so in this thread, fingers crossed it will also fix your system.

PartialVolume, 2026-10-14:
These are the shredos_lite_defconfig versions, this is the version that's fully functional except for the 100MB of AMD drivers that are not included, so it can boot in 512MB. Although the Dell Wyse 3040 has 2GB of memory that's actually not enough now for the ShredOS full version with all the AMD drivers.

[shredos-2025.11_32.1_x86-64_v0.43.1_20261014.img](https://drive.google.com/file/d/11p6zGp9rnj4mIC0_-wz1FAf-2Hom9z5d/view?usp=sharing)

[shredos-2025.11_32.1_x86-64_v0.43.1_20261014.iso](https://drive.google.com/file/d/1Izl615M4QinWyDV75Bgh8RbUzK9MsACo/view?usp=sharing)

#### checksums
sha1sum 8cd61f29e71be38dfd6bfe73bea649ccb40bdb0c  shredos-2025.11_32.1_x86-64_v0.43.1_20261014.img
sha1sum 23c580511f8eec6fdf02f73b5ce094450ee9772d  shredos-2025.11_32.1_x86-64_v0.43.1_20261014.iso

Solverz-0, 2026-10-14:
[image]

Okay it detects it now using your image (thank you!), but for some reason it sees it as a 2GB device instead of 8GB. There is no such thing as a 2GB Dell Wyse 3040, so this is strange. The wipe also fails.

PartialVolume, 2026-10-14:
Those mmcblkboot0 and 1 are not usually accessible like a normal disc but the mmcblk1 should normally be capable of being wiped. mmcblk1 may just be worn out. However you might still be able to read it. ALT-F2 then hexedit /dev/mmcblk1, does that show anything? Its odd your Wyse 3040 has those boot partitions called mmcblkboot0 where on my Wyse 3040 they are lk1boot0 and lk1boot1. Maybe they are running different firmware. You could try a zeros write, no blanking. If that fails try a ones write, no blanking to see if one or the other works. While you are in that terminal can you run `lsblk` and see if lsblk also says it's 2GB.

PartialVolume, 2026-10-14:
If you reboot and press F2 when the DELL logo appears at the bottom under maintenance there is an option to wipe the internal eMMC. It maybe worth trying that and seeing if it comes up with any errors. I'm just wiping the eMMC on mine via the bios, then I'll run a PRNG wipe on mine and let you know how it goes.

PartialVolume, 2026-10-14:
The discrepancy between your eMMC model name (M MOR ) and mine (H8G4A) seems to explain the problem. (According to Gemini)

The presence of M MOR vs. H8G4A as the Product Name string in the eMMC Card Identification (CID) register explains the issue.

This behavior points to eMMC chip failure/firmware panic mode.

**Understanding the Model Strings:**
H8G4A (SK Hynix 8GB eMMC): (On my Wyse 3040)
This is the normal Product Name (PNM) returned by a healthy SK Hynix 8GB eMMC module (such as the SK Hynix H26M41203HPR / H8G4a series). When fully operational, the internal controller reports its exact hardware ID and full capacity to the host system.

M MOR (Hynix Controller Fallback / Boot ROM State):M MOR is a specific fallback product string exposed by SK Hynix eMMC internal memory controllers when the chip enters a ROM boot-mode / emergency state.   What "M MOR" Means for the UnitWhen an SK Hynix eMMC chip suffers severe block degradation, worn-out NAND flash cells, or corrupted controller microcode, the onboard memory controller fails to load its main firmware from the internal flash array.

When this happens:The controller drops into a bare-metal Boot Loader / Diagnostic Mode. It overrides the normal CID register string and reports the product name as M MOR (often with a Serial Number of 00000000 and Revision 00).   The host system cannot access the main flash partition (ROM1 User Area) and only sees a minimal stub or default addressable space (frequently truncation down to **2GB**, 128MB, or raw boot hardware partitions).

**Diagnostic Verification**
You can confirm this at the OS level (e.g., from a Linux live boot) by inspecting the card identification registers in sysfs:

#### Check product name
`cat /sys/block/mmcblk0/device/name`

#### Check serial number and revision
`cat /sys/block/mmcblk0/device/serial`
`cat /sys/block/mmcblk0/device/hwrev`

If /sys/block/mmcblk0/device/name reads M MOR and the serial number returns 0x00000000, the eMMC memory controller has permanently locked into its failure/panic state.

**Conclusion & Resolution OptionsHardware Failure:** The 2GB Wyse 3040 is not a different factory SKU; it is an 8GB unit whose SK Hynix eMMC chip has reached end-of-life or suffered unrecoverable internal controller corruption.

**BGA Replacement Required:** Because eMMC memory on the Wyse 3040 is soldered directly to the system board (BGA), software tools cannot re-flash or recover an SK Hynix chip once it has dropped into the M MOR state. 

**Workaround:** If you want to continue using that specific Wyse 3040 board, the most practical solution is to run your operating system directly from an external USB 3.0 drive or set up a PXE network boot, ignoring the soldered internal eMMC entirely.

Solverz-0, 2026-10-14:
hexedit shows a bunch of hex values
lsblk reports 1.8GB
The zeros and ones writes fail
The wipe in the bios fails too
name shows M MOR
serial 0x00000615
hwrev 0x0

So yeah, looks like the emmc chip is toast, sadly. Thank you for testing on your system!

PartialVolume, 2026-10-14:
Just for reference, I successfully wiped the 8GB eMMc on my DELL Wyse 3040. Interesting speed profile. I also wiped it using the Bios wipe on boot option. Unfortunately a replacement  H26M41203HPR if you can get one is about $20 for one. You can buy a 3040 on ebay for about the same price.

[image]

[image]

[image]

[nwipe_report_2018-08-20-16-23-39_Model_MMC_H8G4a__Serial__S_N__unknown__device_mmcblk1.pdf](https://github.com/user-attachments/files/33460073/nwipe_report_2018-08-20-16-23-39_Model_MMC_H8G4a__Serial__S_N__unknown__device_mmcblk1.pdf)

Solverz-0, 2026-10-14:
Thanks for that, that is interesting indeed.

Yeah so I did have a look, I cannot find a replacement chip with an identical part number even on aliexpress. But if that is what they are going for, no point cost wise to bother replacing it IMHO. :(

paddymccdrc, 2026-10-15:
@PartialVolume that has worked however we have the issue again where the drivers for the network adapter are not talking to the server and taking the customer file and PDFs back / sending to server. I think this was an AMD driver issue before when we had it with HP laptops?

PartialVolume, 2026-10-15:
@paddymccdrc Is that a specific system that doesn't work that used to work?

paddymccdrc, 2026-10-15:
@PartialVolume its specifically on this Igel M250 loading via PXE.

We load ShredOS via our PXE Boot menu. We import the customer file & send the PDFs back to the server via TFTP.

We had this issue in the past with some HP machines. Its a strange one as the network is obviously working until it gets to the handover to the ShredOS and it fails to connect to our server at 192.168.... and falls back to default settings.

paddymccdrc, 2026-10-15:
@PartialVolume i thought i would try the version of Shredos you configured before with some AMD drivers for us and try wiping a USB drive via the Igel device and this one is loading the customer file, connecting to our server via TFTP.

So it looks like this version from earlier in the thread is missing that driver:

[shredos-2025.11_32.1_x86-64_v0.43.1_20261014.iso](https://drive.google.com/file/d/1Izl615M4QinWyDV75Bgh8RbUzK9MsACo/view?usp=sharing)

PartialVolume, 2026-10-15:
Can you let me know the output of lspci -k.

 I'm assuming there is some difference between the defconfigs. You don't know the original filename that you downloaded that worked?  I can then compare the defconfigs and see what has changed.

PartialVolume, 2026-10-15:
Added missing support for Intel Low Power Subsystem Support for exposure of eMMC drives on various systems. #528

PartialVolume, 2026-10-15:
> 25\. [@PartialVolume](https://github.com/PartialVolume) i thought i would try the version of Shredos you configured before with some AMD drivers for us and try wiping a USB drive via the Igel device and this one is loading the customer file, connecting to our server via TFTP.
>      So it looks like this version from earlier in the thread is missing that driver:
>      [shredos-2025.11_32.1_x86-64_v0.43.1_20261014.iso](https://drive.google.com/file/d/1Izl615M4QinWyDV75Bgh8RbUzK9MsACo/view?usp=sharing)

It could be this firmware we are missing from the lite versions. LINUX_FIRMWARE_MICROCHIP_VSC85XX_PHY which is a physical-layer network driver designed to support Microchip (formerly Vitesse and Microsemi) VSC85xx family Gigabit Ethernet transceivers, but depends whether that's your hardware.

But it would be useful to know which .iso or .img worked so I can compare the configs.

PartialVolume, 2026-10-16:
Added firmware for ethernet devices. #529