PartialVolume/shredos.x86_64 #304: Freezing long before completion on all wipes.

thomasZimmerman0, 2024-12-09:
Trying to wipe 3 storage devices to re-sell a computer. Between 10-30 minutes the operation completely fails. Here's some things I've tried.

1. Updating BIOS to latest version.
2. Created bootable USB with Rufus using latest release .img release 
3. Used Ventoy with .img, multiple iso's, and previous release.

Anyone have any idea what might be causing this?

PartialVolume, 2024-12-09:
Have you tried using the nomodeset .iso or adding nomodeset to the kernel command line in both grub files? This disables the DRM graphics drivers and if it then works identifies the DRM driver for whatever graphics hardware you have as the cause of the hang.

thomasZimmerman0, 2024-12-10:
> Have you tried using the nomodeset .iso or adding nomodeset to the kernel command line in both grub files? This disables the DRM graphics drivers and if it then works identifies the DRM driver for whatever graphics hardware you have as the cause of the hang.

Yes I have tried both of those things separately. I do have a bit of an **UPDATE**. I was trying to wipe 3 drives at once, and after running into this problem over and over I tried to give wiping them one at a time a try. One of them finally made it all the way through a DOD short wipe, another is getting close to the same right now, but on the last one it crashed at about 87% completion which is making me think it's something with this drive in particular.

PartialVolume, 2024-12-11:
That makes me wonder whether you have a CPU memory fault. I'm routinely running eight disc wipes with no freezing.

Firminator, 2024-12-11:
Or the laptop or drives are overheating?

thomasZimmerman0, 2025-01-09:
Sorry for the long wait time for this issue. I've come to the conclusion that it was in fact the one drive causing the issue, probably something to do with its age (It's 8 years old). I would like to add though I haven't had any problems with it for any other purpose; temperatures are normal and it's quite cold where I am anyhow. 

Regardless, I made it through enough passes on all of the drives to be satisfied so I'm closing the issue. Thank you all for your suggestions.

closed 2025-01-09 by thomasZimmerman0

Firminator, 2025-01-10:
I was test driving the new version of SeaTools for Linux ( https://www.seagate.com/support/downloads/seatools/ ) recently, and they do have a 'Fix' option in there which is 'fixing' bad sectors by remapping them. After all bad sectors have been remapped the drive can then be wiped without ShredOS/nwipe erroring out with the -1 error or just freezing like what you experienced. That was my workaround to get a  drive wiped which was erroring out prior.
The advandage is that this is basically on-demand remapping and you don't have to wait for the remapping to occur at whatever time it will occur :) 
There is also a Windows version.