PartialVolume/shredos.x86_64 #305: Error when I try shredos with SSD

alfb29, 2024-12-11:
Hello
Shredos is install on a hdd ventoy. I boot on ventoy and use shredos to erase a 256Go SSD.
I tried many time and always have an error. 
Can you help me to understand what is the problem?
[nwipe_log_20241211-134028.txt](https://github.com/user-attachments/files/18100075/nwipe_log_20241211-134028.txt)
[nwipe_report_2024-12-11-13-49-11_Model_SPCC_Solid_State_Serial_AA230216S3025610321.pdf](https://github.com/user-attachments/files/18100076/nwipe_report_2024-12-11-13-49-11_Model_SPCC_Solid_State_Serial_AA230216S3025610321.pdf)

Thanks

PartialVolume, 2024-12-11:
Looks like your SSD is faulty. With the method you have selected which happens to be the default, 4 complete reads or writes will be performed. So the first 25% will be a random write, 26-50% will be verification of that random write, which is when you get a pass and sync error.

Because it failed at just over 28% it's very likely all the data on the disc has been overwritten. You could check this with a hex editor. However it failed on prng verification indicating a fault on the drive. It could be a single stuck bit in a memory location. Instead of prng you could do a zeros wipe and verification (blanking pass off), if it fails you might have a stuck bit high, in which case a one's pass and verification may pass. Irrespective there is something wrong with you drive.

PartialVolume, 2024-12-11:
Equally it could also me a CPU memory fault however this might possibly show as an error in the dmesg.txt file.

The SSD looks like it's been powered up for just under six months. What's the history of this SSD and the PC/laptop it's plugged into?

PartialVolume, 2024-12-11:
However from anecdotal evidence on Reddit there is a certain amount of negativity regarding the reliability of SPCC (Silicon Power) drives. So I'd bet on the drive being faulty. Anybody else had experience of SPCC drives and failures within 6 months?

PartialVolume, 2024-12-12:
Method: zeros, verification last pass, blanking off. Will write zeros to the disc and read it back to verify it is indeed zeros.

Post the certificate so I can read it.

Then do
Method: ones, verification last pass, blanking off. Will write ones to the disc and read it back to verify it is indeed ones.

Post the certificate

If it fails on either I wouldn't be installing anything on it. I would scrap the drive and buy a new one, preferably a drive with a good reputation for reliability.

Have you ever used the hex reader on ShredOS?

It allows you to view blocks of the disc as hexadecimal and ASCII. ALT F2 to switch to a virtual terminal, type hexedit /dev/sda or whatever the drive name is. It will show the first block, use page down/up to move through disc. Shift > to move to the last block, shift < to go back to the start. After the ones wipe you should see all FFFF everywhere. FFFF is hexadecimal for all binary ones. If you see all FFFF in the last block you can be pretty confident your disc is wiped. If verification passes you can be 100% confident. However, like I said any failures whichever method you use indicates a bad drive. Scrap it. It's really not worth the time loading a OS on a suspect disc, you could end up with all sorts of weird and obscure problems with the OS or programs you run.

ShredOS/nwipe is a pretty good way of validating your hardware because it does write and read every block on the disc and also uses all available CPU memory and tends to warm things up too.

There are alternate reasons for using other  procedures (firmware based) for wiping SSDs securely discussed elsewhere however a single wipe with prng and verification should result in no errors if the drive is any good (and kept within it's operating temperature).

Regarding prngs such as xoro etc. in this case it really doesn't matter what you use. When doing a one's or zeros wipe the prng type is irrelevant.